esphome/issues #106: Can't add sensor in Home Assistant

bphermansson, 2019-03-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Esphome through Docker, Home Assistant in a Virtualenv. 

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

ADC

**Description of problem:**
I've built a sensor unit with the latest Esphome. It works fine and I can see it under Config/Integrations in Home Assistant. There it looks like this:

Green house moist
sensor.green_house_moist

I then try to add "sensor.green_house_moist" in groups.yaml to list it with my other sensors. But it won't show up? Do I have to do anything else?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: greenhousemoist
  platform: ESP8266
  board: d1_mini_lite

wifi:
  ssid: 'NETGEAR83'
  password: ''

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: adc
    pin: A0
    name: "Green house moist"
    update_interval: 60s

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-07:
This is a Home Assistant issue. A couple of releases ago Home Assistant switched to the new "Lovelace" UI and `groups` is no longer used. Try pressing the `Configure UI` in the top right of the HA interface and adding the entity from there.

bphermansson, 2019-03-07:
Ok great, that worked! Thank you for a fast reply!

closed 2019-03-07 by bphermansson