esphome/issues #107: mi flora battery level not updating

raccettura, 2019-03-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

https://esphome.io/components/sensor/xiaomi_miflora.html

**Description of problem:**
miflora sensor works several times over (i've got at least 7 or so working) just fine.  All sensors show except battery_level is never getting data.  Stock config other than changing name and mac address.

Running firmware v3.1.9 on all the sensors that are working. 

I've still got one or two sensors on v2 something but the batteries are dead so until I replace them I can't test them, but I've haven't managed to get older firmware to work with esphome.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-03-07:
I remember someone mentioning in the past that the devices don't send their battery level in every update, and there's no way to force them to send it. I believe it should come through eventually.

raccettura, 2019-03-07:
Interesting. I know the built in HA sensor seemed to capture it, but that seemed to run less often by default.  I tried adjusting the BLE sensor's scan_interval to slow down the polling but that doesn't seem to have made an effect.  I'm still seeing very frequent updates (but no battery).

```
esp32_ble_tracker:
  scan_interval: 900s
```

I'll leave it running for at least 12hrs and see if a battery level comes through at some point.

OttoWinter, 2019-03-08:
From past logs I looked at from other people it appears newer versions of the MiFlora just do not send this data via BLE.

The reason it shows up in other integrations is that they use a different form of communication. The ESPHome integration is entirely passive and just listens to what the MiFlora is sending itself; so it can't command it to send a battery level packet.

Because the battery level doesn't change too fast the device itself is probably just not sending it too often, and when it sends it the ESP32 might not be active atm.

This passive BLE tracking of course also means that having the ESP track the sensor does not affect the battery in any way (which is also a great advantage of esphome over the HA integration :D )

raccettura, 2019-03-08:
@OttoWinter oh that's really interesting.  I didn't realize it's a passive approach. I (wrongly) assumed it was doing the same as the python version and polling. 

I couldn't get it to work with the 2.x version of the firmware, when I upgraded to the latest v3 it worked, so I don't think I can help debug if battery level is a version thing or not.  AFAIK all the hardware is the same it's just software that's been incremented over time, my sensors were purchased from different places at different times (and some are even rebadged).  

Thanks for the info.

aalbul, 2019-03-11:
Hi guys and thanks for starting this conversation.
Currently i have more than 13 devices and i monitor all of them for more than a day.
None of them reported battery level yet.
Is it possible to support active polling for battery level once a day or so in a future releases?

OttoWinter, 2019-03-11:
@aalbul That would be a significant amount of work (and having active and passive BLE active at the same time on the ESP32 is surely very unstable). Honestly if it's just battery level I do not care. If you want to check battery level I think you can still go into the miflora app (or whatever app that uses) and check manually. It's not like that information is vital to the working of the device.

Plus it's not entirely known what exactly causes this. Could be that a recent fw update from the company just broke battery level reporting and it comes back soon. Can you try updating to 3.x as @raccettura mentioned? If that works I think we can close this issue.

aalbul, 2019-03-11:
All of them are running latest version (3.1.9 i think)

jokerigno, 2019-04-08:
Same issue. But add battery level is not a top priority for daily use.

james-fry, 2019-04-16:
+1 I also see this. Added a mi flora about 1 week back and batter was never updated.
Latest HA, Hassio ESPHome addon, and mi flora firmware.
IMO if this is not fixed its not a big deal... but the ESPhome docs should be updated to remove battery, or add a caveat that battery is deprecated

tiagofreire-pt, 2019-05-02:
This is weird as Home Assistant 0.92.1 on a Rpi 3B+ detects all this entities with no effort via miflora platform.

adrianmihalko, 2019-05-05:
Subscribe, I am experiencing same "problem".

OttoWinter, 2019-05-07:
@tiagofreire-pt As has been said before, the two use very different communication mechanisms.

CiTroNaK, 2019-05-10:
Same problem with Xiaomi MiFlora v3.2.1 with ESP32 as a tracker.

Aiace88, 2019-05-28:
> Lo stesso problema con Xiaomi MiFlora v3.2.1 con ESP32 come tracker.

idem

jkoenig72, 2019-05-30:
Same issue. Running for more than 48 hours now...

kastro123, 2019-05-31:
Same issue, battery level is never reported.
I am also using this script on RPi0W and am getting battery level also. Same flora device.
https://github.com/ThomDietrich/miflora-mqtt-daemon/blob/master/README.md

roygbiv856, 2019-06-04:
I am also having the same no battery reporting issue. With esphome is it still possible to change the reporting rate for the other sensors (temp, moisture, etc.)? Sometimes my sensors report over 200 times while I'm at work. This seems like it will kill the battery life

OttoWinter, 2019-07-11:
Closing as won't fix - I received my MiFlora and was able to confirm that the device does not send battery level data anymore (not even in a new packet type).

closed 2019-07-11 by OttoWinter

james-fry, 2019-07-11:
thanks for confirming, Otto

tiagofreire-pt, 2019-07-11:
@OttoWinter , that is weird as the Xiaomi app reports the battery percentage every time it connects to the sensor.

brandond, 2019-07-11:
I think Otto has said in the past that the app does active polling to get the battery level, while esphome passively listens for broadcast messages. The broadcasts no longer contain the battery level data.