esphome/issues #109: Dallas Temperature Sensors stopped working after upgrading to 1.11.X branch

mannkind, 2019-03-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

MacOS: 10.14.3  
Python: 2.7.15  
Pip: 19.0.3  
ESPHome(yaml): 1.11.X 

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**Affected component:**

Dallas

**Description of problem:**

After upgrading from 1.10.1 -> 1.11.X (0,1,2 doesn't matter) my Dallas temperature sensors stopped working. 
Debug over MQTT says...
> [19:22:59][W][sensor.dallas:084]:   Found no sensors!

**Problem-relevant YAML-configuration entries:**
```yaml
dallas:
  - pin: GPIO3

# Individual sensors
sensor:
  - platform: dallas
    #name: "First Floor Temperature"
    name: "Temperature 008"
    address: 0x6804175194DEFF28
    filters:
      - lambda: return x * (9.0/5.0) + 32.0;
    unit_of_measurement: "°F"

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

Downgrading to 1.10.1 makes the temperature sensors work again.
I looked over the change log/documentation, but nothing stood out to me.

OttoWinter, 2019-03-08:
That is interesting. I do not remember anything having changed there in 1.11 - dallas has been untouched for quite a few releases now (as well as the core components it uses).

I can say that my dallas sensors (ds18b20) still work fine in 1.11.X.

So from that I think the only explanation I can come up with now is that the timing has changed slightly in this version - maybe some methods are no longer in instruction cache because of new code pushing it out of there. These dallas sensors have always been quite strict with timing.

Can you try with latest dev version?

mannkind, 2019-03-09:
Thanks for the quick response! 

Indeed it does seem like my ds18b20s all start working again on the dev branch. 
I double-checked the older versions again for fun.

1.12.0-dev: working.
1.11.0-1.11.2: No working; no sensors found. 
1.10.1: working.

I did update the original post to note that I'm actually experiencing the issue on multiple devices, in case it matters. 

If there's nothing to be done, I can close this issue.

OttoWinter, 2019-03-09:
If it's fixed I think we can close it.

There was no change in dev version, so I think that makes it more likely to be what i initially thought: a timing issue.

closed 2019-03-09 by OttoWinter