esphome/issues #110: Magic Home LED Controller help

tatunts, 2019-03-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266

**Affected component:**
ESP8266 Software PWM

**Description of problem:**

I am trying to configure a Magig Home LED controller (https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller) previously flashed with Tasmota. On a specific device, I am using a regular with only LED strip, not RGB, by using only GPIO12 (green channel). I was able to create a firmware using a monochromatic light and the PWM software output, flashed and tested. I am able to turn on and off the controller but not the brightness levels. Anyone using this device as well? Thanks.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ${project_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_psw}
  fast_connect: true
  ap:
    ssid: ${friendly_name}
    password: ${ap_psw}

logger:

web_server:

api:
  password: ${api_psw}

ota:

##connect led to GPIO green channel (GPIO12)

output:
  - platform: esp8266_pwm
    pin: GPIO12
    id: main_light

light:
  - platform: monochromatic
    output: main_light
    name: ${light_entity_id} 
    default_transition_length: 1s
    effects:
      - strobe:
          name: Strobe Effect
          colors:
            - state: true
              brightness: 100%
              duration: 600ms
            - state: true
              duration: 600ms
              brightness: 0%
              
sensor:
  - platform: wifi_signal
    name: ${friendly_name} - Sinal Wifi
    update_interval: 60s

status_led:
  pin:
    number: GPIO2
    inverted: true
```

**Traceback (if applicable):**

```
not available
```

**Additional information and things you've tried:**

brandond, 2019-03-09:
What happens when you try to set the brightness? Does it crash, no change, slider isnt available in HA, etc?

tatunts, 2019-03-09:
Hello @brandond, thanks for your reply. The problem was fixed after a full restart of the device. I've tried so many yaml configurations, re-flashed many times, no thing worked. After disconnecting from the 12v power supply, the controller was fully functioning with dimming and transitions. Thanks!

closed 2019-03-09 by tatunts