esphome/issues #111: Guru Meditation Error with Rotary Encoder

elmicha, 2019-03-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker stable.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 DevKit V1.

**Affected component:**

https://esphome.io/components/sensor/rotary_encoder.html

**Description of problem:**

After turning the rotary encoder knob too fast or for too long, I get a Guru meditation error. If I turn the knob slowly, it works without problem, but if I turn it fast, the error occurs. `transition_length: 0s` doesn't help, and deleting the logger.log action also doesn't help.

**Problem-relevant YAML-configuration entries:**
```yaml

mqtt:
  broker: 192.168.7.77
  username: esphome
  topic_prefix: esphome/1

sensor:
  - platform: rotary_encoder
    name: "Rotary Encoder"
    pin_a: 19
    pin_b: 18
    filters:
      - or:
        - debounce: 0.1s
        - delta: 10
      - lambda: |-
          if (x < 0.0) return 0.0;
          if (x > 255.0) return 255.0;
          return x;
    on_value:
      then:
        - light.turn_on:
            id: led_id
            brightness: !lambda "return x / 255;"
            transition_length: 0s
        - logger.log:
            format: "brightness set to %.2f%s"
            args: [ 'x*100/255.0', '"%"' ]
    
output:
  - platform: ledc
    pin: 27
    frequency: 1000 Hz
    id: pwm_output

light:
  - platform: monochromatic
    output: pwm_output
    name: "LED-Platte"
    id: led_id
```

**Traceback (if applicable):**

```
[12:56:11][D][light.state:462]: 'LED-Platte' Turning ON
[12:56:11][D][light.state:469]:   Brightness: 41%
[12:56:11][D][main:052]: brightness set to 41.18%
[12:56:11][D][sensor.sensor:131]: 'Rotary Encoder': Sending state 111.00000 steps with 0 decimals of accuracy
[12:56:11][D][light.state:462]: 'LED-Platte' Turning ON
[12:56:11][D][light.state:469]:   Brightness: 44%
[12:56:11]Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
[12:56:11]Core 1 register dump:
[12:56:11]PC      : 0x40080f8d  PS      : 0x00060034  A0      : 0x8008104c  A1      : 0x3ffc0bb0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d esphome1 -t idedata
WARNING Decoded 0x40080f8d: esphome::sensor::RotaryEncoderSensor::process_state_machine_()
[12:56:13]A2      : 0x3ffb3550  A3      : 0x3ffb18c0  A4      : 0x80081edf  A5      : 0x3ffb18c0  
[12:56:13]A6      : 0x0000dea0  A7      : 0x3ffb196c  A8      : 0xbad00bad  A9      : 0x0000011b  
[12:56:13]A10     : 0x3ffb34f8  A11     : 0x3ffb18a0  A12     : 0x80081c80  A13     : 0x3ffb1890  
[12:56:13]A14     : 0x0000dea0  A15     : 0x3ffb196c  SAR     : 0x00000013  EXCCAUSE: 0x00000007  
[12:56:13]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[12:56:13]Core 1 was running in ISR context:
[12:56:13]EPC1    : 0x40062235  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080f8d
[12:56:13]
[12:56:13]Backtrace: 0x40080f8d:0x3ffc0bb0 0x40081049:0x3ffc0bd0 0x40081099:0x3ffc0bf0 0x40081ac5:0x3ffc0c10 0x40062232:0x00000000
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40080f8d: esphome::sensor::RotaryEncoderSensor::process_state_machine_()
WARNING Decoded 0x40081049: esphome::sensor::RotaryEncoderSensor::encoder_isr_()
WARNING Decoded 0x40081099: __onPinInterrupt at esp32-hal-gpio.c
WARNING Decoded 0x40081ac5: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/xtensa_vectors.S:1105
[12:56:13]
[12:56:13]Rebooting...
```

**Additional information and things you've tried:**

I tried the beta channel (same problem) and the dev channel (rotary knob doesn't do anything apparently).

OttoWinter, 2019-03-10:
That has been fixed here: https://github.com/esphome/esphome-core/pull/535

closed 2019-03-10 by OttoWinter

elmicha, 2019-03-10:
Thanks! But on the dev channel I don't get anything if I turn the knob. 

```
[17:08:06][I][application:097]: esphome-core version 1.12.0-dev compiled on Mar 10 2019, 16:10:04
...
[17:08:06][C][sensor.rotary_encoder:121]: Rotary Encoder 'Rotary Encoder'
[17:08:06][C][sensor.rotary_encoder:121]:   Unit of Measurement: 'steps'
[17:08:06][C][sensor.rotary_encoder:121]:   Accuracy Decimals: 0
[17:08:06][C][sensor.rotary_encoder:121]:   Icon: 'mdi:rotate-right'
[17:08:06][C][sensor.rotary_encoder:122]:   Pin A: GPIO19 (Mode: INPUT)
[17:08:06][C][sensor.rotary_encoder:123]:   Pin B: GPIO18 (Mode: INPUT)
```
I can change the brightness with MQTT, but the rotary encoder gives no log output and does not work.
I guess that should be a new issue?