esphome/issues #112: AJ-SR04M/JSN-SR04T v2.0 gives incorrect values

initrd, 2019-03-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU v2

**Affected component:**
https://esphome.io/components/sensor/ultrasonic.html

**Description of problem:**
AJ-SR04M/JSN-SR04T v2.0 are generally the most available waterproof ultrasonic sensors. Using it with the ultrasonic sensor component does not give good values. v2.0 seems to be incompatible with the older version, which worked fine with code for the HC-SR04. Here is a log sample:

```
[17:49:32][V][sensor.ultrasonic:053]: Echo took 10µs (0.001715m)
[17:49:32][V][sensor.sensor:020]: 'Level 1': Received new state 0.001715
[17:49:38][V][sensor.ultrasonic:053]: Echo took 10µs (0.001715m)
[17:49:38][V][sensor.sensor:020]: 'Level 1': Received new state 0.001715
[17:49:44][V][sensor.ultrasonic:053]: Echo took 32µs (0.005488m)
[17:49:44][V][sensor.sensor:020]: 'Level 1': Received new state 0.005488
[17:49:50][V][sensor.ultrasonic:053]: Echo took 10µs (0.001715m)
[17:49:50][V][sensor.sensor:020]: 'Level 1': Received new state 0.001715
[17:49:56][V][sensor.ultrasonic:053]: Echo took 12µs (0.002058m)
[17:49:56][V][sensor.sensor:020]: 'Level 1': Received new state 0.002058
[17:50:02][V][sensor.ultrasonic:053]: Echo took 9µs (0.001543m)
[17:50:02][V][sensor.sensor:020]: 'Level 1': Received new state 0.001543
[17:50:08][V][sensor.ultrasonic:053]: Echo took 13µs (0.002229m)
[17:50:08][V][sensor.sensor:020]: 'Level 1': Received new state 0.002229
[17:50:14][V][sensor.ultrasonic:053]: Echo took 10µs (0.001715m)
[17:50:14][V][sensor.sensor:020]: 'Level 1': Received new state 0.001715
[17:50:20][V][sensor.ultrasonic:053]: Echo took 6µs (0.001029m)
[17:50:20][V][sensor.sensor:020]: 'Level 1': Received new state 0.001029
[17:50:26][V][sensor.ultrasonic:053]: Echo took 13µs (0.002229m)
[17:50:26][V][sensor.sensor:020]: 'Level 1': Received new state 0.002229
[17:50:26][D][sensor.sensor:131]: 'Level 1': Sending state 0.00214 m with 2 decimals of accuracy
```

**Problem-relevant YAML-configuration entries:**
```yaml
 sensor:
   - platform: ultrasonic
     trigger_pin: D7
     echo_pin: D6
     name: "Level 1"
     update_interval: 6s
     filters:
       - filter_nan:
       - sliding_window_moving_average:
           window_size: 10
           send_every: 10
           send_first_at: 10

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
Here is the [spec sheet [PDF]](https://www.jahankitshop.com/getattach.aspx?id=4635&Type=Product) of the sensor. There are posts in the [Arduino forum](https://forum.arduino.cc/index.php?topic=474759.0) about similar problems and fixes (post #16 from that link).

I'll be happy to ship a sensor or two for testing!

OttoWinter, 2019-03-10:
The pulse_time option has already been added in dev and will be part of 1.12.0

closed 2019-03-10 by OttoWinter

initrd, 2019-03-14:
Thank you for the quick response! I'm not able to get the sensor working with either 1.12.0b1 or esphome_core_version: dev (1.13.0-dev), even without pulse_time: 20us. In both the instances, the sensor times out:

1.12.0b1, default pulse_time:
```
[20:57:39][I][application:097]: esphome-core version 1.12.0b1 compiled on Mar 13 2019, 20:56:33
[20:57:39][C][sensor.ultrasonic:061]:   Echo Pin: GPIO12 (Mode: INPUT)
[20:57:39][C][sensor.ultrasonic:062]:   Trigger Pin: GPIO13 (Mode: OUTPUT)
[20:57:39][C][sensor.ultrasonic:063]:     Pulse time: 10 µs
[20:57:39][C][sensor.ultrasonic:064]:     Timeout: 58309 µs
...
[20:57:53][D][sensor.ultrasonic:055]: Distance measurement timed out!
[20:57:53][V][sensor.sensor:020]: 'Level 1': Received new state nan
[20:57:59][D][sensor.ultrasonic:055]: Distance measurement timed out!
[20:57:59][V][sensor.sensor:020]: 'Level 1': Received new state nan
[20:58:05][D][sensor.ultrasonic:055]: Distance measurement timed out!
[20:58:05][V][sensor.sensor:020]: 'Level 1': Received new state nan
```

1.12.0b1, pulse_time: 20us:
```
[21:03:35][I][application:097]: esphome-core version 1.12.0b1 compiled on Mar 13 2019, 20:59:34
...
[21:03:35][C][sensor.ultrasonic:061]:   Echo Pin: GPIO12 (Mode: INPUT)
[21:03:35][C][sensor.ultrasonic:062]:   Trigger Pin: GPIO13 (Mode: OUTPUT)
[21:03:35][C][sensor.ultrasonic:063]:     Pulse time: 20 µs
[21:03:35][C][sensor.ultrasonic:064]:     Timeout: 58309 µs
[21:03:35][C][sensor.ultrasonic:065]:   Update Interval: 6000 ms
...
[21:03:46][D][sensor.ultrasonic:055]: Distance measurement timed out!
[21:03:46][V][sensor.sensor:020]: 'Level 1': Received new state nan
[21:03:52][D][sensor.ultrasonic:055]: Distance measurement timed out!
[21:03:52][V][sensor.sensor:020]: 'Level 1': Received new state nan
[21:03:58][D][sensor.ultrasonic:055]: Distance measurement timed out!
[21:03:58][V][sensor.sensor:020]: 'Level 1': Received new state nan
[21:04:04][D][sensor.ultrasonic:055]: Distance measurement timed out!
```

1.13.0-dev, default pulse_time:
```
[20:46:34][I][application:097]: esphome-core version 1.13.0-dev compiled on Mar 13 2019, 20:44:35
[20:46:40][D][sensor.ultrasonic:055]: Distance measurement timed out!
[20:46:40][V][sensor.sensor:020]: 'Level 1': Received new state nan
[20:46:42]pm open,type:0 0
[20:46:46][D][sensor.ultrasonic:055]: Distance measurement timed out!
[20:46:46][V][sensor.sensor:020]: 'Level 1': Received new state nan
```

At first I thought my hardware was broken, but reverting to 1.11.2 at least did not show timeouts.