esphome/issues #113: max6675 has 30% failure rate

poldim, 2019-03-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Rpi
**ESP (ESP32/ESP8266, Board/Sonoff):**

[ESP32 NodeMCU
](https://hackster.imgix.net/uploads/attachments/475327/wemos-esp32_vUeL0JD5hZ.jpg)
**Affected component:**

https://esphome.io/components/sensor/max6675.html

**LOG:**
```
[00:09:34][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x05FC)
[00:09:39][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x04EC)
[00:09:44][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=33.8°C
[00:09:49][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=29.5°C
[00:09:54][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x0514)
[00:09:59][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x04CC)
[00:10:04][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=33.2°C
[00:10:09][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=29.5°C
[00:10:14][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x0364)
[00:10:19][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x0324)
[00:10:24][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=23.8°C
[00:10:29][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=23.0°C
[00:10:34][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=22.2°C
[00:10:39][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=21.5°C
[00:10:44][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=21.0°C
[00:10:49][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=20.5°C
[00:10:54][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=20.2°C
[00:10:59][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x027C)
[00:11:04][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x026C)
[00:11:09][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=19.0°C
[00:11:14][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=18.8°C
[00:11:19][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x0254)
[00:11:24][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=18.5°C
[00:11:29][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=18.0°C
[00:11:34][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=18.0°C
[00:11:39][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=18.0°C
[00:11:44][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x023C)
[00:11:49][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=17.8°C
[00:11:54][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=67.2°C
[00:11:59][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=129.2°C
[00:12:04][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=103.8°C
[00:12:09][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=80.5°C
[00:12:14][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=64.5°C
[00:12:19][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=53.5°C
[00:12:24][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=46.2°C
[00:12:29][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=41.0°C
[00:12:34][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x04B4)
[00:12:39][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=35.0°C
[00:12:44][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x0424)
[00:12:49][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x03F4)
[00:12:54][W][sensor.max6675:058]: Got invalid value from MAX6675Sensor (0x03CC)
[00:12:59][D][sensor.max6675:064]: 'BBQ Temp': Got temperature=29.2°C
```

**Description of problem:**
Without physically changing the setup (device laying on the table), the unit appears to be unable to read ~30% of the values are unable to be read.

**Problem-relevant YAML-configuration entries:**
```
spi:
  miso_pin: 19
  clk_pin: 18

sensor:
  - platform: max6675
    name: "BBQ Temp"
    cs_pin: 15
    update_interval: 5s
```

**Additional information and things you've tried:**

- Physically checked connections
- Moved data pin
- Searched all the issues
- Searched to make sure this is the correct board type to use. [lolin32](https://github.com/esphome/esphome-core/issues/121) does not work.

kfirsa, 2019-05-02:
Well it seems that we are working on the same bbq project.
in my use case the reading are not accurate as well
```
[22:48:41][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.2Â°C
[22:48:46][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.2Â°C
[22:48:51][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01E4)
[22:48:56][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.0Â°C
[22:49:01][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.0Â°C
[22:49:06][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.2Â°C
[22:49:11][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:49:16][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.0Â°C
[22:49:21][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:49:26][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=15.0Â°C
[22:49:31][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01DC)
[22:49:32][I][ota:152]: Boot seems successful, resetting boot loop counter.
[22:49:36][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01DC)
[22:49:41][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:49:46][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01DC)
[22:49:51][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:49:56][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01DC)
[22:50:01][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:50:06][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:50:11][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.5Â°C
[22:50:16][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:50:21][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01DC)
[22:50:26][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:50:31][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:50:36][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.5Â°C
[22:50:41][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:50:46][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.5Â°C
[22:50:51][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:50:56][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.8Â°C
[22:51:01][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.5Â°C
[22:51:06][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:11][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:16][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:21][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:26][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:31][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:36][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x01D4)
[22:51:41][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.5Â°C
[22:51:46][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.5Â°C
[22:51:51][D][sensor.max6675:056]: 'BBQ Temperature': Got temperature=14.2Â°C
```
i am using a wemos d1 mini , re-flashed it to default and added a very basic code.
used the same pins, hardware, etc
```C++
// Sample Wemos D1 with MAX6675  Sketch
// D1 Wiring 
//Pin   Function                        ESP-8266 Pin
//TX    TXD                             TXD
//RX    RXD                             RXD
//A0    Analog input, max 3.3V input    A0
//D0    IO                              GPIO16
//D1    IO, SCL                         GPIO5
//D2    IO, SDA                         GPIO4
//D3    IO, 10k Pull-up                 GPIO0
//D4    IO, 10k Pull-up, BUILTIN_LED    GPIO2
//D5    IO, SCK                         GPIO14
//D6    IO, MISO                        GPIO12
//D7    IO, MOSI                        GPIO13
//D8    IO, 10k Pull-down, SS           GPIO15
//G     Ground                          GND
//5V    5V                              -
//3V3   3.3V                            3.3V
//RST   Reset                           RST

#include "max6675.h"

int ktcSO = 12;
[... 36 lines cut ...]
22:47:28.008 -> Deg C = 75.00	 Deg F = 167.00
22:47:29.067 -> Deg C = 74.00	 Deg F = 165.20
22:47:30.133 -> Deg C = 73.25	 Deg F = 163.85
22:47:31.226 -> Deg C = 73.00	 Deg F = 163.40
22:47:32.281 -> Deg C = 72.00	 Deg F = 161.60
22:47:33.336 -> Deg C = 69.25	 Deg F = 156.65
22:47:34.421 -> Deg C = 64.50	 Deg F = 148.10
22:47:35.475 -> Deg C = 60.25	 Deg F = 140.45
22:47:36.537 -> Deg C = 57.50	 Deg F = 135.50
22:47:37.594 -> Deg C = 55.50	 Deg F = 131.90
22:47:38.683 -> Deg C = 53.50	 Deg F = 128.30
22:47:39.731 -> Deg C = 51.50	 Deg F = 124.70
22:47:40.826 -> Deg C = 51.25	 Deg F = 124.25
22:47:41.890 -> Deg C = 49.75	 Deg F = 121.55
22:47:42.942 -> Deg C = 48.00	 Deg F = 118.40
22:47:43.997 -> Deg C = 47.50	 Deg F = 117.50
22:47:45.088 -> Deg C = 45.75	 Deg F = 114.35
22:47:46.137 -> Deg C = 45.00	 Deg F = 113.00
22:47:47.194 -> Deg C = 44.00	 Deg F = 111.20
22:47:48.257 -> Deg C = 43.50	 Deg F = 110.30
22:47:49.342 -> Deg C = 43.25	 Deg F = 109.85
22:47:50.403 -> Deg C = 42.50	 Deg F = 108.50
22:47:51.455 -> Deg C = 42.50	 Deg F = 108.50
22:47:52.540 -> Deg C = 42.25	 Deg F = 108.05
```

poldim, 2019-05-02:
@kfirsa Nice! I didn't think to test mine with basic arduino code, but I wouldn't want to drag my programming desktop over to the grill to read the serial bus anyway... :-p

I'm assuming you did this on the latest version of ESPHome, although looking at the history, it hasn't changed in a few months: https://github.com/esphome/esphome-core/blob/dev/src/esphome/sensor/max6675_sensor.cpp

@OttoWinter  Is line 55 correct here `  float temperature = float(val >> 3) / 4.0f;`?

kfirsa, 2019-05-03:
@poldim have you verify the accuracy of your sensor reading?
i have made two changes in the code that seems to help.
```C++
void MAX6675Sensor::read_data_() {
  this->enable();
  delay(100);
  uint8_t data[2];
  this->read_array(data, 2);
  uint16_t val = data[1] | (uint16_t(data[0]) << 8);
  val = val*1.95;
  this->disable();
``` 
increasing the delay reduce the number of failed samples occasionally 
multiply val in 1.95 for accurate reading 

maybe it could be added as a factor in the yaml file if the sensor needs calibration

OttoWinter, 2019-05-08:
@kfirsa According to the datasheet the current implementation is correct. Conversion time is 0.22s max (line 17) and t_css is min 100ns (esphome uses 1ms, which is plenty). Also the data conversion is correct - bits 3 through 14 represent data and bit 2 represents an open thermocouple input (if set, it is flagged as invalid data).

@poldim Yes the line is correct according to the datasheet.

@kfirsa You could try adding more delay to the conversion (against what the datasheet says) to 250ms (line 17). For the multiplication, see sensor filters. Please also post how you have done your wiring - too long cables and/or if T- is not grounded would result in this error.

gavindsilver, 2019-05-15:
im getting the same, how can i calibrate or adjust these figures? is it possible via the yaml?

`[04:13:47][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x041C)
[04:13:57][D][sensor.max6675:056]: 'Viking Range Temp': Got temperature=34.0°C
[04:14:07][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x045C)
[04:14:17][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x0484)
[04:14:27][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x04AC)
[04:14:37][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x04CC)
[04:14:47][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x04F4)
[04:14:57][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x0514)
[04:15:07][D][sensor.max6675:056]: 'Viking Range Temp': Got temperature=41.8°C
[04:15:17][D][sensor.max6675:056]: 'Viking Range Temp': Got temperature=42.8°C
[04:15:27][D][sensor.max6675:056]: 'Viking Range Temp': Got temperature=44.0°C
[04:15:37][D][sensor.max6675:056]: 'Viking Range Temp': Got temperature=45.0°C
[04:15:47][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x05C4)
[04:15:57][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x05E4)
[04:16:07][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x05FC)
[04:16:17][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x0624)
[04:16:27][D][sensor.max6675:056]: 'Viking Range Temp': Got temperature=50.0°C
[04:16:37][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x066C)
[04:16:47][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x068C)
[04:16:57][W][sensor.max6675:050]: Got invalid value from MAX6675Sensor (0x06A4)`

kfirsa, 2019-05-15:
@OttoWinter, the probes do not touch a metal surface (as i am testing) so grounding the probe/metal to ground should not be an issue in this stage.
that dose not explain the stable results in the "native arduino" on the same hardware.
the entire setup consist of 40cm probe and spi/power cables are 10cm jumpers.

found some reference about adding a 0.01uf cap on the probe leads that could decrease the failures, have not tested it yet.

i am now having doubts about the probe and 6675 and going to order the 31855 as a replacement, checking further those components (6675 and 31855) they are not ideal for the task.

DrJeff, 2019-05-20:
I get similar problems in ESPHome reads @ 50ish but with Arduino flashed it reads as it should 80ish no drops in arduino even with direct read of SPI no library
```
#define CLK 9
#define DBIT 10 // so
#define CS 13

#include <SoftwareSerial.h>

int v = 0;
float Ctemp, Ftemp;

void setup()   {
  Serial.begin(9600);
  pinMode(CLK, OUTPUT);
  pinMode(DBIT, INPUT);
  pinMode(CS, OUTPUT);
  digitalWrite(CS, HIGH);
  digitalWrite(CLK, LOW);
}

void loop()   {
  v = spiRead();
  if (v == -1)   {
    Serial.print("No sensor \n");
  }
  else   {
    Ctemp =  v * 0.25;
    Ftemp = (Ctemp * 9 / 5) + 32;
    Serial.println(Ftemp);
   // Serial.print("\n");
  }
  delay(1000);
}

int spiRead()   {
  int value = 0;
  digitalWrite(CS,LOW);
  delay(2);
  digitalWrite(CS,HIGH);
  delay(220);

  /* Read the chip and return the raw temperature value */
  /* Bring CS pin low to allow us to read the data from
   the conversion process */

  digitalWrite(CS,LOW);
  /* Cycle the clock for dummy bit 15 */
  digitalWrite(CLK,HIGH);
  delay(1);
  digitalWrite(CLK,LOW);

  /*
   Read bits 14-3 from MAX6675 for the Temp. Loop for each bit reading
   the value and storing the final value in 'temp'
   */

  for (int i=14; i>=0; i--) {
    digitalWrite(CLK,HIGH);
    value += digitalRead(DBIT) << i;
    digitalWrite(CLK,LOW);
  }
  // check bit D2 if HIGH no sensor
  if ((value & 0x04) == 0x04) return -1;

  // shift right three places
  return value >> 3;
}
```

poldim, 2019-05-20:
How do I use the original file for this as a custom component?
https://github.com/adafruit/MAX6675-library/blob/master/max6675.h

Hopefully this will give the accurate data we're after...

ax42, 2019-05-24:
Getting the same issue (ironically also making a BBQ sensor).

thoj, 2019-05-30:
Same issue here. This is for a wood stove project though. I have tested the MAX6675 chip with the same probe and an Arduino mega and it works all the time and has no accuracy issues.

Marlor, 2019-06-07:
Still happening in ESPHome 1.13.4

Analog thermometer: ~24°C
https://github.com/YuriiSalimov/MAX6675_Thermocouple implementation: 25°C 
ESPHome 1.13.4: 13°C and alot of invalid values

Same placement, same hardware only difference is the Firmware

Marlor, 2019-06-08:
Fixed in dev thanks to "Add more efficient SPI implementation"
No more invalid values and the right temperature.

OttoWinter, 2019-06-09:
@Marlor Entirely not intended but I'll take it lol. Now that I think of it, the SPI clock phase could have been the problem. Great that this has been resolved.

closed 2019-06-09 by OttoWinter

poldim, 2019-06-10:
How do I get the dev version if I’m on a docket container?  Is there a
container version for dev?