esphome/issues #114: Sonoff Basic - Unable to upload new firmware following update to 1.11.2

bluesheep, 2019-03-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on RPi3.
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic with previously (1.10.2) flashed firmware.
**Affected component:**

N/A

**Description of problem:**
I have updated to 1.11.2 and can no longer upload the compiled firmware to the Sonoff Basic.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: sonoffbasic1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: <SNIP>
  password: <SNIP>
  
  manual_ip:
    static_ip: <SNIP>
    gateway: 192.168.1.1
    subnet: 255.255.255.0

api:
  password: <SNIP>

logger:

ota:
  password: <SNIP>

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/sonoffbasic2
Processing sonoffbasic2 (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.11.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.1.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/sonoffbasic2/src/main.cpp.o
Linking .pioenvs/sonoffbasic2/firmware.elf
Retrieving maximum program size .pioenvs/sonoffbasic2/firmware.elf
Building .pioenvs/sonoffbasic2/firmware.bin
Checking size .pioenvs/sonoffbasic2/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  40.3% (used 33016 bytes from 81920 bytes)
PROGRAM: [===       ]  32.2% (used 330124 bytes from 1023984 bytes)
========================= [SUCCESS] Took 22.32 seconds =========================
INFO Successfully compiled program.
INFO Connecting to <SNIP>
INFO Uploading /config/esphome/sonoffbasic2/.pioenvs/sonoffbasic2/firmware.bin (334272 bytes)
ERROR Error binary size: Error: Couldn't prepare flash memory for update. Is the binary too big? Please try restarting the ESP.
```

**Additional information and things you've tried:**
I have tried:
* restarting the Sonoff Basic (by turning off at wall)
* Stripping out yaml file after the `ota:` section to attempt a smaller compile size

OttoWinter, 2019-03-11:
What do you see when viewing the logs in another window while performing the OTA update?

bluesheep, 2019-03-11:
This is what I've got:
```
INFO Reading configuration...
INFO Starting log output from <SNIP> using esphome API
INFO Connecting to <SNIP>:6053 (<SNIP>)
INFO Successfully connected to <SNIP>
[13:57:06][I][application:097]: You're running esphomelib v1.10.1 compiled on Mar  9 2019, 17:31:15
[13:57:06][C][status_led:023]: Status LED:
[13:57:06][C][status_led:024]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[13:57:06][C][wifi:341]: WiFi:
[13:57:06][C][wifi:240]:   SSID: <SNIP>
[13:57:06][C][wifi:241]:   IP Address: <SNIP>
[13:57:06][C][wifi:243]:   BSSID: 7E:D2:94:B4:D6:B0
[13:57:06][C][wifi:245]:   Hostname: 'sonoffbasic1'
[13:57:06][C][wifi:250]:   Signal strength: -70 dB ▂▄▆█
[13:57:06][C][wifi:251]:   Channel: 13
[13:57:06][C][wifi:252]:   Subnet: 255.255.255.0
[13:57:06][C][wifi:253]:   Gateway: 192.168.1.1
[13:57:06][C][wifi:254]:   DNS1: 0.0.0.0
[13:57:06][C][wifi:255]:   DNS2: 0.0.0.0
[13:57:06][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'Sonoff Basic Button'
[13:57:06][C][binary_sensor.gpio:024]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[13:57:06](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[13:57:06](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[13:57:06][C][switch.gpio:066]:   Restore Mode: Restore (Default to OFF)
[13:57:06][C][logger:099]: Logger:
[13:57:06][C][logger:100]:   Level: DEBUG
[13:57:06][C][logger:101]:   Log Baud Rate: 115200
[13:57:06][C][api:072]: API Server:
[13:57:06][C][api:073]:   Address: <SNIP>:6053
[13:57:06][C][ota:129]: Over-The-Air Updates:
[13:57:06][C][ota:130]:   Address: <SNIP>:8266
[13:57:06][C][ota:132]:   Using Password.
[13:57:50][D][ota:180]: Starting OTA Update from 192.168.1.21...
[13:57:50][W][ota:297]: Preparing OTA partition failed! Is the binary too big?
```

brandond, 2019-03-11:
If you did your initial flash upload using an older release, it inherited from PlatformIO a significant (but unused) SPIFFS partition that reduced the amount of space available for OTA on 1mb devices to an almost unusable level. Unfortunately, the flash layout cannot currently be changed via OTA - so you'll have to reflash via UART in order to recover the space.

@OttoWinter since you've already rebuilt a bunch of the OTA stuff, do you think you might be able to allow for resizing the flash layout via OTA? I don't see any reason why you couldn't overwrite the SPIFFS space with an OTA image that redefined the SPIFFS size and offset for the next boot, it's just that right now the OTA code doesn't allow for that.

OttoWinter, 2019-03-11:
@brandond I think it already does that (for ESP8266s). On an ESP8266 OTA upload the entire fw is written, including partitions etc. AFAIK, esphome does already write over all SPIFFS data.

I think SPIFFS is even just a virtual fs that accesses the ESPs flash directly from the fw. I think there's another issue here.

brandond, 2019-03-12:
I am pretty sure it just rewrites the 'sketch' portion of flash. Last time I looked, the OTA process:
1. Calculate SPIFFS_START - OTA_SIZE (rounded up to block size)
2. If result is before end of current sketch, fail
3. Write OTA to flash, starting at calculated offset, ending at start of SPIFFS
4. Set bootloader flags in RTC memory instructing it to copy new sketch into place on boot
5. Reboot

The bootloader somewhat blindly copies the new sketch from the temporary location to the start of sketch memory, and then boots it. There is no way to update the bootloader from OTA, and there is a completely different OTA process for writing files to the SPIFFS partition.

Seems to me like a better option would be to use an A/B partition layout and just switch which one was booted so you could always fall back to a known-good setting. This is what commercial frameworks like Tuya do.

OttoWinter, 2019-03-12:
> Calculate SPIFFS_START

Yes, but SPIFFS_START is just a constant that gets included in the fw binary (see ld script). So when you upload an esphome ota you will essentially remove the spiffs. It may still have data until it gets overridden, but that space is free to use.

> Seems to me like a better option would be to use an A/B partition layout

Yes that's also what's done for the ESP32 - for the ESP8266 we use nonos sdk however where this is nit possible AFAIK (bootloader would need to support it). RTOS SDK does support it but the arduino core is not using it.

Also up until now I have not seen much benefit to that. ESP32 has been using that process for esphome and the OTA safe mode is already good enough.

james-davis-protonmail, 2019-03-12:
Don't you need to add the board floash mode to it? like this:
```
esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
```

I have just bought 10 sonoff basic (some of them I have modified by removing the led and adding a PIR sensor) and did not have any problem flashing them.

OttoWinter, 2019-03-12:
@james-davis-protonmail No you don't. That is the default since 1.11

bluesheep, 2019-03-12:
In the interests of full diligence I added the `board_flash_mode: dout` to the config. I can confirm that it didn't resolve the issue.

bluesheep, 2019-03-12:
I was reminded that I read about a similar issue with Tasmota (I migrated from that to esphome)

Details are on their [wiki page](https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade) (such as I understand them)

bachya, 2019-03-21:
I'm experiencing the same thing: after a successful initial flash (using an existing Tasmota install as the "uploading mechanism"), flashing new builds via OTA (using Docker) on my Sonoff Basics fails.

**Example Flash and Output:**

```
$ docker run --rm -v $(pwd)/esphome:/config -it esphome/esphome office_salt_lamp.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting office_salt_lamp/.pioenvs
INFO Compiling app...
INFO Running:  platformio run -d office_salt_lamp
...
INFO Successfully compiled program.
INFO Connecting to 172.16.11.16
INFO Uploading office_salt_lamp/.pioenvs/office_salt_lamp/firmware.bin (331696 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error receiving acknowledge receive OK: timed out
WARNING OTA v2 method failed. Trying with legacy OTA...
INFO Starting on 0.0.0.0:19092
INFO Upload size: 331696
INFO Sending invitation to 172.16.11.16
..........
ERROR No response from the ESP
```

**Example Logs on Device (during upload):**

```
$ docker run --rm -v $(pwd)/esphome:/config -it esphome/esphome office_salt_lamp.yaml logs
INFO Reading configuration...
INFO Starting log output from 172.16.11.16 using esphome API
INFO Connecting to 172.16.11.16:6053 (172.16.11.16)
INFO Successfully connected to 172.16.11.16
[17:03:30][I][application:097]: esphome-core version 1.12.1 compiled on Mar 20 2019, 18:24:29
[17:03:30][C][wifi:371]: WiFi:
[17:03:30][C][wifi:253]:   SSID: 'MySSID'
[17:03:30][C][wifi:254]:   IP Address: 172.16.11.16
[17:03:30][C][wifi:256]:   BSSID: FE:EC:DA:32:AA:47
[17:03:30][C][wifi:257]:   Hostname: 'office_salt_lamp'
[17:03:30][C][wifi:261]:   Signal strength: -55 dB ▂▄▆█
[17:03:30][C][wifi:262]:   Channel: 11
[17:03:30][C][wifi:263]:   Subnet: 255.255.255.0
[17:03:30][C][wifi:264]:   Gateway: 172.16.11.1
[17:03:30][C][wifi:265]:   DNS1: 1.1.1.1
[17:03:30][C][wifi:266]:   DNS2: 1.0.0.1
[17:03:30][C][switch.gpio:049]: GPIO Switch 'Office Salt Lamp'
[17:03:30][C][switch.gpio:050]:   Pin: GPIO12 (Mode: OUTPUT)
[17:03:30][C][switch.gpio:066]:   Restore Mode: Restore (Defaults to OFF)
[17:03:30][C][logger:142]: Logger:
[17:03:30][C][logger:143]:   Level: DEBUG
[17:03:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[17:03:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[17:03:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[17:03:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[17:03:30][C][ota:127]: Over-The-Air Updates:
[17:03:30][C][ota:128]:   Address: 172.16.11.16:8266
[17:05:35][D][ota:179]: Starting OTA Update from 172.16.10.19...
[17:05:35][W][ota:314]: Preparing OTA partition failed! 'ERROR[0]: No Error
```

I've tried `dev`, `beta`, and `latest` Docker images with the same results.

OttoWinter, 2019-03-22:
@bachya

> ERROR Error receiving acknowledge receive OK: timed out

That should only happen with really bad network connections, but anyway if you see that message the OTA actually passed correctly - your PC just didn't get the acknowledgement back from the ESP saying that the OTA finished. It will still apply.

> [17:05:35][W][ota:314]: Preparing OTA partition failed! 'ERROR[0]: No Error

That is more interesting. The only way I think that could happen is when you start a new OTA update when the old one failed while uploading. A power cycle should fix this. However I just found how to reset the update after a failed, which should fix this.

closed 2019-03-24 by OttoWinter

bachya, 2019-03-25:
@OttoWinter Should this be available in the latest `esphome/esphome:dev` Docker image? I pulled the most up-to-date version of that image this morning and got the following results (both before and after manually restarting the Sonoff):

**Flash and Output:**

```
INFO Successfully compiled program.
INFO Connecting to 172.16.11.16
INFO Uploading office_salt_lamp/.pioenvs/office_salt_lamp/firmware.bin (328752 bytes)
ERROR Error binary size: Error: Couldn't prepare flash memory for update. Is the binary too big? Please try restarting the ESP.
WARNING OTA v2 method failed. Trying with legacy OTA...
INFO Starting on 0.0.0.0:29256
INFO Upload size: 328752
INFO Sending invitation to 172.16.11.16
..........
ERROR No response from the ESP
```

**Device Logs:**

```
INFO Reading configuration...
INFO Starting log output from 172.16.11.16 using esphome API
INFO Connecting to 172.16.11.16:6053 (172.16.11.16)
INFO Successfully connected to 172.16.11.16
[15:07:22][I][application:097]: esphome-core version 1.12.1 compiled on Mar 20 2019, 18:24:29
[15:07:22][C][wifi:371]: WiFi:
[15:07:22][C][wifi:253]:   SSID: 'MY_SSID'
[15:07:22][C][wifi:254]:   IP Address: 172.16.11.16
[15:07:22][C][wifi:256]:   BSSID: FE:EC:DA:32:AA:47
[15:07:22][C][wifi:257]:   Hostname: 'office_salt_lamp'
[15:07:22][C][wifi:261]:   Signal strength: -48 dB ▂▄▆█
[15:07:22][C][wifi:262]:   Channel: 11
[15:07:22][C][wifi:263]:   Subnet: 255.255.255.0
[15:07:22][C][wifi:264]:   Gateway: 172.16.11.1
[15:07:22][C][wifi:265]:   DNS1: 1.1.1.1
[15:07:22][C][wifi:266]:   DNS2: 1.0.0.1
[15:07:22][C][switch.gpio:049]: GPIO Switch 'Office Salt Lamp'
[15:07:22][C][switch.gpio:050]:   Pin: GPIO12 (Mode: OUTPUT)
[15:07:22][C][switch.gpio:066]:   Restore Mode: Restore (Defaults to OFF)
[15:07:22][C][logger:142]: Logger:
[15:07:22][C][logger:143]:   Level: DEBUG
[15:07:22](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:07:22](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:07:22](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:07:22](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:07:22](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:07:22](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:07:43][D][ota:179]: Starting OTA Update from 172.16.10.19...
[15:07:43][D][ota:353]: OTA in progress: 0.2%
[15:07:44][D][ota:353]: OTA in progress: 8.8%
[15:07:45][D][ota:353]: OTA in progress: 9.3%
[15:07:47][D][ota:353]: OTA in progress: 9.8%
[15:07:48][D][ota:353]: OTA in progress: 10.2%
[15:07:49][D][ota:353]: OTA in progress: 21.3%
[15:07:50][D][ota:353]: OTA in progress: 33.2%
[15:07:51][D][ota:353]: OTA in progress: 44.5%
[15:07:52][D][ota:353]: OTA in progress: 56.2%
[15:07:53][D][ota:353]: OTA in progress: 57.7%
[15:07:54][D][ota:353]: OTA in progress: 58.2%
[15:07:56][D][ota:353]: OTA in progress: 64.9%
[15:07:57][D][ota:353]: OTA in progress: 76.1%
[15:07:58][D][ota:353]: OTA in progress: 87.0%
[15:07:59][D][ota:353]: OTA in progress: 93.6%
[15:07:59][W][ota:410]: Error client disconnected while receiving data!
[15:07:59][W][ota:386]: Update end failed! Error: ERROR[0]: No Error

[15:07:59][D][api:073]: Disconnecting Home Assistant 0.90.1 (172.16.10.55)
[15:07:59][D][api:546]: Client 'Home Assistant 0.90.1 (172.16.10.55)' connected successfully!
[15:08:33][D][ota:179]: Starting OTA Update from 172.16.10.19...
[15:08:33][W][ota:314]: Preparing OTA partition failed! 'ERROR[0]: No Error
```

brandond, 2019-03-25:
Note that this fix is client-side, so you'll have to successfully flash the new code onto the device before you can take advantage of it. If you're currently unable to OTA a device due to this error, it won't fix that for you.

bachya, 2019-03-26:
I believe there's still an issue here.

I flashed a base firmware via serial to reset things. Then, I used the `dev` Docker image to upload new firmware via OTA. Below are the console outputs:

**Flash and Output:**

```
NFO Successfully compiled program.
INFO Connecting to 172.16.11.16
INFO Uploading office_salt_lamp/.pioenvs/office_salt_lamp/firmware.bin (337968 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error receiving acknowledge receive OK: timed out
WARNING OTA v2 method failed. Trying with legacy OTA...
INFO Starting on 0.0.0.0:51323
INFO Upload size: 337968
INFO Sending invitation to 172.16.11.16
..........
ERROR No response from the ESP
```

**Device Logs:**

```
[16:55:32][D][ota:179]: Starting OTA Update from 172.16.10.19...
[16:55:32][D][ota:353]: OTA in progress: 0.2%
[16:55:33][D][ota:353]: OTA in progress: 13.5%
[16:55:34][D][ota:353]: OTA in progress: 23.1%
[16:55:35][D][ota:353]: OTA in progress: 27.9%
[16:55:36][D][ota:353]: OTA in progress: 41.4%
[16:55:37][D][ota:353]: OTA in progress: 43.0%
[16:55:39][D][ota:353]: OTA in progress: 52.3%
[16:55:40][D][ota:353]: OTA in progress: 52.8%
[16:55:41][D][ota:353]: OTA in progress: 53.3%
[16:55:42][D][ota:353]: OTA in progress: 53.7%
[16:55:43][D][ota:353]: OTA in progress: 54.2%
[16:55:44][D][ota:353]: OTA in progress: 54.7%
[16:55:45][D][ota:353]: OTA in progress: 55.2%
[16:55:46][D][ota:353]: OTA in progress: 55.6%
[16:55:48][D][ota:353]: OTA in progress: 56.1%
[16:55:49][D][ota:353]: OTA in progress: 56.6%
[16:55:50][D][ota:353]: OTA in progress: 57.1%
[16:55:51][D][ota:353]: OTA in progress: 69.2%
[16:55:52][D][ota:353]: OTA in progress: 81.9%
[16:55:53][D][ota:353]: OTA in progress: 93.4%
[16:55:54][W][ota:415]: Error client disconnected while receiving data!
[16:55:54][W][ota:386]: Update end failed! Error: ERROR[0]: No Error

[16:55:54][D][api:073]: Disconnecting 172.16.10.55
[16:55:54][D][api:073]: Disconnecting Home Assistant 0.90.1 (172.16.10.55)
[16:55:54][D][api:546]: Client 'Home Assistant 0.90.1 (172.16.10.55)' connected successfully!
```

Despite the aforementioned errors, it looks like the OTA succeeds: everything appears in Home Assistant as it should. Note that WiFi strength isn't an issue here: the Sonoff is right next to my AP.

bachya, 2019-04-22:
@OttoWinter This is still an issue on the latest `dev` image:

```
[00:17:37][D][ota:179]: Starting OTA Update from 172.16.10.102...
[00:17:37][D][ota:353]: OTA in progress: 0.2%
[00:17:39][D][ota:353]: OTA in progress: 9.8%
[00:17:40][D][ota:353]: OTA in progress: 20.4%
[00:17:41][D][ota:353]: OTA in progress: 30.4%
[00:17:42][D][ota:353]: OTA in progress: 32.9%
[00:17:43][D][ota:353]: OTA in progress: 34.0%
[00:17:44][D][ota:353]: OTA in progress: 44.5%
[00:17:45][D][ota:353]: OTA in progress: 52.2%
[00:17:46][D][ota:353]: OTA in progress: 52.7%
[00:17:47][D][ota:353]: OTA in progress: 53.2%
[00:17:48][D][ota:353]: OTA in progress: 53.6%
[00:17:50][D][ota:353]: OTA in progress: 54.1%
[00:17:51][D][ota:353]: OTA in progress: 54.9%
[00:17:52][D][ota:353]: OTA in progress: 55.4%
[00:17:53][D][ota:353]: OTA in progress: 57.0%
[00:17:54][D][ota:353]: OTA in progress: 67.9%
[00:17:55][D][ota:353]: OTA in progress: 78.6%
[00:17:56][D][ota:353]: OTA in progress: 83.6%
[00:17:57][D][ota:353]: OTA in progress: 94.5%
[00:17:58][W][ota:415]: Error client disconnected while receiving data!
[00:17:58][W][ota:386]: Update end failed! Error: ERROR[0]: No Error
```

In this case, it appears that the firmware update never takes.

OttoWinter, 2019-04-22:
@bachya I'm not sure what it would be, from the logs it looks like the OTA itself is working at a reasonable speed (so no timeout because of too slow wifi connection).

What could be an issue is if your OS would ignore the `SNDBUF` socket argument esphome sets when uploading. To confirm, what does the progress bar on the uploader look like while doing the update? Does it immediately fill up while the ESP is still at a lower OTA percentage?

I can up the OTA timeout during upload, maybe that will help.

bachya, 2019-04-22:
@OttoWinter It looks like the progress goes to this:

```
Uploading: [============================================================] 100%
```

...in less than a second. Meanwhile, this sequence on the device:

```
[18:00:52][D][ota:083]: Starting OTA Update from 172.16.12.1...
[18:00:52][D][ota:257]: OTA in progress: 0.2%
[18:00:53][D][ota:257]: OTA in progress: 7.3%
[18:00:54][D][ota:257]: OTA in progress: 18.2%
[18:00:56][D][ota:257]: OTA in progress: 28.0%
[18:00:57][D][ota:257]: OTA in progress: 28.5%
[18:00:58][D][ota:257]: OTA in progress: 28.9%
[18:01:00][D][ota:257]: OTA in progress: 29.4%
[18:01:01][D][ota:257]: OTA in progress: 35.1%
[18:01:02][D][ota:257]: OTA in progress: 35.6%
[18:01:03][D][ota:257]: OTA in progress: 36.1%
[18:01:04][D][ota:257]: OTA in progress: 46.0%
[18:01:06][D][ota:257]: OTA in progress: 50.9%
[18:01:07][D][ota:257]: OTA in progress: 51.4%
[18:01:08][D][ota:257]: OTA in progress: 51.9%
[18:01:09][D][ota:257]: OTA in progress: 52.4%
[18:01:10][D][ota:257]: OTA in progress: 52.8%
[18:01:11][D][ota:257]: OTA in progress: 53.3%
[18:01:13][D][ota:257]: OTA in progress: 53.6%
[18:01:14][D][ota:257]: OTA in progress: 63.0%
[18:01:15][D][ota:257]: OTA in progress: 75.1%
[18:01:16][D][ota:257]: OTA in progress: 86.0%
[18:01:17][D][ota:257]: OTA in progress: 96.3%
[18:01:17][W][ota:323]: Error client disconnected while receiving data!
[18:01:17][W][ota:290]: Update end failed! Error: ERROR[0]: No Error
```

...takes around 15-20 seconds.

bachya, 2019-04-23:
@OttoWinter Looks like you pushed new code. To confirm, I'll have to manually flash this onto a Sonoff first, yes?

OttoWinter, 2019-04-24:
> ...in less than a second. Meanwhile, this sequence on the device:

Ok, that sounds like my assumption was correct then. Something in the network is buffering the request (either OS or some other hop) and ignoring `SNDBUF`.
Does someone know if it's possible for a user-application to check if the send buffer has been fully acknowledged by the remote party?

> @OttoWinter Looks like you pushed new code. To confirm, I'll have to manually flash this onto a Sonoff first, yes?

Uhm no the code change is client-side (I just increased the timeout to 15s instead of 10s, hacky but might just work in your case). This would require installing the dev version of esphome.

bachya, 2019-04-24:
The latest `dev` Docker image appears to work. 👍

alfredopironti, 2019-04-30:
Hi All,

I seem to have a related issue. Using esphome 1.12.2; flashed a Sonoff Basic upgrading from Tasmota (uploaded esphome firmware via the Tasmota web interface). I actually have about 10 Sonoff Basic devices that I ported from Tasmota and could re-flash successfully several times, but this one seems to be having an issue.

From the esphome commad line, I see
```
$ esphome garden/swimming_pool.yaml upload
INFO Reading configuration...
INFO Resolving IP address of swimming_pool.local
INFO  -> 192.168.1.52
INFO Uploading garden/swimming_pool/.pioenvs/swimming_pool/firmware.bin (332560 bytes)
ERROR Error receiving version: timed out
```

From the device logs, I see:
```
[01:22:58][D][ota:179]: Starting OTA Update from 192.168.1.54...
WARNING Disconnected from API.
INFO Connecting to swimming_pool.local:6053 (192.168.1.52)
WARNING Couldn't connect to API (Error connecting to 192.168.1.52: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to swimming_pool.local:6053 (192.168.1.52)
INFO Successfully connected to swimming_pool.local
[01:23:29][D][api:546]: Client 'Home Assistant 0.92.1 (192.168.1.111)' connected successfully!
```
During the failed update, the device also goes unavailable on HA, so I suspect it may actually be rebooting.

I don't really know where to start debugging this, as the information I have at hand is not that much. This is the configuration I'm trying to upload:
```
esphome:
  name: swimming_pool
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "<REDACTED>"
  password: "<REDACTED>"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# In this device the Sonoff button is disabled.
# We let GPIO pin 14 turn the relay on/off.
# As usual we don't specify a name, so that it won't
# show up in HA
binary_sensor:
  - platform: gpio
    id: external_button
    pin:
      number: GPIO1
      mode: INPUT
      inverted: True
    on_press:
      - switch.toggle: relay
  - platform: gpio
    name: "Swimming Pool Door"
    device_class: "door"
    pin:
      number: GPIO14
      mode: INPUT
      inverted: True
  - platform: gpio
    name: "Swimming Pool Safety Pump Power"
    device_class: "problem"
    pin:
      number: GPIO3
      mode: INPUT
      inverted: True

switch:
  - platform: gpio
    name: "Swimming Pool Pump"
    icon: "mdi:pool"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
```
The current running configuration is set to use the embedded Sonoff Basic button (GPIO0 with INPUT_PULLUP mode) for the physical switch to toggle the relay, and does not have the mdi:pool icon for the GPIO12 switch. Literally those 3 lines differ.

Any hints or ideas before I have to open the Sonoff to re-flash via UART? Thanks!

alfredopironti, 2019-05-08:
Closing the loop. I finally opened the Sonoff and re-flashed via UART. Flashing was a breeze and the device is working well now.

I just suspect the previous Tasmota firmware (memory layout?) wasn't allowing a smooth upgrade.

OttoWinter, 2019-05-09:
> I just suspect the previous Tasmota firmware (memory layout?) wasn't allowing a smooth upgrade.

Probably not. ESPHome intentionally does not use any settings storage (like tasmota etc do) as that causes a whole lot of problems usually.

Anyway, if it works now and we don't know what caused it, there's not much that can be done.

alfredopironti, 2019-05-09:
Hi Otto, thanks for the clarification. When I migrated from Tasmota to ESPHome, I used the OTA feature of Tasmota to upload the ESPHome firmware. Could it be that this placed the firmware in an unexpected memory region (due to the settings storage still being present) that worked in "most of the cases"?

Just trying to provide hints if that can help documenting the Tasmota -> ESPHome migration.

I agree there's not much debug info to work on, and a re-flash from UART fixed it; so if the above doesn't help the rest would just be speculation. Thanks for your support, and for the exciting work you're doing -- using ESPHome was eye-opening :-)

HFsi, 2019-05-09:
> Probably not. ESPHome intentionally does not use any settings storage (like tasmota etc do) as that causes a whole lot of problems usually.

@OttoWinter Don´t you even think of it as a future feature? Maybe for WiFi settings, to not have to reflash the hole device in case of a network change