esphome/issues #115: MIC sensor, status fast change

Iv711, 2019-03-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266. NodeMCU
**Affected component:**

https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**
Hi! (Im sorry for my English, this is all Google;))
I use the microphone sensor, but if the noise occurs and quickly subsides, the sensor status quickly changes off-on-off. Is it possible to make a status delay after triggering, so that the on status would remain at least 10 seconds?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
  - platform: gpio
    pin:
      number: D2
      inverted: True
    name: "Mic Kitchen sensor"
    device_class: sound
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-11:
As I wrote before, see the binary_sensor.gpio page. It contains your answer a couple of lines down. (specifically filters)

closed 2019-03-11 by OttoWinter

Iv711, 2019-03-11:
Thx. I saw the link that I did not look under it.