esphome/issues #116: INA219 reports wrong current

bboser, 2019-03-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

* Hass.io Docker on Ubuntu

**Adafruit Feather Huzzah32 (ESP32)**

**Affected component:**

INA219

**Description of problem:**

EspHome reports correct bus and shunt voltage but wrong (too small) current in home assistant. I have two INA219, both with `Rshunt = 1Ohm`. Reported results:

```
I2C address 0x40:  Vshunt = 0.06V,   Current = 0.001A
I2C address 0x41:  Vshunt = 0.02V,   Current = 0.000A
```

Since Rshunt = 1Ohm, the reported current (in Amps) should equal Vshunt (in Volts).

I also tried leaving the default for Rshunt, but the problem persists (I then do get small values for the current, but still off).

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 23
  scl: 22
  scan: True

sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 1.0 ohm
    bus_voltage:
      name: "Smoke Supply"
      unit_of_measurement: "V"
    current: 
      name: "Smoke Current"
      unit_of_measurement: "A"
    shunt_voltage:
      name: "Smoke Shunt Voltage"
    max_voltage: 15V
    max_current: 50mA
  
  - platform: ina219
    address: 0x41
    shunt_resistance: 1.0 ohm
    bus_voltage:
      name: "CPU Supply"
      unit_of_measurement: "V"
    current: 
      name: "CPU Current"
      unit_of_measurement: "A"
    shunt_voltage:
      name: "CPU Shunt Voltage"
    max_voltage: 15V
    max_current: 100mA
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

questrov, 2019-04-06:
Same here, seems to track reasonably but is 2 decimal places off; 1.5A on DVM is 0.015A HA/ESPHome logs.  Changing shunt_resistance value does not seem to have the desired effect of scaling the output...

markusschierz, 2019-04-06:
Same here. 
Tested on a bare esp32 and a wemos D1 mini. 
using the adafruit library and example in arduino ide the sensor reports correct values.

markusschierz, 2019-04-09:
A quick and dirty workaround is to use filters and multiply:
`sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "Garden 003 Current"
      filters:
        - multiply: 100`
for me that works for the moment and delivers correct values.

bboser, 2019-04-14:
Thanks for posting the work-around! 
Works perfectly.

closed 2019-04-14 by bboser