esphome/issues #117: Neopixel White LED are independent / unable to TurnOff

Bierchermuesli, 2019-03-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker:
- ottowinter/esphomeyaml:latest --> Works like a charm...
- esphome/esphome:1.11.0 --> Neopixel does not work at all
- esphome/esphome --> White LED are independent
- esphome/esphome:dev -->White LED are independent

**ESP (ESP32/ESP8266, Board/Sonoff):**
- Wemos D1 Pro
- NodeMcu V2

**Affected component:**
- Neopixel

**Description of problem:**
When turning off the light in home-assistant, RGB LED's goes of but the White LED's stays independent On on current `white_value`. Adjusting the `brightness` does only affect RGB LED's

Only way to turn off the strip is to set `white_value` to `0` and then TurnOff the light. 

Although I really liked to just have the (Warm-)White LED on - which is not possible otherwise (always dimmed by `brightness`)... 

Same configuration with older Docker image worked  - so I assume its a neopixel bug

**Problem-relevant YAML-configuration entries:**
```
light:
    - platform: neopixelbus
      pin: GPIO3
      num_leds: 61
      name: "Chochi"
      variant: SK6812
      type: GRBW
      method: ESP8266_DMA
logger:
  baud_rate: 0
  level: debug
```
also tried with `GPIO2` and `ESP8266_UART1`

Thanks!!

OttoWinter, 2019-03-12:
Yes this was intentional, before it was not possible to control only the white LED of the light strip. I guess this is a side effect now.

The issue is in HA, 0% brightness means OFF so you either get the ability to control white LED independently or having to click one extra button for turning it off.

I know it is not nice, but with HA's architecture there's nothing that can be done. See https://github.com/home-assistant/architecture/issues/123

Bierchermuesli, 2019-03-12:
nice thanks! I did some research but was not able to find this reference :-/

if its in HA, why it is working with older esphomeyaml version? some API changes?

so, can be closes (I guess...?)

OttoWinter, 2019-03-12:
Somewhere in discord, but searching there is a mess. Problem was that for RGBW lights it was impossible to control white value without turning at least one RGB channel on.

closed 2019-03-17 by OttoWinter