esphome/issues #118: ds18b20 not recognized

fragolinux, 2019-03-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
esphome 1.11.2 as addon in hass.io on ubuntu 18.04

**ESP (ESP32/ESP8266, Board/Sonoff):**
it's a witty board (it has integrated rgb led and 1 button and 1 ldr), i selected esp01_1m as board

**Affected component:**
ds18b20 dallas temperature sensor

**Description of problem:**
it cannot detect any sensor ("**[20:51:20][W][sensor.dallas:084]:   Found no sensors!**"), while on tasmota with "sensors" build it works without any issue... oh, even static ip does not work, i commented it out, on reboot it was still on previous dhcp address (different from the static one, which is out of the dhcp range)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: witty
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ....................
  password: ..................
#  manual_ip:
#    static_ip: 192.168.1.115
#    gateway: 192.168.1.1
#    subnet: 255.255.255.0

logger:
  level: VERY_VERBOSE

api:
  password: ................

ota:
  password: ................

light:
  - platform: rgb
    name: "LED RGB"
    red: output_red
    green: output_green
    blue: output_blue

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO15
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO13

web_server:
  port: 80

dallas:
  - pin: GPIO16

sensor:
  - platform: adc
    pin: A0
    name: "LDR"
    update_interval: 5s
#  - platform: dallas
#    name: "Temperatura"
#    index: 0
```

**Traceback (if applicable):**
```

```

**Additional information and things you've tried:**
as said, in tasmota it just works, on same pin (gpio16), i come from there and this is my 1st esphomized device...

fragolinux, 2019-03-11:
this are the logs, from esphome web gui:

```
INFO Reading configuration...
INFO Starting log output from witty.local using esphome API
INFO Connecting to witty.local:6053 (192.168.1.225)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.225: timed out). Re-Trying in 1 seconds
WARNING Error resolving IP address of witty.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving address with mDNS: Did not respond. Maybe the device is offline.). Re-Trying in 2 seconds
INFO Connecting to witty.local:6053 (192.168.1.225)
INFO Successfully connected to witty.local
[21:05:49][I][application:098]: esphome-core version 1.11.2 compiled on Mar 11 2019, 20:47:36
[21:05:49][C][wifi:348]: WiFi:
[21:05:49][C][wifi:244]:   SSID: [redacted]
[21:05:49][C][wifi:245]:   IP Address: 192.168.1.225
[21:05:49][C][wifi:247]:   BSSID: [redacted]
[21:05:49][C][wifi:248]:   Hostname: 'witty'
[21:05:49][C][wifi:252]:   Signal strength: -61 dB ▂▄▆█
[21:05:49][C][wifi:253]:   Channel: 8
[21:05:49][C][wifi:254]:   Subnet: 255.255.255.0
[21:05:49][C][wifi:255]:   Gateway: 192.168.1.1
[21:05:49][C][wifi:256]:   DNS1: 192.168.1.1
[21:05:49][C][wifi:257]:   DNS2: 0.0.0.0
[21:05:49][C][output.esp8266_pwm:028]: ESP8266 PWM:
[21:05:49][C][output.esp8266_pwm:029]:   Pin: GPIO15 (Mode: OUTPUT)
[21:05:49][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[21:05:49][C][output.esp8266_pwm:028]: ESP8266 PWM:
[21:05:49][C][output.esp8266_pwm:029]:   Pin: GPIO12 (Mode: OUTPUT)
[21:05:49][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[21:05:49][C][output.esp8266_pwm:028]: ESP8266 PWM:
[21:05:49][C][output.esp8266_pwm:029]:   Pin: GPIO13 (Mode: OUTPUT)
[21:05:49][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[21:05:49][C][logger:159]: Logger:
[21:05:49][C][logger:160]:   Level: VERY_VERBOSE
[21:05:49][C][logger:161]:   Log Baud Rate: 115200
[21:05:49][C][logger:162]:   Hardware UART: UART0
[21:05:49][C][light.state:266]: Light 'LED RGB'
[21:05:49][C][light.state:268]:   Default Transition Length: 1000 ms
[21:05:49][C][light.state:269]:   Gamma Correct: 2.80
[21:05:49][C][sensor.dallas:079]: DallasComponent:
[21:05:49][C][sensor.dallas:080]:   Pin: GPIO16 (Mode: OUTPUT)
[21:05:49][C][sensor.dallas:081]:   Update Interval: 60000 ms
[21:05:49][W][sensor.dallas:084]:   Found no sensors!
[21:05:49][C][sensor.adc:036]: ADC Sensor 'LDR'
[21:05:49][C][sensor.adc:036]:   Unit of Measurement: 'V'
[21:05:49][C][sensor.adc:036]:   Accuracy Decimals: 2
[21:05:49][C][sensor.adc:036]:   Icon: 'mdi:flash'
[21:05:49][V][sensor.adc:036]:   Unique ID: '5ccf7fc17bb8-adc'
[21:05:49][C][sensor.adc:041]:   Pin: GPIO17 (Mode: INPUT)
[21:05:49][C][sensor.adc:061]:   Update Interval: 5000 ms
[21:05:49][C][web_server:139]: Web Server:
[21:05:49][C][web_server:140]:   Address: witty.local:80
[21:05:49][C][api:089]: API Server:
[21:05:49][C][api:090]:   Address: witty.local:6053
[21:05:49][C][ota:129]: Over-The-Air Updates:
[21:05:49][C][ota:130]:   Address: witty.local:8266
[21:05:49][C][ota:132]:   Using Password.
[21:05:50][VV][api:501]: on_hello_request_(client_info='Home Assistant 0.89.1')
[21:05:50][V][api:504]: Hello from client: 'Home Assistant 0.89.1 (192.168.1.254)'
[21:05:50][VV][api:523]: on_connect_request_(password='Nutella@01')
[21:05:50][D][api:535]: Client 'Home Assistant 0.89.1 (192.168.1.254)' connected successfully!
[21:05:50][VV][api:570]: on_device_info_request_
[21:05:50][VV][api:593]: on_list_entities_request_
[21:05:50][VV][api:597]: on_subscribe_states_request_
[21:05:51][VV][component:134]: Running interval 'update':0 with interval=5000 last_execution=9531 (now=14538)
[21:05:51][D][sensor.adc:093]: 'LDR': Got voltage=0.05V
[21:05:51][V][sensor.sensor:020]: 'LDR': Received new state 0.046875
```

fragolinux, 2019-03-11:
this are the logs:

```
INFO Reading configuration...
INFO Starting log output from witty.local using esphome API
INFO Connecting to witty.local:6053 (192.168.1.225)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.225: timed out). Re-Trying in 1 seconds
WARNING Error resolving IP address of witty.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving address with mDNS: Did not respond. Maybe the device is offline.). Re-Trying in 2 seconds
INFO Connecting to witty.local:6053 (192.168.1.225)
INFO Successfully connected to witty.local
[21:05:49][I][application:098]: esphome-core version 1.11.2 compiled on Mar 11 2019, 20:47:36
[21:05:49][C][wifi:348]: WiFi:
[21:05:49][C][wifi:244]:   SSID: [redacted]
[21:05:49][C][wifi:245]:   IP Address: 192.168.1.225
[21:05:49][C][wifi:247]:   BSSID: [redacted]
[21:05:49][C][wifi:248]:   Hostname: 'witty'
[21:05:49][C][wifi:252]:   Signal strength: -61 dB ▂▄▆█
[21:05:49][C][wifi:253]:   Channel: 8
[21:05:49][C][wifi:254]:   Subnet: 255.255.255.0
[21:05:49][C][wifi:255]:   Gateway: 192.168.1.1
[21:05:49][C][wifi:256]:   DNS1: 192.168.1.1
[21:05:49][C][wifi:257]:   DNS2: 0.0.0.0
[21:05:49][C][output.esp8266_pwm:028]: ESP8266 PWM:
[21:05:49][C][output.esp8266_pwm:029]:   Pin: GPIO15 (Mode: OUTPUT)
[21:05:49][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[21:05:49][C][output.esp8266_pwm:028]: ESP8266 PWM:
[21:05:49][C][output.esp8266_pwm:029]:   Pin: GPIO12 (Mode: OUTPUT)
[21:05:49][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[21:05:49][C][output.esp8266_pwm:028]: ESP8266 PWM:
[21:05:49][C][output.esp8266_pwm:029]:   Pin: GPIO13 (Mode: OUTPUT)
[21:05:49][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[21:05:49][C][logger:159]: Logger:
[21:05:49][C][logger:160]:   Level: VERY_VERBOSE
[21:05:49][C][logger:161]:   Log Baud Rate: 115200
[21:05:49][C][logger:162]:   Hardware UART: UART0
[21:05:49][C][light.state:266]: Light 'LED RGB'
[21:05:49][C][light.state:268]:   Default Transition Length: 1000 ms
[21:05:49][C][light.state:269]:   Gamma Correct: 2.80
[21:05:49][C][sensor.dallas:079]: DallasComponent:
[21:05:49][C][sensor.dallas:080]:   Pin: GPIO16 (Mode: OUTPUT)
[21:05:49][C][sensor.dallas:081]:   Update Interval: 60000 ms
[21:05:49][W][sensor.dallas:084]:   Found no sensors!
[21:05:49][C][sensor.adc:036]: ADC Sensor 'LDR'
[21:05:49][C][sensor.adc:036]:   Unit of Measurement: 'V'
[21:05:49][C][sensor.adc:036]:   Accuracy Decimals: 2
[21:05:49][C][sensor.adc:036]:   Icon: 'mdi:flash'
[21:05:49][V][sensor.adc:036]:   Unique ID: '5ccf7fc17bb8-adc'
[21:05:49][C][sensor.adc:041]:   Pin: GPIO17 (Mode: INPUT)
[21:05:49][C][sensor.adc:061]:   Update Interval: 5000 ms
[21:05:49][C][web_server:139]: Web Server:
[21:05:49][C][web_server:140]:   Address: witty.local:80
[21:05:49][C][api:089]: API Server:
[21:05:49][C][api:090]:   Address: witty.local:6053
[21:05:49][C][ota:129]: Over-The-Air Updates:
[21:05:49][C][ota:130]:   Address: witty.local:8266
[21:05:49][C][ota:132]:   Using Password.
[21:05:50][VV][api:501]: on_hello_request_(client_info='Home Assistant 0.89.1')
[21:05:50][V][api:504]: Hello from client: 'Home Assistant 0.89.1 (192.168.1.254)'
[21:05:50][VV][api:523]: on_connect_request_(password='.........')
[21:05:50][D][api:535]: Client 'Home Assistant 0.89.1 (192.168.1.254)' connected successfully!
[21:05:50][VV][api:570]: on_device_info_request_
[21:05:50][VV][api:593]: on_list_entities_request_
[21:05:50][VV][api:597]: on_subscribe_states_request_
[21:05:51][VV][component:134]: Running interval 'update':0 with interval=5000 last_execution=9531 (now=14538)
[21:05:51][D][sensor.adc:093]: 'LDR': Got voltage=0.05V
[21:05:51][V][sensor.sensor:020]: 'LDR': Received new state 0.046875
```

closed 2019-03-13 by OttoWinter

fragolinux, 2019-03-13:
to get the fix do i need to switch to the beta or dev channel?

OttoWinter, 2019-03-14:
either one