esphome/issues #119: Upgraded to espHOME Connectivity issues

talk2dug, 2019-03-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I have Docker running HA with HASSIO installed. Running on a mac mini with Debian 9.
**ESP (ESP32/ESP8266, Board/Sonoff):**

I have a mix of esp-01 esp32 and esp8285
**Affected component:**

No components. 

**Description of problem:**
Before the update, everything was working as expected. After the update, I can not add them to HA through the new configure ESPHome gui. It asks for the password, I supply it, and I can watch my ping just start timing out. This happens for ALL of my esp-01 and esp8285 devices, which I have many spread through out my house, and way up in in flood lights. 

I was OK with the delay in response I was experiencing, and even had my wife not complaining. And then the new update happened. Now my devices (that have the esp01) are no longer in my system, because i can't add them via the new config. 

As soon was i send the password, the pings start timing out, and then I get the incorrect password, which i assume is just a generic error message, because the password is right and used on the esp32s, which are rocking.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphomeyaml:
  name: flood_004
  platform: ESP8266
  board: esp8285

wifi:
  ssid: 'Sorrento2'
  password: '0000011111'
  manual_ip:
  # Set this to the IP of the ESP
    static_ip: 192.168.0.250
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.0.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
mqtt:
  broker: '192.168.0.142'
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'Il2ejp22'

ota:
  password: 'Il2ejp22'
switch:
  - platform: gpio
    name: "Flood 004"
    pin: GPIO12
    id: relay
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
My first round of troubleshooting was to move my Deco M5 WAPs around, thinking maybe it was a weak connection in the mesh. 
Then i found a forum that said the latest update to the M5s caused issues with ESPs and Raspberry Pis, so i put all three of them back to factory defaults, and set it back up. Same result.

So then I said EFF it and went to walmart and bought as brand spanking new google wifi mesh. Set it up, got all my 96 devices to show up, went to setup the remaining espHome devices, and yup, you guessed it....... ping of death.....

Ping times normally just sitting there not trying to talk to it, are great, between 6 and 40 avg. 

Updating the devices is going to be a huge pain in the ass, as a few of them are litterally 50' up inside a water proof and weather proof box for my flood lights. 

So, can we not use the must supply password and go back to pure MQTT?

OttoWinter, 2019-03-12:
> [...] and then I get the incorrect password, which i assume is just a generic error message, because the password is right and used on the esp32s, which are rocking.

It is not, the invalid password message is only printed when the password is incorrect - at least from what I remember writing in the code.

> So, can we not use the must supply password and go back to pure MQTT?

What do you mean? Updating the ESP OTA without a password when a password is specified in the config? No that is not possible and would totally defeat the purpose of a password: limiting access.

talk2dug, 2019-03-13:
You misunderstood what I was referring too. The password, was never a requirements when adding devices to HA. Now I must supply the password to HA before the devices is available in HA, this is new and has caused all of my esp01 to not be useful

talk2dug, 2019-03-13:
And I am not sure how it can be an invalid passwords, where the unit drops off the network. I have also copied and pasted the config file for everyone of my units to make sure this is never an issue, Its NOT a wrong password. I do not get a bad password message for my ESP32s and its the same password.

talk2dug, 2019-03-13:
[image]
This is new. This is what's causing my units to no longer show up in HA I never had to go through this before and my units just updated to mqtt broker. Now, they do no update.

talk2dug, 2019-03-13:
[image]

And here is what happens when I supply the password and press OK.

talk2dug, 2019-03-13:
[image]
Last one and ill stop blowing up this thread. This is a good one,. ESP32 with the same password. I know for 110% its not the password. I use this password for all of my internal network units and have for years. Its also the same password on ALL of my espHOMEyaml units. So if it were a password issue, then this unit would not have been added to HA.

brandond, 2019-03-13:
Honestly it sounds like they're crashing and rebooting when you try to connect. Have you tried connecting one via UART and collecting the serial logs when you try to add it in HA?

talk2dug, 2019-03-13:
OK!!!! I have sorted out what is going on.......  As it turns out, I had a whole lot of things get updated. Not just ESPhome but also, mqtt. 

In my config it was setup for discovery enabled at the root, as well as within mqtt. For whatever reason, the built in mqtt service with the user homeasissitant was enabled. This was causing issues with the add-on broker running in anonymous. 

I ended up spinning up a mqtt broker outside of hassio. Remove discover at the root and left it on the mqtt broker settings. 

That finally brought me back to where I was prior to all of the updates. 

Now about the dropping. I have never been able to connect to the esp-01 after that are setup, I read this was normal due to some reasons I don't remember right now. Thus OTA is not available for the esp-01. This was never an issue as I decided to use all of my esp-01 for devices that are just dumb, on off nothing else, and I have a ton of them. 

The updates broke the devices from communicating with the mqtt broker that's installed from the add-on due to the ESPhome auto discovery requires the OTA password to communicate with the device, which I fully don't get why its this way, when publishing directly to MQTT broker can be secure with ssl.

Yes the ESP-01 are still crap and I can't talk to them without them dropping, but now they are publishing and subscribing to my mqtt broker as they should and use to before the CF of the year, for me at least. This has been a hair pulling nightmare.

OttoWinter, 2019-03-13:
> The updates broke the devices from communicating with the mqtt broker that's installed from the add-on due to the ESPhome auto discovery requires the OTA password to communicate with the device, [...]

1. I do not understand this sentence that well.
2. OTA does not happen over MQTT

talk2dug, 2019-03-13:
I know Ota doesn't happen over mqtt, thats not what i am discussing. I meant to say HA is requesting the API password, not the OTA password. I mixed up the terminology when typing.

Whats happening now, if you have discovery on in the configuration.yaml, when a new device is found via ESP, HA asks for the password of the device, before it will add it to the HA system. This wasn't the way before the updates. Before, a new found device would just show up under the MQTT integration window and would automatically be added to HA.

[image]

This image is what I am talking about, this was not there before the updates.

OttoWinter, 2019-03-14:
I think you're confused here.

The window you're seeing is for the "native API" - a completely different system from MQTT.

Pre-1.10, only MQTT was used to communicate with HA and devices were auto-added to HA when the ESP connected to the MQTT broker.

If you enable the native API (with `api:` in the config), there will be an additional server opened on the ESP that HA will talk directly to (with no broker in between). Additionally, this channel supports authentication.

If you put a password in the setup wizard, this is entirely expected and in fact correct.

I do not understand what the issue here is, can you clear that up?

NearlCrews, 2019-03-14:
I hit a similar issue with an ESP8266 when I had both MQTT and the API enabled in the ESPHome configuration. I don't remember which build it was though. The ESP8266 would crash and reboot if I connected to the API and also had MQTT enabled. Everything worked as expected once I disabled MQTT and stuck with the API instead, basically following [this guide](https://esphome.io/components/api.html). There's no reason to have them both turned on, so try sticking with one or the other. Personally, I ended up disabling MQTT on all of the ESPHome devices, stuck with the API, and never looked back.

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter