esphome/issues #120: Issue with Remote Transmitter and custom RCSwitch code

stich86, 2019-03-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 + FS1000A (315Mhz)
**Affected component:**

**Description of problem:**

i'm trying to use a Remote Transmitter part to open a gate. I'm successfully using rc-switch (with custom protocol) library with a simple sketch that works without problem. In my sketch i'm using code in decimal (no binary), but i've converted my dec 2 bin and still not works :(

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: testesp01
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'XXXXX'
  password: 'XXXXX'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'XXXXX'

ota:
  password: 'XXXXX'

remote_transmitter:
  pin: D1

switch:
  - platform: remote_transmitter
    name: "Cancello"
    rc_switch_raw:
      code: '00110000110011011001100011110000'
      protocol:
        pulse_length: 810
        sync: [4, 4]
        zero: [2, 2]
        one: [1, 1]
        inverted: False
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-12:
see carrier duty cyle option as the docs specify

stich86, 2019-03-12:
sorry, it was an old yaml file, i've under remote transmitter:

  carrier_duty_percent: 100%

stich86, 2019-03-12:
i've done some test with BusPirate, it looks like that RCSwitch library on this project doesn't generate correct sync bit:

This is my code sent by rc-switch on an arduino sketch (that open gate):

[image]

And this is the one generated by esphome:

[image]

using same protocol definition and code

stich86, 2019-03-12:
i'm trying also to change sync bit to different value.. but it doesn't create it :(

OttoWinter, 2019-03-13:
The sync bit is sent at the end of each transmission, not at the beginning (rc switch also does this).

What might be your issue is that while rc switch resends the packet multiple times by default, esphome does not do so, but that can be configured (see docs).

Then it would *look* like the sync bit is sent at the beginning of a transmission.

stich86, 2019-03-14:
I’m not sure. All the test I’ve done (and also the wiki stated for rc-switch) the RF is composed by:

Sync bit + Manufactur code + action code. In my case just the first two are available because it’s a gate opener.

I’ve tried what you suggest but there is not sync bit trace on the capture. Using raw component and my timings (first two number are sync bits and the the code) works.

OttoWinter, 2019-03-14:
Well the code says something else: https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L506 - the sync bit is sent after all data bits. ESPHome does the same: https://github.com/esphome/esphome-core/blob/dev/src/esphome/remote/rc_switch_protocol.cpp#L66

OttoWinter, 2019-07-11:
Closing due to inactivity.

closed 2019-07-11 by OttoWinter