esphome/issues #122: Command "python setup.py egg_info" failed with error code 1

SebDominguez, 2019-03-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS 10.13.6

I really can't see what I'm missing there.
Just trying to follow the "Getting Started" guide, the first command `pip install esphome` give me an error 
```
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-install-44NNHn/esphome/
```

**Traceback (if applicable):**

```bash
➜  ~ python --version
Python 2.7.10
➜  ~ pip install esphome
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting esphome
  Using cached https://files.pythonhosted.org/packages/84/a0/90ffeb0be888b7ec744e00ff2a3a4e5de608664ef916f76b1a321fe4b717/esphome-1.11.2.tar.gz
    Complete output from command python setup.py egg_info:
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    error in esphome setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected ',' or end-of-list in typing>=3.0.0;python_version<"3.5" at ;python_version<"3.5"

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-install-44NNHn/esphome/
➜  ~ sudo pip install esphome
Password:
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
The directory '/Users/sebastien/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/sebastien/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting esphome
  Downloading https://files.pythonhosted.org/packages/84/a0/90ffeb0be888b7ec744e00ff2a3a4e5de608664ef916f76b1a321fe4b717/esphome-1.11.2.tar.gz (427kB)
    100% |████████████████████████████████| 430kB 10.2MB/s
    Complete output from command python setup.py egg_info:
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    error in esphome setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected ',' or end-of-list in typing>=3.0.0;python_version<"3.5" at ;python_version<"3.5"

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-install-8Uvpj_/esphome/
➜  ~ pip3 install esphome
Collecting esphome
  Using cached https://files.pythonhosted.org/packages/84/a0/90ffeb0be888b7ec744e00ff2a3a4e5de608664ef916f76b1a321fe4b717/esphome-1.11.2.tar.gz
Collecting voluptuous>=0.11.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/59/95/fa6218477c6999c9b7fdfab7c12c1bd4da2d5930f5eb2b232ec74eb344e7/voluptuous-0.11.5-py2.py3-none-any.whl
Collecting pyyaml>=3.12 (from esphome)
  Using cached https://files.pythonhosted.org/packages/9e/a3/1d13970c3f36777c583f136c136f804d70f500168edc1edea6daa7200769/PyYAML-3.13.tar.gz
Collecting paho-mqtt>=1.3.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/25/63/db25e62979c2a716a74950c9ed658dce431b5cb01fde29eb6cba9489a904/paho-mqtt-1.4.0.tar.gz
Collecting colorlog>=3.1.2 (from esphome)
  Using cached https://files.pythonhosted.org/packages/68/4d/892728b0c14547224f0ac40884e722a3d00cb54e7a146aea0b3186806c9e/colorlog-4.0.2-py2.py3-none-any.whl
Collecting tornado>=5.0.0 (from esphome)
  Using cached https://files.pythonhosted.org/packages/b1/7f/8109821ff9df1bf3519169e34646705c32ac13be6a4d51a79ed57f47686e/tornado-6.0.1.tar.gz
Collecting protobuf>=3.4 (from esphome)
  Using cached https://files.pythonhosted.org/packages/fb/4f/b4ab63001f81ccc3ef754965070000f056b2c13fadf93ef46ac231e3086d/protobuf-3.7.0-cp37-cp37m-macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl
Collecting tzlocal>=1.4 (from esphome)
  Using cached https://files.pythonhosted.org/packages/cb/89/e3687d3ed99bc882793f82634e9824e62499fdfdc4b1ae39e211c5b05017/tzlocal-1.5.1.tar.gz
Collecting pyserial<4,>=3.4 (from esphome)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting ifaddr>=0.1.6 (from esphome)
  Using cached https://files.pythonhosted.org/packages/9f/54/d92bda685093ebc70e2057abfa83ef1b3fb0ae2b6357262a3e19dfe96bb8/ifaddr-0.1.6.tar.gz
Collecting platformio>=3.5.3 (from esphome)
  Using cached https://files.pythonhosted.org/packages/f1/92/202b47ed8b4e8dcef98a74426dbffa139d10d11f01a10fe6fc98728fbade/platformio-3.6.5.tar.gz
    Complete output from command python setup.py egg_info:
    PlatformIO Core v3.6.5 does not run under Python version 3.7.2 (default, Feb 12 2019, 08:16:38)
    [Clang 10.0.0 (clang-1000.11.45.5)].
    Minimum supported version is 2.7, please upgrade Python.
    Python 3 is not yet supported.

[... 20 lines cut ...]
    100% |████████████████████████████████| 491kB 2.3MB/s
Collecting protobuf>=3.4 (from esphome)
  Downloading https://files.pythonhosted.org/packages/fb/4f/b4ab63001f81ccc3ef754965070000f056b2c13fadf93ef46ac231e3086d/protobuf-3.7.0-cp37-cp37m-macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl (1.3MB)
    100% |████████████████████████████████| 1.3MB 5.5MB/s
Collecting tzlocal>=1.4 (from esphome)
  Downloading https://files.pythonhosted.org/packages/cb/89/e3687d3ed99bc882793f82634e9824e62499fdfdc4b1ae39e211c5b05017/tzlocal-1.5.1.tar.gz
Collecting pyserial<4,>=3.4 (from esphome)
  Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 2.8MB/s
Collecting ifaddr>=0.1.6 (from esphome)
  Downloading https://files.pythonhosted.org/packages/9f/54/d92bda685093ebc70e2057abfa83ef1b3fb0ae2b6357262a3e19dfe96bb8/ifaddr-0.1.6.tar.gz
Collecting platformio>=3.5.3 (from esphome)
  Downloading https://files.pythonhosted.org/packages/f1/92/202b47ed8b4e8dcef98a74426dbffa139d10d11f01a10fe6fc98728fbade/platformio-3.6.5.tar.gz (112kB)
    100% |████████████████████████████████| 112kB 15.1MB/s
    Complete output from command python setup.py egg_info:
    PlatformIO Core v3.6.5 does not run under Python version 3.7.2 (default, Feb 12 2019, 08:16:38)
    [Clang 10.0.0 (clang-1000.11.45.5)].
    Minimum supported version is 2.7, please upgrade Python.
    Python 3 is not yet supported.

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-install-18hcoxyb/platformio/
```

I do have Platformio-IDE already installed via Atom's Package Manager and I thought that this was the problem. I tried to uninstall it but that doesn't seems to change anything.

OttoWinter, 2019-03-12:
Try updating setuptools

```bash
pip install -U setuptools
```

SebDominguez, 2019-03-12:
```bash
➜  ~ pip install -U setuptools
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting setuptools
  Using cached https://files.pythonhosted.org/packages/d1/6a/4b2fcefd2ea0868810e92d519dacac1ddc64a2e53ba9e3422c3b62b378a6/setuptools-40.8.0-py2.py3-none-any.whl
Installing collected packages: setuptools
  Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
Could not install packages due to an EnvironmentError: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/__init__.py', "[Errno 1] Operation not permitted: '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/__init__.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/markers.py', "[Errno 1] Operation not permitted: '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/markers.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu', "[Errno 1] Operation not permitted: '/private/var/folders/6k/tvzjvt115k50pvl1r5gx6yrm0000gn/T/pip-uninstall-srksiu'")]

➜  ~ sudo pip install -U setuptools
Password:
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
The directory '/Users/sebastien/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/sebastien/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/d1/6a/4b2fcefd2ea0868810e92d519dacac1ddc64a2e53ba9e3422c3b62b378a6/setuptools-40.8.0-py2.py3-none-any.whl (575kB)
    100% |████████████████████████████████| 583kB 4.0MB/s
Installing collected packages: setuptools
  Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
Could not install packages due to an EnvironmentError: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/private/tmp/pip-uninstall-SUuGmZ/__init__.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-SUuGmZ/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/private/tmp/pip-uninstall-SUuGmZ/__init__.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-SUuGmZ/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/private/tmp/pip-uninstall-SUuGmZ/markers.py', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-SUuGmZ/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/private/tmp/pip-uninstall-SUuGmZ/markers.pyc', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-SUuGmZ/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/private/tmp/pip-uninstall-SUuGmZ', "[Errno 1] Operation not permitted: '/private/tmp/pip-uninstall-SUuGmZ'")]
```

OttoWinter, 2019-03-12:
If not even a setuptools install/update works then there must be something else wrong with your system.

I'm assuming you installed python via homebrew, what do you see with `brew doctor`?

SebDominguez, 2019-03-12:
Python 2.7.10 is shipped with macOS by default.

Only Python3 was installed via brew to see if using version 2.7.10 was the problem.

output of brew doctor 
```bash
➜  ~ brew doctor
Your system is ready to brew.
```

OttoWinter, 2019-03-12:
The macOS system-shipped Python has a bunch of issues with package installation - I do not know exactly what but all site-packages are in `/System` and thus write privileges often fail. Not being able to install setuptools is a clear indication this is the case. If I remember correctly some official python guides even say not to use the system-shipped one.

> Only Python3 was installed via brew to see if using version 2.7.10 was the problem.

platformio is incompatible with python 3 (esphome *is* compatible, but platformio is a requirement). I would suggest installing python 2 with homebrew.

SebDominguez, 2019-03-12:
Solved!
I've installed Python 2.7.16 with brew and the installation was successful this time.

closed 2019-03-12 by SebDominguez