esphome/issues #123: Feature Request : on_long_click automation

SebDominguez, 2019-03-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 18.10

**ESP (ESP32/ESP8266, Board/Sonoff):**

Shelly1

**Affected component:**

Binary Sensor

**Description of problem:**

My goal: make the Shelly1 board sending me click / press events via MQTT.
I wan't to be able to detect single click, longue click, double clicks.

I've somewhat achieved this using `on_press` `on_click` `on_double_click` but the problem is that if I try to fire any `on_click`events, `on_press` is also triggered.

Maybe there is a workaround using some complex filters but I'm really not that familiar with that.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO5
      inverted: false
      mode: INPUT
    id: "sw"
    on_press:
      then:
      - mqtt.publish:
          topic: some/topic
          payload: "press!"
    on_click:
      min_length: 1s
      max_length: 3s
      then:
      - mqtt.publish:
          topic: some/topic
          payload: "long click!"
    on_double_click:
      min_length: 50ms
      max_length: 350ms
      then:
      - mqtt.publish:
          topic: some/topic
          payload: "double click!"
```

It would be really nice to have an `on_long_click` automation : then I could use the `on_click event` to detect simple click instead of long one.

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-12:
That's what on_multi_press is for.

closed 2019-03-12 by OttoWinter

SebDominguez, 2019-03-12:
Well I'm still having a really hard time to prevent single click from being fired when I'm trying to do a double click.

What would be a good way to do it?

OttoWinter, 2019-03-12:
@SebDominguez Yes that's *exactly* what on_multi_click is for. on_click and so fire immediately when the time has been reached, and so are unsuitable for use with different click types.

on_multi_click solves this by making the timing entirely user-configurable. The docs page there even has an example for how to distinguish between short/long/double click

SebDominguez, 2019-03-12:
Sorry I missed that one!