esphome/issues #124: Rotary encoder component broken in dev branch

puuu, 2019-03-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip/git

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

rotary encoder

**Description of problem:**
Using the dev branch, the rotary_encoder do not change its state when rotating. Version 1.11.2 of esphome-core works fine.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: rot_test
  platform: ESP8266
  board: d1_mini
  esphome_core_version:
    #tag: v1.11.2 # ok
    #commit: 6088a6d2e256511067d7b6da19a3638b74cf42d7 # ok
    #commit: acccea739754fc12689a880b1e5ef7ae8eed318a # comiler error
    #commit: e691aa9ae1b54fa1748fe8a0f0cf2b69eae876ea # comiler error
    tag: dev # broken

logger:

sensor:
  - platform: rotary_encoder
    name: rot_left
    id: rot_left
    pin_a:
      number: D2
      mode: INPUT_PULLUP
    pin_b:
      number: D1
      mode: INPUT_PULLUP
    on_value:
      then:    
        - lambda: |-
            ESP_LOGD("Rot", "Got: %f", x);
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

HA-TB303, 2019-03-13:
Yup, same for me. This commit seems to be the culprit:
https://github.com/esphome/esphome/commit/7c65d4497602cf339233fe451cb055d4fea329c8#commitcomment-32688860

OttoWinter, 2019-03-13:
I do not think it was that commit, but https://github.com/esphome/esphome-core/commit/e691aa9ae1b54fa1748fe8a0f0cf2b69eae876ea - can you try with latest dev again, there is a small fix.

HA-TB303, 2019-03-13:
I rebuild, restarted, clean the build and re-uploaded to the esp, but the encoder does not work on rotation.

HA-TB303, 2019-03-15:
Anything more we can do to log or debug?

HA-TB303, 2019-03-17:
Still not working in esphome-core version 1.13.0-dev compiled on Mar 17 2019, 11:06:12

puuu, 2019-03-17:
I can confirm that commit esphome/esphome-core@fe04ee5332e0b35e1591b4df238cf9fb14197769 does not fix this issue yet. I suppose the problem is the LUT, because printing `store_.state` from `loop()` will show different states depending on the rotation, but none of them is longer than 5 bit.

closed 2019-03-17 by OttoWinter

puuu, 2019-03-18:
Thanks, now it is working!