esphome/issues #125: ESP32 not setting the hostname correctly via DHCP

NearlCrews, 2019-03-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

I'm assuming wifi: https://esphome.io/components/wifi.html

**Description of problem:**
The ESP32 does not seem to set a hostname although it grabs a DHCP address. This problem only seems to occur with the ESP32. My ESP8266 modules set the hostname correctly. I've tried various combinations of dev and stable for both arduino and esphome without any luck. This looks identical to Issue #64 which was closed.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: nodemcu_plantmon
  platform: ESP32
  board: nodemcu-32s
  esphome_core_version: dev
  arduino_version: dev

wifi:
  ssid: 'SSID'
  password: 'password'
  domain: '.domain.local'

api:

# Enable logging
logger:

ota:

esp32_ble_tracker:

sensor:
  - platform: wifi_signal
    name: "NodeMCU PlantMon WiFi"

binary_sensor:
  - platform: status
    name: "NodeMCU PlantMon Status"

time:
  - platform: sntp
    id: sntp_time
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org

switch:
  - platform: restart
    name: "NodeMCU PlantMon Restart"
```

**Traceback (if applicable):**

```
[12:08:05]▒[I][logger:119]: Log initialized
[12:08:05][C][ota:479]: There have been 1 suspected unsuccessful boot attempts.
[12:08:05][I][application:055]: Running through setup()...
[12:08:05][D][binary_sensor:037]: 'NodeMCU PlantMon Status': Sending state OFF
[12:08:05][D][esp32_ble_tracker:165]: Starting scan...
[12:08:05][C][wifi:027]: Setting up WiFi...
[12:08:06][D][wifi:246]: Starting scan...
[12:08:06][D][esp32_ble_tracker:431]: Found device 04:21:4C:03:32:9A RSSI=-81
[12:08:06][D][esp32_ble_tracker:452]:   Address Type: PUBLIC
[12:08:07][D][wifi:261]: Found networks:
[12:08:07][I][wifi:295]: - 'SSID' (78:8A:20:54:0A:D6) ▂▄▆█
[12:08:07][D][wifi:296]:     Channel: 1
[12:08:07][D][wifi:297]:     RSSI: -53 dB
[12:08:07][I][wifi:295]: - 'SSID' (78:8A:20:51:E3:69) ▂▄▆█
[12:08:07][D][wifi:296]:     Channel: 6
[12:08:07][D][wifi:297]:     RSSI: -77 dB
[12:08:07][D][wifi:299]: - '' (7A:8A:20:54:0A:D6) ▂▄▆█
[12:08:07][D][wifi:299]: - '' (7A:8A:20:51:E3:69) ▂▄▆█
[12:08:07][D][wifi:299]: - 'other-ssid' (C0:A0:0D:B4:49:AD) ▂▄▆█
[12:08:07][D][wifi:299]: - '' (E2:A0:0D:B4:49:AD) ▂▄▆█
[12:08:07][D][wifi:299]: - '' (02:A0:0D:B4:49:AD) ▂▄▆█
[12:08:07][D][wifi:299]: - 'another-ssid' (D2:A0:0D:B4:49:AD) ▂▄▆█
[12:08:07][I][wifi:167]: WiFi Connecting to 'SSID'...
[12:08:08][I][wifi:340]: WiFi connected!
[12:08:08][C][wifi:228]:   SSID: 'SSID'
[12:08:08][C][wifi:229]:   IP Address: 10.10.10.10
[12:08:08][C][wifi:231]:   BSSID: 78:8A:20:54:0A:D6
[12:08:08][C][wifi:232]:   Hostname: 'nodemcu_plantmon'
[12:08:08][C][wifi:236]:   Signal strength: -47 dB ▂▄▆█
[12:08:08][C][wifi:237]:   Channel: 1
[12:08:08][C][wifi:238]:   Subnet: 255.255.254.0
[12:08:08][C][wifi:239]:   Gateway: 10.10.10.1
[12:08:08][C][wifi:240]:   DNS1: 10.10.10.1
[12:08:08][C][wifi:241]:   DNS2: 0.0.0.0
[12:08:09][D][binary_sensor:037]: 'NodeMCU PlantMon Status': Sending state ON
[12:08:09][C][time.sntp:029]: Setting up SNTP...
[12:08:09][C][api:023]: Setting up Home Assistant API server...
[12:08:09][C][ota:127]: Over-The-Air Updates:
[12:08:09][C][ota:128]:   Address: nodemcu_plantmon.domain.local:3232
[12:08:09][I][application:089]: setup() finished successfully!
[12:08:09][I][application:097]: esphome-core version 1.12.0-dev compiled on Mar 13 2019, 11:59:26
[12:08:09][C][wifi:332]: WiFi:
[12:08:09][C][wifi:228]:   SSID: 'SSID'
[12:08:09][C][wifi:229]:   IP Address: 10.78.51.83
[12:08:09][C][wifi:231]:   BSSID: 78:8A:20:54:0A:D6
[12:08:09][C][wifi:232]:   Hostname: 'nodemcu_plantmon'
[12:08:09][C][wifi:236]:   Signal strength: -44 dB ▂▄▆█
[12:08:09][C][wifi:237]:   Channel: 1
[12:08:09][C][wifi:238]:   Subnet: 255.255.254.0
[12:08:09][C][wifi:239]:   Gateway: 10.78.50.1
[12:08:09][C][wifi:240]:   DNS1: 10.78.50.1
[12:08:09][C][wifi:241]:   DNS2: 0.0.0.0
[12:08:09][C][binary_sensor.status:035]: Status Binary Sensor 'NodeMCU PlantMon Status'
[12:08:09][C][binary_sensor.status:035]:   Device Class: 'connectivity'
[12:08:09][C][logger:142]: Logger:
[12:08:09][C][logger:143]:   Level: DEBUG
[12:08:09][C][logger:144]:   Log Baud Rate: 115200
[12:08:09][C][logger:145]:   Hardware UART: UART0
[12:08:09][C][esp32_ble_tracker:703]: BLE Tracker:
[12:08:09][C][esp32_ble_tracker:704]:   Scan Interval: 300 s
[12:08:09][C][switch.restart:028]: Restart Switch 'NodeMCU PlantMon Restart'
[12:08:09][C][switch.restart:028]:   Icon: 'mdi:restart'
[12:08:09][C][sensor.wifi_signal:027]: WiFi Signal 'NodeMCU PlantMon WiFi'
[12:08:09][C][sensor.wifi_signal:027]:   Unit of Measurement: 'dB'
[12:08:09][C][sensor.wifi_signal:027]:   Accuracy Decimals: 0
[12:08:09][C][sensor.wifi_signal:027]:   Icon: 'mdi:wifi'
[12:08:09][C][time.sntp:055]: SNTP Time:
[12:08:09][C][time.sntp:056]:   Server 1: '0.pool.ntp.org'
[12:08:09][C][time.sntp:057]:   Server 2: '1.pool.ntp.org'
[12:08:09][C][time.sntp:058]:   Server 3: '2.pool.ntp.org'
[12:08:09][C][time.sntp:059]:   Timezone: 'EST5EDT4,M3.2.0/2,M11.1.0/2'
[12:08:09][C][api:099]: API Server:
[12:08:09][C][api:100]:   Address: nodemcu_plantmon.domain.local:6053
[12:08:09][C][ota:127]: Over-The-Air Updates:
[12:08:09][C][ota:128]:   Address: nodemcu_plantmon.domain.local:3232
```

**Additional information and things you've tried:**
I've tried various combinations of dev and stable for both arduino and esphome without any luck. The ESP8266 modules do not have this issue. This looks identical to [issue 62](Also, from DHCP I'm not seeing a hostname:

```DHCPREQUEST for 10.10.10.10 from 44:19:b6:4a:67:98 via mvneta1```

From an ESP8266:

```DHCPREQUEST for 10.10.10.11 from dc:4f:22:10:8c:ec (nodemcu_sump) via mvneta1```

OttoWinter, 2019-03-13:
It doesn't sound identical because before the hostname was `espressif`, not an empty string. What could be an issue is the hostname itself: I don't remember the exact requirements for hostnames, but I think there was a length limit. Maybe this is the case here.

NearlCrews, 2019-03-13:
I don't believe that's it. I've got an ESP8266 that has a 21 character name without any issues. The one above is shorter than that. Oddly enough, the first time it booted it gave the hostname successfully. Subsequent DHCP renews were lacking the hostname and it happened after it'd been up for about 12 hours. The expressif name rings a bell though: it switched from the hostname, to "expressif", to no name at all. 

To test this further and remove any chance of a funky MAC association, I just unplugged the ESP32, plugged in a brand new one, and reflashed it with the exact same yaml:

`DHCPREQUEST for 10.10.10.13 (10.10.10.1) from 24:0a:c4:1c:86:cc (espressif) via mvneta1`

It assigned the `espressif` name even though I've got the other hostname in the config. Let me know if there's anything you'd like me to test. With the latter, it looks pretty close to the previous issue. Also tried commenting out `esp32_ble_tracker` without any luck.

NearlCrews, 2019-03-14:
Alright, one more update: it seemed to have registered the hostname in the last few hours. It definitely isn't doing it on the initial flash though. I flashed it again and the name showed up in the DHCP request correctly. Could this be some odd timing issue?

OttoWinter, 2019-03-14:
Could be, but in that case it would definitely be an upstream issue in esp-idf. ESPHome just tells the SDK that the hostname is X, and esp-idf handles all the internals (which are also closed source so no way we can fix it).

NearlCrews, 2019-03-17:
Closing this since it seems to be a problem upstream.

closed 2019-03-17 by NearlCrews

pszafer, 2019-04-29:
@OttoWinter maybe this might help you somehow - https://github.com/rjwats/esp8266-react/pull/17
FYI, I have same problem with ESP32.

OttoWinter, 2019-05-01:
@pszafer Hmm that might actually explain this behavior. On the first connect attempt, the hostname is correctly set after the DHCP config call. However after that they are no longer set. Not sure if this is really the case but setting the hostname again after each config call can't hurt either. Adding it to the code.

cpw, 2019-05-02:
Just found this issue with my brand new esp32 - it's not requesting a DHCP hostname correctly. It looks like it only works for the initial offer and then never asks for the hostname again.