esphome/issues #126: Relay switch off/on when it lose wifi connection

ToRvaLDz, 2019-03-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff T2/T3 US 

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
The device is used to power on the lights and it has poor wifi signal, when the wifi connection is lost, the device wait some minutes, then the light blink, I can hear the relay switchin off/on.
I will solve the wifi problem asap, but this could be a big issue if my router die.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: stanza2 
  platform: ESP8266
  board: esp01_1m

wifi:
  use_address: 192.168.27.78
  ap:
    ssid: stanza2
  ssid: xxxxxx
  password: xxxxxx

ota:
  safe_mode: True
  password: xxxxxx

api:
  password: xxxxxx

logger: 

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    internal: True
    name: "Touch1"
    on_press:
      - switch.toggle: Light1 
  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    internal: True
    name: "Touch2"
    on_press: 
      - switch.toggle: Light2 
  - platform: gpio
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    internal: True
    name: "Touch3"
    on_press: 
      - switch.toggle: Light3       
  - platform: status
    name: "Stanza2"

output:
  - platform: esp8266_pwm
    id: output1
    pin: GPIO12
  - platform: esp8266_pwm
    id: output2
    pin: GPIO5
  - platform: esp8266_pwm
    id: output3
    pin: GPIO4    

switch:
  - platform: output
    name: "Light1"
    id: Light1
    output: output1
  - platform: output
    name: "Light2"
    id: Light2
    output: output2
  - platform: output
    name: "Light3"
    id: Light3
    output: output3

sensor:
  - platform: wifi_signal
    name: "WiFi Stanza2"
    update_interval: 60s    

```

**Traceback (if applicable):**

```
INFO Successfully connected to 192.168.27.78
[18:40:08][I][application:098]: esphome-core version 1.11.2 compiled on Mar 13 2019, 17:01:11
[18:40:08][C][wifi:348]: WiFi:
[18:40:08][C][wifi:244]:   SSID: 'XXXXX'
[18:40:08][C][wifi:245]:   IP Address: 192.168.27.78
[18:40:08][C][wifi:247]:   BSSID: 28:24:FF:73:3E:6B
[18:40:08][C][wifi:248]:   Hostname: 'stanza2'
[18:40:08][C][wifi:252]:   Signal strength: -68 dB ▂▄▆█
[18:40:08][C][wifi:253]:   Channel: 6
[18:40:08][C][wifi:254]:   Subnet: 255.255.255.0
[18:40:08][C][wifi:255]:   Gateway: 192.168.27.254
[18:40:08][C][wifi:256]:   DNS1: 192.168.27.254
[18:40:08][C][wifi:257]:   DNS2: 0.0.0.0
[18:40:08][C][binary_sensor.status:047]: Status Binary Sensor 'Stanza2'
[18:40:08][C][binary_sensor.status:047]:   Device Class: 'connectivity'
[18:40:08][C][binary_sensor.gpio:022]: GPIO Binary Sensor 'Touch1'
[18:40:08][C][binary_sensor.gpio:023]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08][C][output.esp8266_pwm:029]:   Pin: GPIO12 (Mode: OUTPUT)
[18:40:08][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[18:40:08][C][output.esp8266_pwm:028]: ESP8266 PWM:
[18:40:08][C][output.esp8266_pwm:029]:   Pin: GPIO5 (Mode: OUTPUT)
[18:40:08][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[18:40:08][C][output.esp8266_pwm:028]: ESP8266 PWM:
[18:40:08][C][output.esp8266_pwm:029]:   Pin: GPIO4 (Mode: OUTPUT)
[18:40:08][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[18:40:08][C][switch.output:041]: Output Switch 'Light1'
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:40:08][C][sensor.wifi_signal:042]: WiFi Signal 'WiFi Stanza2'
[18:40:08][C][sensor.wifi_signal:042]:   Unit of Measurement: 'dB'
[18:40:08][C][sensor.wifi_signal:042]:   Accuracy Decimals: 0
[18:40:08][C][sensor.wifi_signal:042]:   Icon: 'mdi:wifi'
[18:40:08][C][api:089]: API Server:
[18:40:08][C][api:090]:   Address: stanza2.local:6053
[18:40:08][C][ota:129]: Over-The-Air Updates:
[18:40:08][C][ota:130]:   Address: stanza2.local:8266
[18:40:08][C][ota:132]:   Using Password.
WARNING Disconnected from API.
INFO Connecting to 192.168.27.78:6053 (192.168.27.78)
INFO Successfully connected to 192.168.27.78

```

**Additional information and things you've tried:**
I tried to setup it as both light and switch because the switch has the restore_mode configuration
I tried to set reboot_timeout 0s but I lost the wifi connection to the switch until power off/on
I tried fast_connect: True
I tried different arduino core version

Cannot debug by usb while connected to 220v and out of the wall box the wifi signal is strong.

Tnx

brandond, 2019-03-13:
API, MQTT, and WiFi each have their own reboot_timeout. You can set them to 0 to disable, but as you noticed, sometimes a reboot is necessary to get the device to reconnect to wifi. I don't think there's a perfect way to get the GPIOs to avoid momentarily toggling as the device reboots; they all have a default state that the chip resets them to when it reboots.

OttoWinter, 2019-03-14:
Exactly, the GPIOs toggling on reboot is not fixable - before esphome gains control of the pins on reset they can do all kinds of things, some pins even produce a 1kHz signal while the bootloader is loading.

So there's no way to fix that directly. WiFi reconnecting would be another issue (and should be discussed in the existing issues that are open).

closed 2019-03-14 by OttoWinter