esphome/issues #127: ai-tinker esp32-cam onboard camera ov2614 not work

hellerxx, 2019-03-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
The board pin connect in:
https://github.com/donny681/ESP32_CAMERA_QR

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: door
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "ssid"
  password: "pw"

esp32_camera:
  name: My Camera
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  resolution: 640x480
  jpeg_quality: 10
  test_pattern: true

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-14:
The camera is completely handled by the esp32-camera library, so any issue would be upstream there.

The issue must be somewhere in the i2c or powerdown related pins - looking at [this example](https://github.com/espressif/arduino-esp32/blob/fa61b3bffedc0c5b65de69c1a550c7c07d437faa/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino#L57) the power down pin does exist at pin GPIO32. Can you try enabling that?

hellerxx, 2019-03-14:
yap! working!! lose the powerdown pin  in .yaml. now, how to use the camera in webstream or homeassistant?  the component example not stated
and the camera working in 800x600, but 1024x768 or higher, the board run-time error.

[00:04:27]e���0x���98C�[I][logger:119]: Log initialized
[00:04:27][C][ota:479]: There have been 1 suspected unsuccessful boot attempts.
[00:04:27][I][application:055]: Running through setup()...
[00:04:27][E][camera.c:233] camera_fb_init(): Allocating 153 KB frame buffer Failed
[00:04:27][E][camera.c:1007] camera_init(): Failed to allocate frame buffer
[00:04:27]E (182) gpio: gpio_isr_handler_remove(380): GPIO isr service is not installed, call gpio_install_isr_service() first
[00:04:27][E][camera.c:1104] esp_camera_init(): Camera init failed with error 0x101
[00:04:27][E][esp32_camera:018]: esp_camera_init failed: ESP_ERR_NO_MEM
[00:04:27][E][component:151]: Component was marked as failed.

OttoWinter, 2019-03-16:
> how to use the camera in webstream or homeassistant?

for that you need to use a dev homeassistant instance for now. The camera integration in HA has not been merged yet. You need to install HA from this PR: https://github.com/home-assistant/home-assistant/pull/22107/files

> and the camera working in 800x600, but 1024x768 or higher, the board run-time error.

The error you're seeing means your module does not have enough RAM for that resolution. For higher resolutions you need boards with external PSRAM.

OttoWinter, 2019-03-17:
I will close this issue because the camera is working. If something else turns up re-open it.

closed 2019-03-17 by OttoWinter

hellerxx, 2019-03-17:
The borad has  PSRAM is not running，In Borad PSRAM chip is IPS6404LSQ
[C][esp32_camera:044]: ESP32 Camera:
[C][esp32_camera:045]:   Name: My Camera
[C][esp32_camera:046]:   Board Has PSRAM: NO

hellerxx, 2019-03-17:
@OttoWinter

hellerxx, 2019-03-17:
ok,thx

nicolas-fricke, 2019-04-24:
I am currently also trying to set up an ESP32cam ([this model](https://www.aliexpress.com/item/ESP32-CAM-WiFi-WiFi-Module-ESP32-serial-to-WiFi-ESP32-CAM-Development-Board-5V-Bluetooth-with/32981449079.html?spm=a2g0s.9042311.0.0.3ca64c4de8aBzo)) and am getting the same issue. I have tried to add the `GPIO32` as a `power_down_pin`, but without avail.

@hellerxx could you share your final, now working YAML configuration?

This is my current config:
```yaml
esphome:
  name: livingroom_cam
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: My WiFi
  password: secret

# Enable Home Assistant API
api:
  password: secret

esp32_camera:
  name: My Camera
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  resolution: 640x480
  jpeg_quality: 10
  test_pattern: true
```

And while booting, I'm also getting the very same, descriptive `Setup Failed: ERROR`:
```
[...]
[09:54:31][C][esp32_camera:044]: ESP32 Camera:
[09:54:31][C][esp32_camera:045]:   Name: My Camera
[09:54:31][C][esp32_camera:046]:   Board Has PSRAM: YES
[09:54:31][C][esp32_camera:048]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[09:54:31][C][esp32_camera:049]:   VSYNC Pin: 25
[09:54:31][C][esp32_camera:050]:   HREF Pin: 23
[09:54:31][C][esp32_camera:051]:   Pixel Clock Pin: 22
[09:54:31][C][esp32_camera:052]:   External Clock: Pin:0 Frequency:20000000
[09:54:31][C][esp32_camera:053]:   I2C Pins: SDA:26 SCL:27
[09:54:31][C][esp32_camera:054]:   Reset Pin: -1
[09:54:31][C][esp32_camera:075]:   Resolution: 640x480 (VGA)
[09:54:31][E][esp32_camera:092]:   Setup Failed: ERROR
```

Interestingly, in the debug output it doesn't even mention the powerdown pin that I set up.. 👀 

Does anyone see what I might be doing wrong here?