esphome/issues #128: Homekit support

aardjoun, 2019-03-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266 sonoff basic
**Affected component:**

Homekit support

**Description of problem:**
The esphome based esp isn't discovered in Apple Home App.
Home Assistant Homekit component work with HA switches 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: lampe_cuisine
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'ssid'
  password: 'password'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'password'

ota:
  password: 'password'

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Lampe cuisine"
    on_press:
      - switch.toggle: fakebutton

switch:
  - platform: template
    name: "Lampe cuisine"
    optimistic: true
    id: fakebutton
    turn_on_action:
    - switch.turn_on: relay
    # - light.turn_off: led
    turn_off_action:
    - switch.turn_off: relay
    # - light.turn_on: led
  - platform: gpio
    id: relay
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

# light:
#   - platform: monochromatic
#     name: "Lampe led cuisine"
#     output: basic_green_led
#     id: led

sensor:
  - platform: wifi_signal
    name: "Lampe cuisine WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "Lampe cuisine Uptime"
  
text_sensor:
  - platform: version
    name: "Lampe cuisine ESPHome Version"
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-14:
The description is quite sparse. But esphome does not have direct integration into homekit (nor is that planned for the near future). Those things are supposed to be done by a central home automation system like Home Assistant.

This isn't an issue, it's a feature request.

closed 2019-03-14 by OttoWinter

aardjoun, 2019-03-14:
those things are already done by home assistant and works great with tasmota's based esp. How can i make a feature request ?

SebDominguez, 2019-03-14:
I think OP was referring to Homeassistant HomeKit component and the fact that the ESPs running ESPhome weren’t added automatically to the HomeKit bridge like any other switches configured in Homeassistant.

OttoWinter, 2019-03-14:
Then that would be a Home Assistant problem. I don't know HomeKit but probably you just need to restart some stuff. ESPHome registers all entities correctly in HA like any other seitch platform, so that should not be it