esphome/issues #129: ESPHome on aarch64/arm64v8

sachilles, 2019-03-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

* ArchLinuxArm 64 bit on a Raspberry Pi 3
* Docker version 18.09.3-ce, build 774a1f4eee
* Docker image esphome/esphome-aarch64 in Version 1.11.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

* Wemos D1 Mini

**Affected component:**

Probably none.

**Description of problem:**

Using the docker image (as stated above) no firmware can be compiled. It seems that the toolchain has some issues. Two sections below you'll find the output of the (web-) console by trying a clean build.

**Problem-relevant YAML-configuration entries:**

Probably none.

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/wemos_test
Processing wemos_test (platform: espressif8266@1.8.0; framework: arduino; board: d1_mini)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing v1.11.2
Downloading...
Unpacking  [####################################]  100%          
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.11.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.1.3
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/wemos_test/src/main.cpp.o
Generating LD script .pioenvs/wemos_test/ld/eagle.app.v6.common.ld
Compiling .pioenvs/wemos_test/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/wemos_test/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/src/main.cpp.o] Error 2
sh: 1: xtensa-lx106-elf-gcc: Exec format error
*** [.pioenvs/wemos_test/ld/eagle.app.v6.common.ld] Error 2
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o] Error 2
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o] Error 2
========================== [ERROR] Took 26.75 seconds ==========================
```

**Additional information and things you've tried:**

A clean build/compilation on AMD64 plattform and upload via USB works. After that, the Wemos D1 mini is recognized by the aarch64/arm64v8 docker image.

OttoWinter, 2019-03-15:
That is an issue in the platformio toolchains. I would suggest using the armhf images until that is fixed upstream (armhf is most often compatible on aarch64 systems).

deece, 2019-04-23:
Has this been reported upstream yet?

OttoWinter, 2019-04-24:
> Has this been reported upstream to platformio yet?

I think so, but I can't find the issue right now. I'm actually sure if this actually an issue anymore or fixed already, someone could check with the esphome aarch64 docker images, if those work then aarch64 can be enabled in the hass.io addon config.

valvry, 2019-05-24:
> I think so, but I can't find the issue right now. I'm actually sure if this actually an issue anymore or fixed already, someone could check with the esphome aarch64 docker images, if those work then aarch64 can be enabled in the hass.io addon config.

hi how to enable esphome aarch64 in hass.io addon config?
i am install hass.io on tv box, but esphome in addon store not avaible in my sytem.
only dev version can be install on my system, but cant start the sphome.

yesterday I try to edit config esphome dev version like this:
"name": "ESPHome (dev)",
  "version": "dev",
  "slug": "esphome-dev",
  "description": "Development Version! Manage and program ESP8266/ESP32 microcontrollers through YAML configuration files",
  "options": {
    "esphome_version": "dev"
  }

to this:
"name": "ESPHome (dev)",
  "version": "1.12.2",
  "slug": "esphome",
  "description": "Development Version! Manage and program ESP8266/ESP32 microcontrollers through YAML configuration files",
  "options": {
    "ssl": false,
    "certfile": "fullchain.pem",
    "keyfile": "privkey.pem",
    "port": 6052
  }

and edit docker file :
ARG BUILD_FROM=esphome/esphome-hassio-base-amd64:1.5.1

to:
ARG BUILD_FROM=esphome/esphome-hassio-base-aarch64:1.5.1

and copy to local addon hass.io.
they all work fine i was try to compile and upload to device.

my question, can you enable esphome stable version to aarch64 ?

OttoWinter, 2019-05-24:
> only dev version can be install on my system, but cant start the sphome.

Then that looks like aarch64 does not work - the dev version uses the aarch64 base image (and is the only one enabled so that users can test if aarch64 works yet).

Please add why dev does not work (logs) and how you think you got the dev image working (the BUILD_FROM line is overriden by Hass.io - changing it should not do anything as long as you're using the Hass.io interface).

valvry, 2019-05-24:
> Please add why dev does not work (logs) and how you think you got the dev image working (the BUILD_FROM line is overriden by Hass.io - changing it should not do anything as long as you're using the Hass.io interface).

this log when iam start esphome dev version
[image]

and what i mean, i was try again copy esphome addons dev version to local addon and only edit config.json file, and its work fine whitout edit BUILD_FROM

[image]

[image]

OttoWinter, 2019-05-24:
Ok that's another issue - for some reason pip is failing to uninstall the old esphome-dev version from the image. (https://github.com/openwrt/packages/issues/1709 seems related).

What happens when you use the addon from the repo and remove the `esphome_version` line from the add-on config (in Hass.io view)?

valvry, 2019-05-24:
> Ok that's another issue - for some reason pip is failing to uninstall the old esphome-dev version from the image. ([openwrt/packages#1709](https://github.com/openwrt/packages/issues/1709) seems related).
> 
> What happens when you use the addon from the repo and remove the `esphome_version` line from the add-on config (in Hass.io view)?

thakyou for your respone,
i was try use addons from the repo and remove esphome_version in hass.io view error same as above

OttoWinter, 2019-05-28:
Thanks for trying that @valvry - it is much appreciated!

Interesting that it's still showing the same error - if the option is not specified it *should* not even execute that code ever. Or I don't understand the Hass.io options system correctly.

Anyway, from a closer look that appears to be a separate issue with the python shipped with the arm64v8 image by ubuntu.

I've pushed a beta version of ESPHome yesterday, and aarch64 is enabled there. I've tried to set up my RPi to test the change, but so far I've been struggling with setting up the network, so I can't really check if it actually works now. I think in a couple of days I can get that test setup to work (need some components)

If someone can check if the beta image works now that would also be very much appreciated (and if not, error log; I'm pretty sure this time it cannot be the same stacktrace again, as that does not even in the image)

valvry, 2019-05-28:
> Thanks for trying that @valvry - it is much appreciated!
>
> Interesting that it's still showing the same error - if the option is not specified it _should_ not even execute that code ever. Or I don't [...]

Thank you @OttoWinter 
i was trying beta version and its work 👍 :)

OttoWinter, 2019-05-28:
That's great!

sachilles, 2019-05-28:
Hi,

I tried to compile a little test firmware using version 1.13.0b3 and I obtained the same error output.

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/wemos_test
Processing wemos_test (framework: arduino; platform: espressif8266@1.8.0; board: d1_mini)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.2.0
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/wemos_test/src/esphome/components/api/api_message.cpp.o
Compiling .pioenvs/wemos_test/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/wemos_test/src/esphome/components/api/basic_messages.cpp.o
Compiling .pioenvs/wemos_test/src/esphome/components/api/command_messages.cpp.o
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/src/esphome/components/api/api_message.cpp.o] Error 2
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/src/esphome/components/api/command_messages.cpp.o] Error 2
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/src/esphome/components/api/api_server.cpp.o] Error 2
sh: 1: xtensa-lx106-elf-g++: Exec format error
*** [.pioenvs/wemos_test/src/esphome/components/api/basic_messages.cpp.o] Error 2
========================== [ERROR] Took 11.98 seconds ==========================
```

I'm using the image `esphome/esphome-aarch64:1.13.0b3`. What is the difference between this image and the hassio ones?

Thanks in advance!

OttoWinter, 2019-05-29:
@sachilles There shouldn't be any major differences - both use the same base ubuntu image and install python from there. So that shouldn't be the issue.

What could be an issue is the system: I'm not sure if your archlinux is using some different arm64v8 mode than HA or using some other base settings.

Was able to test on a RPi 3b+ Hass.io setup today - works great!

OttoWinter, 2019-05-31:
Update: turns out only the ESP8266 works with aarch64, for ESP32 the platformio package is broken - my first guess is a broken symlink or so.

deylin, 2019-06-02:
With the latest updates of the esphome addon, I am not able to compile (as I could in previous version):
[image]

OvCa77, 2019-06-06:
From issue #411 ESP32 works, with dev version. Not ESP8266

OttoWinter, 2019-06-06:
@OvCa77 Thanks for reporting, I will do the same thing (build a native toolchain and patch platformio in docker image) for ESP8266 👍

OttoWinter, 2019-07-11:
Has been fixed in 1.13 builds

closed 2019-07-11 by OttoWinter

OvCa77, 2019-07-11:
It seems that this issue is still not solved, is there out new version? I am running 1.13.6

OttoWinter, 2019-07-11:
@OvCa77 Ok, it should be fixed in 1.13.6 - I manually built the entire esp32/esp8266 on a native aarch64 system. And at least on my RPi 3b+ it's all working fine now. If you still see the issue, please create a new bug report (this one here is getting too messy) with logs etc attached.