esphome/issues #130: Cannot access node logs after reboot

akasma74, 2019-03-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, WeMos D1 mini

**Affected component:**
logs

**Description of problem:**
My node was working fine for weeks, but today I powered off access point that it was configured to connect to with static ip.
When I powered the AP again in 30 minutes, the node entities remained Unavailable in HA and I couldn't see its logs:

> esphome bathroom_extractor_fan_sensor.yaml logs
> INFO Reading configuration...
> INFO Starting log output from 192.168.0.53 using esphome API
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: timed out). Re-Trying in 1 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: timed out). Re-Trying in 2 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: [Errno 64] Host is down). Re-Trying in 4 seconds

I power cycled the node, and it came back to HA - I can see data coming from entities, its integration components
[image]

However, I'm still unable to see logs

> INFO Reading configuration...
> INFO Starting log output from 192.168.0.53 using esphome API
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: timed out). Re-Trying in 1 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: [Errno 64] Host is down). Re-Trying in 2 seconds

If I delete the node from integrations and try to add it, it fails
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: "bathroom_extractor_fan_sensor"
  platform: ESP8266
  board: d1_mini

wifi:
  fast_connect: true
  ssid: 'xxx'
  password: 'xxx'
  manual_ip:
    static_ip: 192.168.0.53
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'xxx'

ota:
  password: 'xxx'

binary_sensor:
- platform: gpio
  name: "bathroom_extractor_fan"
  pin:
    number: D0 # Only GPIO16 supports pulldown
    mode: INPUT_PULLDOWN_16

sensor:
- platform: dht
  pin: D1
  model: dht22
  update_interval: 10s
  temperature:
    name: "bathroom_shower_temperature"
  humidity:
    name: "bathroom_shower_humidity"
    filters:
      - filter_nan:
      - exponential_moving_average:
          alpha: 0.5
          send_every: 3

switch:
  - platform: restart
    name: "bathroom_extractor_fan_sensor_restart"
```

**Traceback (if applicable):**

**Additional information and things you've tried:**

OttoWinter, 2019-03-15:
> If I delete the node from integrations and try to add it, it fails

This is very likely the OTA safe mode which kicks in if the ESP repeatedly reboots in a short amount of time (for example when WiFi is not active). After 5 (max 10) minutes they should be online again.

> When I tried to connect it again in few minutes, it asked for a password and said Invalid password.

Invalid password can only be caused by an invalid password (only data corruption could cause that, but TCP has that built in). The password was very likely invalid.

> I tried to enter the same password again and it connected, but showed an empty integration (but the node items are visible in UI!).

Yes the empty integration is an issue in HA (see home-assistant-polymer repo). Reloading the page with CTRL+R or F5 make them re-appear

> Still no logs.

What do you mean by no logs? Empty log window or the dashboard cannot connect?

akasma74, 2019-03-15:
> Yes the empty integration is an issue in HA (see home-assistant-polymer repo). Reloading the page with CTRL+R or F5 make them re-appear

I think I cleared cache and restarted both browser and HA

> > Still no logs.
> 
> What do you mean by no logs? Empty log window or the dashboard cannot connect?

I mean command `esphome bathroom_extractor_fan_sensor.yaml logs` still couldn't connect to the node.

Today that node went Unavailable again as I had to power down my AP for about 10 mins.
I did nothing and it came back online in about 15 minutes despite fixed ip and stuff.
I didn't try to get logs this time.

akasma74, 2019-03-21:
I updated esphome and changed my AP. So far didn't have anything like this.
Will close it then.

closed 2019-03-21 by akasma74