esphome/issues #134: cse7766

brianrjones69, 2019-03-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio/pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Sonoff S31
**Affected component:**

https://esphome.io/components/sensor/cse7766.html

**Description of problem:**
Voltage is reported as 0.0 in a no load situation.
Occasionally the values are not correct.
Spamming the logs in Verbose mode. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
uart:
  tx_pin: GPIO1
#  rx_pin: GPIO3
  baud_rate: 4800

sensor:
  - platform: cse7766
    update_interval: 5s
    current:
      name: "Dryer Current"
      id: current_sensor
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every:  10
    voltage:
      name: "Dryer Voltage"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every:  10
    power:
      name: "Dryer Power"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every:  10

```

**Traceback (if applicable):**

```
The below log is from the original code:  0xF2 is the first byte of the packet, it is valid but we say it is not and continue to look for a valid packet start.  Power cycle exceeds range is (I believe, really a underrange due to no load.
[22:56:32][VV][uart:186]:     Read 0b11110010 (0xF2)
[22:56:32][V][sensor.cse7766:055]: CSE7766 reports abnormal hardware: (0xF2)
[22:56:32][V][sensor.cse7766:063]:   Power cycle exceeds range.
[22:56:32][VV][uart:186]:     Read 0b01011010 (0x5A)
[22:56:32](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[22:56:32][VV][uart:186]:     Read 0b00000010 (0x02)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x02!
[22:56:32][VV][uart:186]:     Read 0b11001100 (0xCC)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0xCC!
[22:56:32][VV][uart:186]:     Read 0b01101000 (0x68)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x68!
[22:56:32][VV][uart:186]:     Read 0b00000000 (0x00)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x00!
[22:56:32][VV][uart:186]:     Read 0b00000101 (0x05)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x05!
[22:56:32][VV][uart:186]:     Read 0b11101101 (0xED)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0xED!
[22:56:32][VV][uart:186]:     Read 0b00000000 (0x00)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x00!
[22:56:32][VV][uart:186]:     Read 0b00111101 (0x3D)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x3D!
[22:56:32][VV][uart:186]:     Read 0b11001100 (0xCC)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0xCC!
[22:56:32][VV][uart:186]:     Read 0b00000010 (0x02)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0x02!
[22:56:32][VV][uart:186]:     Read 0b11010000 (0xD0)
[22:56:32][V][sensor.cse7766:071]: Invalid Header Start from CSE7766: 0xD0!
```

**Additional information and things you've tried:**
I have modified my cse7766 sensor files to better match the spec sheet.  As you can see from the logs below I get 48/49 packets from the cse7766 ic in 5 seconds.  I simply accumulate the values individually and at update() divide the value by the count of packets read to get an average.  I think that multiplying by the time between packets and at update() dividing by the time between updates is causing strange values. 

Here is a log from my revised code.
[00:47:46][D][sensor.cse7766:158]: Got voltage=122.0V current=0.0A power=0.0W
[00:47:51][D][sensor.cse7766:153]: Got voltage_acc=5972.29 current_acc=0.00 power_acc=0.00
[00:47:51][D][sensor.cse7766:155]: Got voltage_counts=49 current_counts=49 power_counts=0
[00:47:51][D][sensor.cse7766:158]: Got voltage=121.9V current=0.0A power=0.0W
[00:47:56][D][sensor.cse7766:153]: Got voltage_acc=5862.94 current_acc=0.00 power_acc=0.00
[00:47:56][D][sensor.cse7766:155]: Got voltage_counts=48 current_counts=48 power_counts=0
[00:47:56][D][sensor.cse7766:158]: Got voltage=122.1V current=0.0A power=0.0W
[00:47:56][D][sensor.sensor:100]: 'Dryer Voltage': Sending state 122.01463 V with 1 decimals of accuracy
[00:47:56][D][sensor.sensor:100]: 'Dryer Current': Sending state -0.00000 A with 1 decimals of accuracy
[00:47:56][D][sensor.sensor:100]: 'Dryer Power': Sending state 0.00012 W with 1 decimals of accuracy

Here is a section with a load:
[01:03:04][D][sensor.cse7766:158]: Got voltage=122.9V current=0.0A power=0.0W
[01:03:09][D][sensor.cse7766:153]: Got voltage_acc=5896.98 current_acc=0.00 power_acc=0.00
[01:03:09][D][sensor.cse7766:155]: Got voltage_counts=48 current_counts=48 power_counts=0
[01:03:09][D][sensor.cse7766:158]: Got voltage=122.9V current=0.0A power=0.0W
[01:03:14][D][sensor.cse7766:153]: Got voltage_acc=5855.74 current_acc=78.93 power_acc=33791.97
[01:03:14][D][sensor.cse7766:155]: Got voltage_counts=48 current_counts=48 power_counts=19
[01:03:14][D][sensor.cse7766:158]: Got voltage=122.0V current=1.6A power=1778.5W
[01:03:19][D][sensor.cse7766:153]: Got voltage_acc=5707.72 current_acc=737.90 power_acc=87808.66
[01:03:19][D][sensor.cse7766:155]: Got voltage_counts=48 current_counts=48 power_counts=48
[01:03:19][D][sensor.cse7766:158]: Got voltage=118.9V current=15.4A power=1829.3W
[01:03:24][D][sensor.cse7766:153]: Got voltage_acc=5711.11 current_acc=735.22 power_acc=87523.81
[01:03:24][D][sensor.cse7766:155]: Got voltage_counts=48 current_counts=48 power_counts=48
[01:03:24][D][sensor.cse7766:158]: Got voltage=119.0V current=15.3A power=1823.4W
[01:03:24][D][sensor.sensor:100]: 'Dryer Voltage': Sending state 121.95223 V with 1 decimals of accuracy
[01:03:24][D][sensor.sensor:100]: 'Dryer Current': Sending state 3.23345 A with 1 decimals of accuracy
[01:03:24][D][sensor.sensor:100]: 'Dryer Power': Sending state 543.12854 W with 1 decimals of accuracy

OttoWinter, 2019-03-17:
> Voltage is reported as 0.0 in a no load situation.

Yes that is expected, the sensor does not give valid values in this case.

> Occasionally the values are not correct.

Have you checked with the latest beta?

> Spamming the logs in Verbose mode.

That is also expected, that's what verbose mode is for.

>  I think that multiplying by the time between packets and at update() dividing by the time between updates is causing strange values.

I don't think that would be it. Have you got logs from without the patch to compare?

OttoWinter, 2019-03-17:
Also, why do you have
```yaml
uart:
  tx_pin: GPIO1
#  rx_pin: GPIO3
```

in your config? The CSE7766 is RX-only, no TX pin.

brianrjones69, 2019-03-17:
The sensor does give valid voltage values in the case on a no load.  The power value is not valid - therefore everything is thrown away.
Wow, I commented out the wrong line and it still works!  

The logs I have are from about 2 weeks ago.  Your latest patch made the data much better, but quite frankly, I cant explain the following lines in cse7766.cpp.  

> // Manually discard unreasonable values
>   // CSE766 sends a bunch of invalid data (or packets that are not documented)
>   // sometimes, they pass the checksum test and lead to wrong values being published
>   if (this->voltage_ != nullptr && voltage < 500.0)
>     this->voltage_->publish_state(voltage);
>   if (this->current_ != nullptr && current < 50.0)
>     this->current_->publish_state(current);
>   if (this->power_ != nullptr && power < 12500.0)
>     this->power_->publish_state(power);

BTW I can do a pull request finally.

closed 2019-03-17 by OttoWinter