esphome/issues #135: ble beacon won't work when the ble tracker is active.

HA-TB303, 2019-03-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 0.89, docker espohome dev 1.13

**ESP (ESP32/ESP8266, Board/Sonoff):**

Lolin32

**Affected component:**

esp32_ble_beacon

**Description of problem:**

ble beacon won't work when the ble tracker is active.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_miflora
  platform: espressif32
  board: lolin32

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  
api:
  password: !secret esphome_api_password

#ota:
#  password: !secret esphome_ota_password

logger:

#deep_sleep:
#  run_duration: 300s
#  sleep_duration: 60min

esp32_ble_beacon:
  type: iBeacon
  uuid: '23074df1-5d24-7109-fa9a-9092663b4782'

esp32_ble_tracker:

sensor:
### mi flora sensors
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:66:E7:0D
    temperature:
      name: "Sanseveria hal temperatuur"
    moisture:
      name: "Sanseveria hal vochtigheid"
    illuminance:
      name: "Sanseveria hal licht"
    conductivity:
      name: "Sanseveria hal geleiding"
    battery_level:
      name: "Sanseveria hal batterij"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:66:E7:16
    temperature:
      name: "Sanseveria keuken temperatuur"
    moisture:
      name: "Sanseveria keuken vochtigheid"
    illuminance:
      name: "Sanseveria keuken licht"
    conductivity:
      name: "Sanseveria keuken geleiding"
    battery_level:
      name: "Sanseveria keuken batterij"
```

**Traceback (if applicable):**

```
Showing logs:
[11:11:10]ets Jun  8 2016 00:22:57
[11:11:10]
[11:11:10]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:11:10]configsip: 0, SPIWP:0xee
[11:11:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:11:10]mode:DOUT, clock div:2
[11:11:10]load:0x3fff0018,len:4
[11:11:10]load:0x3fff001c,len:808
[11:11:10]load:0x40078000,len:6084
[11:11:10]load:0x40080000,len:6704
[11:11:10]entry 0x400802e4
[11:11:10][I][logger:119]: Log initialized
[11:11:10][I][application:055]: Running through setup()...
[11:11:10][C][switch.gpio:018]: Setting up GPIO Switch 'relay'...
[11:11:10][E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: 0 NOT_FOUND
[11:11:10][D][switch:031]: 'relay' Turning OFF.
[11:11:10][D][switch:051]: 'relay': Sending state ON
[11:11:10][D][switch:031]: 'relay' Turning OFF.
[11:11:10][C][esp32_ble_beacon:036]: Setting up ESP32 BLE beacon...
[11:11:11][E][esp32_ble_beacon:074]: esp_bluedroid_init failed: 259
[11:11:11][D][esp32_ble_tracker:165]: Starting scan...
[11:11:11][C][wifi:027]: Setting up WiFi...
[11:11:11][D][wifi:271]: Starting scan...
[11:11:14][D][esp32_ble_tracker:431]: Found device 41:16:30:30:C0:F2 RSSI=-32
[11:11:14][D][esp32_ble_tracker:452]:   Address Type: RANDOM
```

**Additional information and things you've tried:**

OttoWinter, 2019-03-17:
BLE Beacon and tracker cannot be used at the same time. I will make it create a validation error for this.

closed 2019-03-17 by OttoWinter

HA-TB303, 2019-03-17:
Ok, thanks.