esphome/issues #137: Bug with the new wifi_info text sensor in 1.12.0?

erazor666, 2019-03-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
esphome is running on Docker (unraid)

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU, esp8266

**Affected component:**
I added the new wifi info sensor to a node, when the problem occurred

**Description of problem:**
I upgraded my ESPhome docker based installation and tried to add the new wifi_info text sensor
It validated ok and flashed, however the errors below came in the log and the node becomes "weird" for the lack of a better word. The temperature sensor on the board is connected to a graph card in Home Assistant and when the node started it would alternated between `nan` and the value really rapidly. I could not access the states UI in home assistant with the node in this condition.
I re flashed the node without the wifi_sensor config and it acts as it would before.
```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/src/app/esphome/api/client.py", line 396, in run
    self._run_once()
  File "/usr/src/app/esphome/api/client.py", line 385, in _run_once
    msg.ParseFromString(raw_msg)
  File "/usr/local/lib/python2.7/dist-packages/google/protobuf/message.py", line 187, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python2.7/dist-packages/google/protobuf/internal/python_message.py", line 1145, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python2.7/dist-packages/google/protobuf/internal/python_message.py", line 1210, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/usr/local/lib/python2.7/dist-packages/google/protobuf/internal/decoder.py", line 592, in DecodeField
    field_dict[key] = _ConvertToUnicode(buffer[pos:new_pos])
  File "/usr/local/lib/python2.7/dist-packages/google/protobuf/internal/decoder.py", line 548, in _ConvertToUnicode
    value = local_unicode(byte_str, 'utf-8')
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe6 in position 73: 'utf8' codec can't decode byte 0xe6 in position 73: invalid continuation byte in field: SubscribeLogsResponse.message
```

**Problem-relevant YAML-configuration entries:**
Original config:
```yaml
esphome:
  name: basement
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'xxx'
  password: 'xxx'
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 10.0.0.82
    # Set this to the IP address of the router. Often ends with .1
    gateway: 10.0.0.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D2
  scl: D1
  scan: false

sensor:
  - platform: htu21d
    temperature:
      name: "Basement Temperature"
    humidity:
      name: "Basement Humidity"
    update_interval: 60s
  - platform: wifi_signal
    name: "WiFi Signal basement"
    update_interval: 60s

```
Then I added the following:
```yaml
text_sensor:
  - platform: wifi_info
    ip_address:
      name: Basement IP adress
    ssid:
      name: Basement SSID
    bssid:
      name: Basement BSSID
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

bennydiamond, 2019-03-18:
Issue seems related to SSID and BSSID reporting.

in wifi_info.cpp both "SSIDWiFiInfo::loop()" and "BSSIDWiFiInfo::loop()" are weirdly written.

```
void SSIDWiFiInfo::loop() {
  const char *ssid = WiFi.SSID().c_str();
  if (this->last_ssid_ != ssid) {
    this->last_ssid_ = ssid;
    this->publish_state(ssid);
  }
}
```

"last_ssid_ " is a string object and "ssid" is a char const pointer. I'm not a C++ wiz but I don't think there is a default operator for string object comparison with a pointer. The "if" statement will almost always return true.

Then, a pointer value is put into "last_ssid_ " and same pointer is passed as argument to "publish_state" method, which only takes in string object(should pass reference to save stack btw, if possible).

Most likely, the fix only involves dropping the char const * declaration and use "WiFi.SSID()" in it's place.

Apply same rule for "BSSIDWiFiInfo::loop()" method.

I would like to test this myself but I don't know how to modify my own setup (running docker addon in Hass.io) to include custom made changes easily.

OttoWinter, 2019-03-18:
@bennydiamond No that's not it. `std::string` has an equals (and not equals) operator overload for const char*. The check is correct.

Anyway, if the check was the issue, it for sure would not end up with that error on the *python* side. The issue must be that somewhere the wifi info text sensor is printing text that is not UTF-8 or garbage.

OttoWinter, 2019-03-18:
Update: this is a classic use-after-free issue. After the `.c_str()` call the `String` object no longer needs to be allocated and is popped from the stack. That deletes the data buffer and thus we're dealing with deallocated storage here, so UB.

bennydiamond, 2019-03-18:
Ok Thanks.

So I assume assigning char const * to Std::String is also possible?

But then, what about the publish_state method? Is the char const * passed as parameter promoted to Std::string automatically?

OttoWinter, 2019-03-18:
> So I assume assigning char const * to Std::String is also possible?

Yes, but that requires another allocation, using the return type of those methods (`String`) directly is more efficient.

> But then, what about the publish_state method? Is the char const * passed as parameter promoted to Std::string automatically

Yes.

bennydiamond, 2019-03-18:
I saw the updated code.

I wonder why not use WiFi.SSID() directly instead of creating a temp string object? Also, why create another string object from the said temp string object when assigning to "last_ssid_"? 

Seems like a couple could cycles could be saved by using WiFi.SSID() directly and not using c_str() for compare.

Again, not C++ wiz. Just trying to gain knowledge :)

OttoWinter, 2019-03-18:
> I wonder why not use WiFi.SSID() directly instead of creating a temp string object?

The return type of `WiFi.SSID()` _is_ `String`.

> Also, why create another string object from "ssid" temp string object when assigning to "last_ssid_"?

It needs to be converted to a `std::string` (esphome use C++ STL types, arduino uses its own `String` class)

> Seems like a couple could cycles could be saved by using WiFi.SSID() directly and not using c_str() for compare.

Oh the clock cycles are not the problem here. ESPHome does not try to be over efficient (except for a few hot code paths) - mostly the ESP stays asleep anyway. The thing that is optimized here is allocations because of heap fragmentation.

closed 2019-03-18 by OttoWinter

bennydiamond, 2019-03-18:
Ohhh. Make sense. Thanks!