esphome/issues #141: Use of !secret throws python error on validate

bennydiamond, 2019-03-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Armv7 SBC running 0.89.2 and ESPHome 1.12.0

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic gen1 and Wemos D1 Mini but it doesn't seem to be target-related.
**Affected component:**

Validation using !secret

**Description of problem:**
Trying to run validate on previously working sketches now throws an error. 

Is related to the use of !secret as removing all of them from a sketch removes the issue. Putting back only a single one and the issue reappears.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: lum_ext_arr
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
    - ssid: !secret secret_wifi1_ssid
      password: !secret secret_wifi1_password
    - ssid: !secret secret_wifi2_ssid
      password: !secret secret_wifi2_password
  manual_ip:
    static_ip: !secret secret_lum_ext_arr_ip
    gateway: !secret secret_gateway_ip
    subnet: !secret secret_netmask

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret secret_api_password

ota:
  password: !secret secret_ota_password
 
web_server:
  port: 80

sensor:
  - platform: wifi_signal
    name: lum_ext_arr_rssid
    update_interval: 60s
    
text_sensor:
  - platform: wifi_info
    ip_address:
      name: lum_ext_arr_ip
    ssid:
      name: lum_ext_arr_ssid
      
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: pushbutton
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: relay
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Configuration is valid!
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 484, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 474, in run_esphome
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 272, in command_config
    safe_print(yaml_util.dump(config))
  File "/opt/esphome/esphome/yaml_util.py", line 69, in dump
    dict_, default_flow_style=False, allow_unicode=True)
  File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 309, in safe_dump
    return dump_all([data], stream, Dumper=SafeDumper, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 281, in dump_all
    dumper.represent(data)
  File "/usr/local/lib/python2.7/dist-packages/yaml/representer.py", line 29, in represent
    node = self.represent_data(data)
  File "/usr/local/lib/python2.7/dist-packages/yaml/representer.py", line 58, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/opt/esphome/esphome/yaml_util.py", line 390, in <lambda>
    represent_odict(dumper, 'tag:yaml.org,2002:map', value)
  File "/opt/esphome/esphome/yaml_util.py", line 315, in represent_odict
    node_value = dump.represent_data(item_value)
  File "/usr/local/lib/python2.7/dist-packages/yaml/representer.py", line 58, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/opt/esphome/esphome/yaml_util.py", line 390, in <lambda>
    represent_odict(dumper, 'tag:yaml.org,2002:map', value)
  File "/opt/esphome/esphome/yaml_util.py", line 315, in represent_odict
    node_value = dump.represent_data(item_value)
  File "/usr/local/lib/python2.7/dist-packages/yaml/representer.py", line 58, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/opt/esphome/esphome/yaml_util.py", line 390, in <lambda>
    represent_odict(dumper, 'tag:yaml.org,2002:map', value)
  File "/opt/esphome/esphome/yaml_util.py", line 315, in represent_odict
    node_value = dump.represent_data(item_value)
  File "/usr/local/lib/python2.7/dist-packages/yaml/representer.py", line 58, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/opt/esphome/esphome/yaml_util.py", line 350, in stringify_representer
    return represent_secret(data)
  File "/opt/esphome/esphome/yaml_util.py", line 331, in represent_secret
    return yaml.ScalarNode(tag=u'!secret', value=_SECRET_VALUES[value])
KeyError: <esphome.core.IPAddress object at 0x76266d30>
```

**Additional information and things you've tried:**

andriej, 2019-03-18:
+1, have exactly the same issue.
docker armhf image, latest version (1.12)

adonno, 2019-03-19:
+1 same for me

frenck, 2019-03-19:
I have the same issue as well.. according to my CI is has been fixed in dev already:

[image]

OttoWinter, 2019-03-20:
@frenck You're using the `:dev` tag, right? I've run into rate limiting with docker hub and thus that image is no longer automatically updated (FAQ also have updated instructions). I think I will remove the `:dev` tag for clarity.

The issue itself is a quick fix, I think I can get a fix out quick.

closed 2019-03-20 by OttoWinter

frenck, 2019-03-20:
Aah, thanks for clearing that up @OttoWinter. I'll simply remove the `dev` build from my CI in that case.

Thanks for fixing it so darn quick! 👍