esphome/issues #142: esphome symbolic link issues with Windows 10

glmnet, 2019-03-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):** Windows 10 x86

**Affected component:**

https://esphome.io/components/esphome.html#esphome-esphome-core-version
esphome tool, when using it with local esphome-core code

**Description of problem:**
Pointing ESPHome to your local copy relies on using Symbolic links
So far I see two issues related, 
1. It seems that Windows won't allow an Administrator to create symbolic links (simlinks) the solution I found so far is to run the tool as Administrator (even when you are an administrator) I use vscode so right click -> Run as administrator (it would be nice to print an error message suggesting this, the current one is something like "you don't held the rights")
2. In python 2.7 many symlink functions are implemented as `return false` and of course it does not work, interestingly I see the symlink (to create sym links) is "fixed" and it creates the symlink the first time you use it, but next time it fails (attached stack trace) this code relies on other symlink functions: unlink, islink and readlink. Those are not working in Python 2.7 (they seem to work in Python 3 which I did not test)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_local
  platform: ESP8266
  board: nodemcu
  # Use a local copy of ESPHome
  esphome_core_version:
    local: C:\glmgit\glmnet_esphome\esphome-core
```

**Traceback (if applicable):**

```
Traceback (most recent call last):
  File "C:\Python27\Scripts\esphome-script.py", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "c:\glmgit\glmnet_esphome\esphome\esphome\__main__.py", line 484, in main
    return run_esphome(sys.argv)
  File "c:\glmgit\glmnet_esphome\esphome\esphome\__main__.py", line 474, in run_esphome
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "c:\glmgit\glmnet_esphome\esphome\esphome\__main__.py", line 277, in command_compile
    exit_code = write_cpp(config)
  File "c:\glmgit\glmnet_esphome\esphome\esphome\__main__.py", line 150, in write_cpp
    writer.write_platformio_project()
  File "c:\glmgit\glmnet_esphome\esphome\esphome\writer.py", line 477, in write_platformio_project
    write_platformio_ini(content, platformio_ini)
  File "c:\glmgit\glmnet_esphome\esphome\esphome\writer.py", line 448, in write_platformio_ini
    symlink_esphome_core_version(CORE.esphome_core_version)
  File "c:\glmgit\glmnet_esphome\esphome\esphome\writer.py", line 231, in symlink_esphome_core_version
    symlink(src_path, dst_path)
  File "c:\glmgit\glmnet_esphome\esphome\esphome\helpers.py", line 155, in symlink
    raise ctypes.WinError()
WindowsError: [Error 183] Cannot create a file when that file already exists.
```

**Additional information and things you've tried:**
I'm already fixing this, creating this issue to track the problem and do PR later hopefully this week, I've got working simlink, unlink, readlink and islink functions working on Windows 10

glmnet, 2019-03-19:
So I made [this](https://github.com/glmnet/esphome/commit/8dea732172f3f6e3111ff7af08273a556c2a4bca#diff-d64f5b555a99da40715b002c40445223L20) changes and would like to know how to proceed in the contribution, should I make a PR?

OttoWinter, 2019-03-19:
have you seen the contributing guide?

glmnet, 2019-03-19:
Ok, there are two issues, 1st i'm using the wrong linter, will look into it.
2nd this
```
Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.14/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/pkg_resources/__init__.py", line 572, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2752, in load_entry_point
    return ep.load()
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2405, in load
    return self.resolve()
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2411, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/travis/build/esphome/esphome/esphome/__main__.py", line 10, in <module>
    from esphome import const, core_config, mqtt, platformio_api, wizard, writer, yaml_util
  File "/home/travis/build/esphome/esphome/esphome/core_config.py", line 7, in <module>
    from esphome import automation, pins
  File "/home/travis/build/esphome/esphome/esphome/automation.py", line 5, in <module>
    import esphome.config_validation as cv
  File "/home/travis/build/esphome/esphome/esphome/config_validation.py", line 12, in <module>
    from esphome import core
  File "/home/travis/build/esphome/esphome/esphome/core.py", line 15, in <module>
    from esphome.helpers import ensure_unique_string, is_hassio
  File "/home/travis/build/esphome/esphome/esphome/helpers.py", line 10, in <module>
    from ctypes import wintypes
  File "/opt/python/2.7.14/lib/python2.7/ctypes/wintypes.py", line 19, in <module>
    class VARIANT_BOOL(_SimpleCData):
ValueError: _type_ 'v' not supported
```
It is related with interop code, of course it is compiling and working on my environment. Can someone help me here?

closed 2019-03-23 by OttoWinter