esphome/issues #143: Incorrect Readings with CSE7766 Power Sensor

NearlCrews, 2019-03-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu with pip.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff S31.

**Affected component:**

[CSE7766 Power Sensor](https://esphome.io/components/sensor/cse7766.html)

**Description of problem:**
The readings from the CSE7766 are typically correct, but will occasionally throw an extremely high value. The outlet is hooked up to a Raspberry Pi and monitor. Typically readings are around 4W when the monitor is off, and 20W when it's turned on. However, looking through the graph of values, I'll occasionally see one that's *way* outside of what is possible, e.g. the Pi and monitor pulling 10,000W. I didn't have this problem when running ESPurna on the S31.

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sonoff_magicmirror_outlet
  platform: ESP8266
  board: esp01_1m
  esphome_core_version: dev
  arduino_version: dev
  on_boot:
    then:
      - switch.turn_on: relay

wifi:
  ssid: 'SSID'
  password: 'password'
  domain: '.domain.local'

api:

# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 4800

ota:

sensor:
  - platform: wifi_signal
    name: "Sonoff MagicMirror Outlet WiFi"
  - platform: cse7766
    current:
      name: "Sonoff MagicMirror Outlet Current"
    voltage:
      name: "Sonoff MagicMirror Outlet Voltage"
    power:
      name: "Sonoff MagicMirror Outlet Power"
    update_interval: 15s

binary_sensor:
  - platform: status
    name: "Sonoff MagicMirror Outlet Status"
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff MagicMirror Outlet Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    pin: GPIO12
    name: "Sonoff MagicMirror Outlet Relay"
    id: relay
  - platform: restart
    name: "Sonoff MagicMirror Outlet Restart"

status_led:
  pin:
    inverted: True
    number: GPIO13

time:
  - platform: sntp
    id: sntp_time
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org
```

**Traceback (if applicable):**

```
[10:12:36][D][sensor.cse7766:136]: Got voltage=122.2V current=0.0A power=4.4W
[10:12:51][D][sensor.cse7766:136]: Got voltage=121.4V current=0.0A power=4.3W
[10:13:06][D][sensor.cse7766:136]: Got voltage=122.2V current=0.0A power=4.4W
[10:13:21][D][sensor.cse7766:136]: Got voltage=121.3V current=0.0A power=4.3W
[10:13:36][D][sensor.cse7766:136]: Got voltage=122.1V current=0.0A power=4.4W
[10:13:51][D][sensor.cse7766:136]: Got voltage=121.9V current=0.0A power=4.3W
[10:14:06][D][sensor.cse7766:136]: Got voltage=122.2V current=0.0A power=4.4W
[10:14:21][D][sensor.cse7766:136]: Got voltage=121.9V current=0.0A power=6.2W
[10:14:36][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=19.7W
[10:14:51][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=19.6W
[10:15:06][D][sensor.cse7766:136]: Got voltage=121.1V current=0.0A power=19.7W
[10:15:21][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=19.0W
[10:15:36][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=7.4W
[10:15:51][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=958.7W
[10:16:06][D][sensor.cse7766:136]: Got voltage=121.8V current=0.0A power=2181.2W
[10:16:21][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=4.3W
[10:16:36][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=4.4W

```

**Additional information and things you've tried:**

I've tried both the stable and beta branches with the same result. There doesn't seem to be any rhyme or reason for when it spits out those values.

NearlCrews, 2019-03-19:
FYI as a temporary fix for throwing up extremely high values:

```
sensor:
  - platform: wifi_signal
    name: "Sonoff MagicMirror Outlet WiFi"
  - platform: cse7766
    current:
      name: "Sonoff MagicMirror Outlet Current"
    voltage:
      name: "Sonoff MagicMirror Outlet Voltage"
    power:
      name: "Sonoff MagicMirror Outlet Power"
      filters:
      - lambda: |-
          if (x > 1000) return {};
          return x;
    update_interval: 15s
```

...which seems to be working:

```
[10:33:26][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=4.4W
[10:33:26][D][sensor.sensor:100]: 'Sonoff MagicMirror Outlet Power': Sending state 4.38769 W with 1 decimals of accuracy
[10:33:41][D][sensor.cse7766:136]: Got voltage=122.1V current=0.0A power=4.3W
[10:33:41][D][sensor.sensor:100]: 'Sonoff MagicMirror Outlet Power': Sending state 4.34868 W with 1 decimals of accuracy
[10:33:56][D][sensor.cse7766:136]: Got voltage=122.1V current=0.0A power=4.3W
[10:33:56][D][sensor.sensor:100]: 'Sonoff MagicMirror Outlet Power': Sending state 4.33688 W with 1 decimals of accuracy
[10:34:11][D][sensor.cse7766:136]: Got voltage=122.0V current=0.0A power=2552.1W
[10:34:26][D][sensor.cse7766:136]: Got voltage=122.1V current=0.0A power=4.4W
[10:34:26][D][sensor.sensor:100]: 'Sonoff MagicMirror Outlet Power': Sending state 4.37376 W with 1 decimals of accuracy
[10:34:41][D][sensor.cse7766:136]: Got voltage=121.9V current=0.0A power=4.3W
```

Obviously doesn't correct the problem, but at least it'll keep the graphs sane for the time being. Let me know if you need any additional details.

OttoWinter, 2019-03-20:
What version of esphome is this? There have been some changes in 1.12 that should fix some of this.

NearlCrews, 2019-03-20:
`Firmware: 1.12.0 (Mar 20 2019, 00:33:47)`

I’ve also got `esphome_core_version: dev` in the configuration file. I just flashed it a few hours ago with the same issues. There’s 6 Sonoff S31s that are having the same issue, although some more frequently than others. 

Let me know if you need any other details. I’d be more than happy to assist.

OttoWinter, 2019-03-20:
Nah, 1.12.1 doesn't fix that. But could you capture some very verbose logs and post them here? I really shouldn't be working on this integration because I don't have a CSE7766 device but I want to finally fix this :D

NearlCrews, 2019-03-20:
I just enabled VERY_VERBOSE and will leave it running for a while. Unfortunately you're stuck with the API logs from the S31 since you can't have it simultaneously hooked up to the USB-to-serial and plugged into an outlet. Well, you could, but a bit of zapping might result :). I'll post the logs after it's been running for a while. I also removed the lambda filter for now.

Do you have a donation account? I'd be more than happy to cover the cost of a few S31s. It's the least I could do for all of the effort you've put into this project.

NearlCrews, 2019-03-20:
Also, there's a different bug but I can open a separate ticket if you need me to: a lot of times the S31 will report 0.0W even though it's pulling power. Sometimes a soft reset of the S31 will fix it, other times I have to physically unplug it and plug it back it. Eventually it'll start reading the power levels again. Literally just had to do that to get readings after I reflashed it.

NearlCrews, 2019-03-20:
Hrmmm.... I'm kind of wondering if the focus should be on it dropping the readings to `Got voltage=0.0V current=0.0A power=0.0W` when it's pulling power. I can't seem to get an accurate reading at this point. Here's where to look in the (large) log dump that I've attached:

```
[09:31:10][D][sensor.cse7766:136]: Got voltage=115.9V current=0.0A power=18.8W
[09:31:34][D][sensor.cse7766:136]: Got voltage=121.4V current=0.0A power=19.6W
[09:31:34][V][sensor.sensor:020]: 'Sonoff MagicMirror Outlet WiFi': Received new state -60.000000
[09:31:48][D][sensor.cse7766:136]: Got voltage=0.0V current=0.0A power=0.0W
[09:32:03][D][sensor.cse7766:136]: Got voltage=0.0V current=0.0A power=0.0W
[09:32:18][D][sensor.cse7766:136]: Got voltage=0.0V current=0.0A power=0.0W
```

The 18-19W reading is correct, the 0W reading is not.

[sonoff-magicmirror-outlet.zip](https://github.com/esphome/issues/files/2988409/sonoff-magicmirror-outlet.zip)

brianrjones69, 2019-03-20:
Otto, we talked about this issue before and your last changes helped.  Last night I was playing around with it a got some strange messages about the checksum being wrong.  Is the new build using 2.5.0?  I forced it to use 2.4.2 and those checksum errors went away, but it might just be a coincidence.
I will create a pull request tonight to see if we can finally put this to rest.

OttoWinter, 2019-03-20:
> Is the new build using 2.5.0?

That must be a coincidence, the default core has not changed.

@NearlCrews Thanks for the logs, that will help

OttoWinter, 2019-03-20:
I see at least one issue now: At around 09:31:32 the protocol got out of sync and was constantly waiting for a 0x5A packet, but because the whole message has an even number of bytes it will never get synced up again (or until the same issue appears again).

OttoWinter, 2019-03-20:
Ok, when you have time and want to try, test with

```yaml
esphome:
  esphome_core_version:
    branch: fix-cse7766
```

NearlCrews, 2019-03-20:
@OttoWinter I just purged all of the graphs in Home Assistant, removed the lambda filters, and reflashed 6 of the S31 outlets with that build. I'll let them run for a while, check the graphs, and report back. Thanks again!

NearlCrews, 2019-03-20:
Spoke too soon. One of them just registered ~38,000 W on the graph. 

[image]

What would be the easiest way to keep the logs rolling with the level of detail that you need? The problem with VERY_VERBOSE is that the logs grow really, really quick. If you can point me to a config change that will provide the necessary logs I'd be more than happy to load it and reflash.

OttoWinter, 2019-03-20:
:( That is not good and so confusing.

You mentioned espurna works correctly, and I found one occurence where the two handle the data differently (first byte, esphome accepts everything whereas espurna accepts only 0x55 and 0xFX).

I've updated esphome to do the same on the fix-cse7766 branch now.

I've also temporarily added a simple switch statement that makes esphome log the data with WARNING level when the power value is above 1000W.

So, if you could again test that would be awesome (just upload again, you should see some files re-compile) :) And set it to warning too, then you won't have so many logs and it will be easier to pin-point too.

Best case, the change fixes the issue. Worst case, we will get some good logs on it :)

NearlCrews, 2019-03-20:
Tried it but two of them are spamming this error like crazy:

``
[15:43:31][W][sensor.cse7766:097]:   Coefficient storage area is abnormal.
[15:43:31][W][sensor.cse7766:096]: CSE7766 reports abnormal hardware: (0x55)
[15:43:31][W][sensor.cse7766:097]:   Coefficient storage area is abnormal.
[15:43:31][W][sensor.cse7766:096]: CSE7766 reports abnormal hardware: (0x55)
``
Same error is repeated about 10x a second.

OttoWinter, 2019-03-20:
Oh shit. I forgot to remove a line, updated it now. This fix is getting quite a long process :(

NearlCrews, 2019-03-20:
Ha, no worries. Just updated them with the latest version and purged the graphs. I'll keep you posted!

jdenver13, 2019-03-20:
Im using the Anoopsyche wifi smart sockets https://www.amazon.co.uk/dp/B07K24QZHN/ref=sspa_dk_detail_4?psc=1&pd_rd_i=B07K24QZHN
they work well except for the current reading showing 0.0A. The sensor chip in them is a CSE7759B http://218.17.109.235/FolderPublish.aspx?code=A75b6a2a006a241e483e7c3d4556bea14#doc/file/2723541
It is similar to the CSE7766. Here's my file:
`substitutions:
  plug_name: smartplug_1
esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp01_1m
  esphome_core_version:
    branch: fix-cse7766
  arduino_version: dev
wifi:
  ssid: 'ssidname'
  password: 'ssidkey'
logger:
  baud_rate: 0
uart:
  tx_pin: TX
  rx_pin: RX
  baud_rate: 4800
web_server:
  port: 80
api:
  password: 'pasword123'
ota:
  password: 'pasword'
time:
- platform: homeassistant
  id: homeassistant_time
binary_sensor:
- platform: gpio
  pin: GPIO12
  filters:
    - invert:
  name: "${plug_name}_button"
status_led:
  pin:
    inverted: True
    number: GPIO13
switch:
- platform: gpio
  name: "${plug_name}_Relay"
  pin: GPIO14
  restore_mode: ALWAYS_OFF
sensor:
- platform: wifi_signal
  name: "${plug_name} WiFi"
- platform: adc
  pin: VCC
  name: "VCC Voltage"
- platform: cse7766
  voltage:
    name: "${plug_name}_Voltage"
  current:
    name: "${plug_name}_Amperage"
  power:
    name: "${plug_name}_Wattage"
    id: "${plug_name}_Wattage"
- platform: total_daily_energy
  name: "${plug_name}_Total Daily Energy"
  power_id: "${plug_name}_Wattage"
  filters:
    - multiply: 0.001
  accuracy_decimals: 2
  unit_of_measurement: kWh
- platform: uptime
  name: ${plug_name}_Uptime Sensor`

NearlCrews, 2019-03-21:
Unfortunately it popped up again after a few hours on 3 of the graphs:

[image]

NearlCrews, 2019-03-21:
A few more on here. Let me know if you need anything else. It looks like it's still giving values > 1000.

[image]

seaverd, 2019-03-21:
FYI - I am watching this thread as I have the exact same issue as the original poster.  Only difference is my device is a Sonoff POW R2...however same sensor.

jdenver13, 2019-03-22:
My device seems to have settled down now since the update, no more spikes in the graphs, still not displaying the current though.

simbesh, 2019-03-22:
I am having the same issue with a Sonoff R2 (v1.0) 
[image]
As you can see the Amperage is correct (0) the Wattage is not (~97)

```
esphome:
  on_boot:
    priority: 49
    then:
      - switch.turn_on: relay
  name: tv_sonoff_pow_r2
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: "Archer 2.4"
  password: "triplets"

# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 4800

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.toggle: relay
  - platform: template
    name: "TV"
    filters:
      - delayed_off: 180s
    lambda: |-
      if (isnan(id(power).state)) {
        return {};
      } else if (id(power).state > 38) {
        // Running
        return true;
      } else {
        // Not running
        return false;
      }
switch:
  - platform: gpio
    name: "TV Sonoff"
    id: relay
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: pow_blue_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    output: pow_blue_led
    id: led

sensor:
  - platform: cse7766
    update_interval: 5s
    power:
      name: "tv_sonoff_power"
      id: power
      on_value_range:
        - above: 38.0
          then:
            - light.turn_on: led
        - below: 37.0
          then:
            - light.turn_off: led
```

NearlCrews, 2019-03-26:
@OttoWinter did you push out a few changes the that branch? I just updated and it seems to be showing amperage now too, which is wasn't before.

```
[23:31:44][D][sensor.sensor:100]: 'Sonoff Basement Fan Outlet Power': Sending state 76.10742 W with 1 decimals of accuracy
[23:31:59][D][sensor.cse7766:162]: Got voltage=122.3V current=0.6A power=76.1W
[23:31:59][D][sensor.sensor:100]: 'Sonoff Basement Fan Outlet Power': Sending state 76.14986 W with 1 decimals of accuracy
[23:32:14][D][sensor.cse7766:162]: Got voltage=122.2V current=0.6A power=76.2W
[23:32:14][D][sensor.sensor:100]: 'Sonoff Basement Fan Outlet Power': Sending state 76.20154 W with 1 decimals of accuracy
```

OttoWinter, 2019-07-11:
Closing because it has been fixed.

closed 2019-07-11 by OttoWinter