esphome/issues #144: text_sensor: wifi_info - ssid crashes node

akasma74, 2019-03-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, Wemos D1 mini

**Affected component:**
wifi_info

**Description of problem:**
If ssid option of wifi_info platform is used, my node constantly goes on - off according to state sensor.

**Problem-relevant YAML-configuration entries:**
```yaml
text_sensor:
  - platform: wifi_info
    ip_address:
      name: $devicename IP address
#    ssid:
#      name: "SSID"
```

in esphome logs I can see:

> [13:02:16][D][text_sensor.text_sensor:020]: 'SSID': Sending state '	'
> [libprotobuf ERROR google/protobuf/wire_format_lite.cc:611] String field 'SubscribeLogsResponse.message' contains invalid UTF-8 data when parsing a protocol buffer. Use the 'bytes' type if you intend to send raw bytes. 
> Exception in thread Thread-1:
> Traceback (most recent call last):
>   File "/usr/local/Cellar/python@2/2.7.15_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
>     self.run()
>   File "/Users/akasma74/devel/pio/lib/python2.7/site-packages/esphome/api/client.py", line 396, in run
>     self._run_once()
>   File "/Users/akasma74/devel/pio/lib/python2.7/site-packages/esphome/api/client.py", line 385, in _run_once
>     msg.ParseFromString(raw_msg)
> DecodeError: Error parsing message

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

HFsi, 2019-03-19:
I saw that this could be caused when using UTF characters in the SSID instead of standard ASCII characters.
Wich is your SSID?

akasma74, 2019-03-19:
Level2

HFsi, 2019-03-20:
Well, not your case then...

OttoWinter, 2019-03-20:
Duplicate of https://github.com/esphome/issues/issues/137

closed 2019-03-20 by OttoWinter