esphome/issues #145: Core 2.3.0 - support request / WiFi issues with 2.4/2.5

andriej, 2019-03-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** HA / venv / latest / n/a

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266

**Affected component:** core

**Description of problem:** Please make 2.3.0 core version compilable. It's the only one I've tested and can consider stable. Even tasmota with latest, yesterday's release (6.5.0) moved back to 2.3.0 as it's the only stable for most of users. 2.4.x is some kind of joke, 2.5 is not better and loosing connectivity in places where 2.3 is rock solid without even one disconnection.

**Problem-relevant YAML-configuration entries:** n/a

**Traceback (if applicable):** compilation with core_version (arduino) 2.3.0:
```
|   |   |-- <ESP8266WiFi> 1.0
Error: Could not find 'eagle.flash.1m0.ld' LD script in LDPATH '/root/.platformio/packages/framework-arduinoespressif8266@1.20300.1/tools/sdk/lib /root/.platformio/platforms/espressif8266@1.5.0/ldscripts /config/h801_lazienka/.pioenvs/h801_lazienka'
Error: Could not find 'eagle.flash.1m0.ld' LD script in LDPATH '/root/.platformio/packages/framework-arduinoespressif8266@1.20300.1/tools/sdk/lib /root/.platformio/platforms/espressif8266@1.5.0/ldscripts /config/h801_lazienka/.pioenvs/h801_lazienka'
*** [.pioenvs/h801_lazienka/firmware.bin] Error 1
========================== [ERROR] Took 18.27 seconds ==========================
```

**Additional information and things you've tried:**

tatunts, 2019-03-20:
Hello @andriej!

I am having the same problem. Have you managed to fix it?

Thanks.

andriej, 2019-03-20:
Unfortunately - for critical devices I'm sticking with tasmota on 2.3 just because - either tasmota or esp on any other (higher) core failed.
That's why I want to move to ESPHome with this (stable) core.

tatunts, 2019-03-20:
Too bad.. I've just discovered this setting and tried to compile again for a couple of devices that did not reconnected after updating to 1.12. Thanks mate!

Sblop, 2019-03-21:
Are there any way around using ESPHOME to build the bin file? - Can I take the code and paste it directly to Arduino 2.3? I have the same problem. All of my Sonoff devices have stopped working. Searching for network. -

andriej, 2019-03-21:
Hard to say. Maybe let's wait for developers (@OttoWinter ?) to drop us an info.
I've hardly tried for longer than week with half of my devices switched to newest cores, but it was too unstable unfortunately.

tatunts, 2019-03-22:
Hello guys, do you know if it possible to connect to the AP network and upload a new FW OTA? I really don't want to take these devices out and re-flash by cable.. I have both AP and WEBSERVER on my config. Thanks.

brandond, 2019-03-22:
Yes, you can connect to the AP hosted by the device if you have OTA enabled. You'll need to specify the AP-mode IP address on the command line - 192.168.4.1

tatunts, 2019-03-22:
That's really odd, I could not upload the firmware to it. I've connected to the AP with my laptop and tried the upload command using `esphome project.yaml upload --upload-port 192.168.4.1`. No idea why.

andriej, 2019-03-22:
Can we please keep this issue core 2.3.0 related?

OttoWinter, 2019-03-23:
1. If you see this issue upgrading from esphome 1.11 then this definitely isn't core 2.3.0 related. Nothing has changed to that regard - only thing that changed was that the user *can* choose to test 2.5.0 now, but it is not the default.

2. ESPHome has developed a bunch of workarounds for some ESP8266 related WiFi problems, and at least to my knowledge it works quite well (except for some cases, but I doubt those are esp sdk-core related. Also I've gotten messages from people saying there setups were very unstable with tasmota and suddenly stable with esphome). Are you saying you tested esphome with the default version and that did not work? Or is this just because the tasmota guys use it?

3. From a quick test compile, it looks like a bunch of code relies on 2.4.x APIs, and it'd be a pain to backport them manually. Unless there's very clear confirmation that 2.3.x would in fact help things with esphome I don't want to go through that trouble.

andriej, 2019-03-23:
Unfortunately I've tested the core as default and also 2.5.0 - still there are dropouts (meaning entities becoming randomly unavailable after time, even with or without sleep).
What can be a sign of some kind of stability, even tasmota in latest release decided to go back to 2.3 as user-experience with newer core went kinda not as planned.

I hope there's someone to offer help with 2.3 at least for test purposes. On my most important hardware through house I have tasmotas with 2.3 and sleep of 250ms without even one loss of connectivity for weeks. That's my 'stability' benchmark, although I'd love to switch it all to ESPHome to have one solution using discovery.

closed 2019-03-24 by OttoWinter