esphome/issues #146: ESP32 with ble_tracker no longer connects to wifi after upgrading to 1.12

pashar1, 2019-03-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io/pip installed on both RPI with Hass.io and ubuntu 16.04 VM
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 
**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: dinningroom
  platform: espressif32
  board: esp32doit-devkit-v1
#  arduino_version: espressif32@1.0.2

wifi:
  ssid: 'my wifi'
  password: 'my-password'
  manual_ip:
    static_ip: 192.168.XXX.XXX
    gateway: 192.168.XXX.1
    subnet: 255.255.255.0
    dns1: 192.168.XXX.XXX

debug:

# Logger must be at least debug (default)
logger:
  level: very_verbose

# Enable Home Assistant API
api:
  password: 'XXXXXX'

ota:
  password: 'XXXXXX'

esp32_ble_tracker:
  
sensor:
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:65:B2:10
    temperature:
      name: "Plant 1 Temperature"
    moisture:
      name: "Plant 1 Moisture"
    illuminance:
      name: "Plant 1 Illuminance"
    conductivity:
      name: "Plant 1 Soil Conductivity"
    battery_level:
      name: "Plant 1 Battery Level"

```

**Traceback (if applicable):**

```
[21:10:09][VV][esp32_ble_tracker:571]:   Service data: ''
[21:10:09][VV][esp32_ble_tracker:586]: Adv data: 02.01.06.13.FF.4C.00.0C.0E.08.CA.67.06.4C.71.CE.12.68.FE.90.FD.50.62. (23 bytes)
[21:10:09][VV][esp32_ble_tracker:313]: Xiaomi no service data
[21:10:09][V][esp32_ble_tracker:425]: Already discovered device 43:B4:86:77:AD:C4
[21:10:09][D][wifi:271]: Starting scan...
[21:10:10][VV][esp32_ble_tracker:537]: Parse Result:
[21:10:10][VV][esp32_ble_tracker:554]:   Address: 62:91:6F:7D:A6:F2 (RANDOM)
[21:10:10][VV][esp32_ble_tracker:556]:   RSSI: -80
[21:10:10][VV][esp32_ble_tracker:557]:   Name: 
....

[21:10:39][VV][esp32_ble_tracker:571]:   Service data: ''
[21:10:39][VV][esp32_ble_tracker:586]: Adv data: 1B.FF.75.00.42.04.01.80.A0.FC.8F.90.F8.6D.2C.FE.8F.90.F8.6D.2B.01.2A.00.00.00.00.00. (28 bytes)
[21:10:39][VV][esp32_ble_tracker:313]: Xiaomi no service data
[21:10:39][V][esp32_ble_tracker:425]: Already discovered device FC:8F:90:F8:6D:2C
[21:10:39][E][wifi:279]: Scan timeout!
[21:10:39][W][wifi:420]: Restarting WiFi adapter...
[21:10:39][V][wifi:1051]: Disabling STA.
[21:10:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[21:10:39]\033 (241979) wifi: stop sw txqWiFi STA stop
[21:10:39]I (241979) wifi: stop sw txq
[21:10:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
[21:10:40][V][wifi:1315]: Event: WiFi STA stop
[21:10:40][VV][esp32_ble_tracker:537]: Parse Result:
```

**Additional information and things you've tried:**

pashar1, 2019-03-20:
additional logs:

`21:11:38][V][esp32_ble_tracker:425]: Already discovered device F0:18:98:A6:52:1A
[21:11:38][E][wifi:102]: Can't connect to WiFi, rebooting...
[21:11:38][I][helpers:256]: Forcing a reboot... Reason: 'wifi'
[21:11:38][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39]\033 (300990) wifi: stop sw txqWiFi STA stop
[21:11:39]I (300990) wifi: stop sw txq
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
[21:11:39]ets Jun  8 2016 00:22:57
[21:11:39]
[21:11:39]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:11:39]configsip: 0, SPIWP:0xee
[21:11:39]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:11:39]mode:DOUT, clock div:2
[21:11:39]load:0x3fff0018,len:4
[21:11:39]load:0x3fff001c,len:928
[21:11:39]ho 0 tail 12 room 4
[21:11:39]load:0x40078000,len:9280
[21:11:39]load:0x40080400,len:5860
[21:11:39]entry 0x40080698
[21:11:39][I][logger:119]: Log initialized
[21:11:39][VV][preferences:036]: LOAD 0: valid=YES, 0=0x0000000A 1=0x1AFA847E (Type=233825507, CRC=0x1AFA847E)
[21:11:39][C][ota:479]: There have been 10 suspected unsuccessful boot attempts.
[21:11:39][VV][preferences:049]: SAVE 0: 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[21:11:39][E][ota:484]: Boot loop detected. Proceeding to safe mode.
[21:11:39][C][wifi:027]: Setting up WiFi...
[21:11:39][V][wifi:1049]: Enabling STA.
[21:11:39]I (74) wifi: wifi driver task: 3ffd21c0, prio:23, stack:3584, core=0
[21:11:39]I (309) wifi: wifi firmware version: 6b44342
[21:11:39]I (310) wifi: config NVS flash: enabled
[21:11:39]I (312) wifi: config nano formating: disabled
[21:11:39]I (333) wifi: Init dynamic tx buffer num: 32
[21:11:39]I (334) wifi: Init data frame dynamic rx buffer num: 10
[21:11:39]I (334) wifi: Init management frame dynamic rx buffer num: 10
[21:11:39]I (337) wifi: Init static rx buffer size: 1600
[21:11:39]I (341) wifi: Init static rx buffer num: 4
[21:11:39]I (345) wifi: Init dynamic rx buffer num: 10
[21:11:39]I (417) wifi: mode : softAP (30:ae:a4:42:ef:bd)
[21:11:39]I (418) wifi: Init max length of beacon: 752/752
[21:11:39]I (419) wifi: Init max length of beacon: 752/752
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[21:11:39][V][wifi:1353]: Event: WiFi AP start
[21:11:39][V][wifi:1302]: Event: WiFi ready
[21:11:39]I (440) wifi: mode : sta (30:ae:a4:42:ef:bc)
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39][V][wifi:1311]: Event: WiFi STA start
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39]I (458) wifi: Set ps type: 0
[21:11:39]
[21:11:39][D][wifi:271]: Starting scan...
[21:11:42][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[21:11:42][V][wifi:1307]: Event: WiFi Scan Done status=0 number=22 scan_id=128
[21:11:42][D][wifi:286]: Found networks:
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 1
[21:11:42][D][wifi:322]:     RSSI: -54 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 1
[21:11:42][D][wifi:322]:     RSSI: -65 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 6
[21:11:42][D][wifi:322]:     RSSI: -66 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 11
[21:11:42][D][wifi:322]:     RSSI: -79 dB
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[... 82 lines cut ...]
[21:13:45][C][wifi:027]: Setting up WiFi...
[21:13:45][V][wifi:1049]: Enabling STA.
[21:13:45]I (780) wifi: wifi driver task: 3ffea0fc, prio:23, stack:3584, core=0
[21:13:45]I (1721) wifi: wifi firmware version: 6b44342
[21:13:45]I (1722) wifi: config NVS flash: enabled
[21:13:45]I (1724) wifi: config nano formating: disabled
[21:13:45]I (1748) wifi: Init dynamic tx buffer num: 32
[21:13:45]I (1748) wifi: Init data frame dynamic rx buffer num: 10
[21:13:45]I (1748) wifi: Init management frame dynamic rx buffer num: 10
[21:13:45]I (1752) wifi: Init static rx buffer size: 1600
[21:13:45]I (1756) wifi: Init static rx buffer num: 4
[21:13:45]I (1760) wifi: Init dynamic rx buffer num: 10
[21:13:45]I (1775) wifi: mode : softAP (30:ae:a4:42:ef:bd)
[21:13:45]I (1776) wifi: Init max length of beacon: 752/752
[21:13:45]I (1777) wifi: Init max length of beacon: 752/752
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[21:13:45][V][wifi:1de : sta (30:ae:a4:42:ef:bc)
[21:13:45]I (1791) wifi: mode : sta (30:ae:a4:42:ef:bc)
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:13:45][V][wifi:1311]: Event: WiFi STA start
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:13:45][D][wifi:271]: Starting scan...
[21:13:45][V][wifi:1311]: Event: WiFi STA start
[21:13:45][VV][esp32_ble_tracker:537]: Parse Result:`

pashar1, 2019-03-20:
Additional logs

`
[21:11:38][V][esp32_ble_tracker:425]: Already discovered device F0:18:98:A6:52:1A
[21:11:38][E][wifi:102]: Can't connect to WiFi, rebooting...
[21:11:38][I][helpers:256]: Forcing a reboot... Reason: 'wifi'
[21:11:38][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39]\033 (300990) wifi: stop sw txqWiFi STA stop
[21:11:39]I (300990) wifi: stop sw txq
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
[21:11:39]ets Jun  8 2016 00:22:57
[21:11:39]
[21:11:39]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:11:39]configsip: 0, SPIWP:0xee
[21:11:39]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:11:39]mode:DOUT, clock div:2
[21:11:39]load:0x3fff0018,len:4
[21:11:39]load:0x3fff001c,len:928
[21:11:39]ho 0 tail 12 room 4
[21:11:39]load:0x40078000,len:9280
[21:11:39]load:0x40080400,len:5860
[21:11:39]entry 0x40080698
[21:11:39][I][logger:119]: Log initialized
[21:11:39][VV][preferences:036]: LOAD 0: valid=YES, 0=0x0000000A 1=0x1AFA847E (Type=233825507, CRC=0x1AFA847E)
[21:11:39][C][ota:479]: There have been 10 suspected unsuccessful boot attempts.
[21:11:39][VV][preferences:049]: SAVE 0: 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[21:11:39][E][ota:484]: Boot loop detected. Proceeding to safe mode.
[21:11:39][C][wifi:027]: Setting up WiFi...
[21:11:39][V][wifi:1049]: Enabling STA.
[21:11:39]I (74) wifi: wifi driver task: 3ffd21c0, prio:23, stack:3584, core=0
[21:11:39]I (309) wifi: wifi firmware version: 6b44342
[21:11:39]I (310) wifi: config NVS flash: enabled
[21:11:39]I (312) wifi: config nano formating: disabled
[21:11:39]I (333) wifi: Init dynamic tx buffer num: 32
[21:11:39]I (334) wifi: Init data frame dynamic rx buffer num: 10
[21:11:39]I (334) wifi: Init management frame dynamic rx buffer num: 10
[21:11:39]I (337) wifi: Init static rx buffer size: 1600
[21:11:39]I (341) wifi: Init static rx buffer num: 4
[21:11:39]I (345) wifi: Init dynamic rx buffer num: 10
[21:11:39]I (417) wifi: mode : softAP (30:ae:a4:42:ef:bd)
[21:11:39]I (418) wifi: Init max length of beacon: 752/752
[21:11:39]I (419) wifi: Init max length of beacon: 752/752
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[21:11:39][V][wifi:1353]: Event: WiFi AP start
[21:11:39][V][wifi:1302]: Event: WiFi ready
[21:11:39]I (440) wifi: mode : sta (30:ae:a4:42:ef:bc)
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39][V][wifi:1311]: Event: WiFi STA start
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39]I (458) wifi: Set ps type: 0
[21:11:39]
[21:11:39][D][wifi:271]: Starting scan...
[21:11:42][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[21:11:42][V][wifi:1307]: Event: WiFi Scan Done status=0 number=22 scan_id=128
[21:11:42][D][wifi:286]: Found networks:
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 1
[21:11:42][D][wifi:322]:     RSSI: -54 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 1
[21:11:42][D][wifi:322]:     RSSI: -65 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 6
[21:11:42][D][wifi:322]:     RSSI: -66 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 11
[21:11:42][D][wifi:322]:     RSSI: -79 dB
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[... 84 lines cut ...]
[21:13:45][V][wifi:1049]: Enabling STA.
[21:13:45]I (780) wifi: wifi driver task: 3ffea0fc, prio:23, stack:3584, core=0
[21:13:45]I (1721) wifi: wifi firmware version: 6b44342
[21:13:45]I (1722) wifi: config NVS flash: enabled
[21:13:45]I (1724) wifi: config nano formating: disabled
[21:13:45]I (1748) wifi: Init dynamic tx buffer num: 32
[21:13:45]I (1748) wifi: Init data frame dynamic rx buffer num: 10
[21:13:45]I (1748) wifi: Init management frame dynamic rx buffer num: 10
[21:13:45]I (1752) wifi: Init static rx buffer size: 1600
[21:13:45]I (1756) wifi: Init static rx buffer num: 4
[21:13:45]I (1760) wifi: Init dynamic rx buffer num: 10
[21:13:45]I (1775) wifi: mode : softAP (30:ae:a4:42:ef:bd)
[21:13:45]I (1776) wifi: Init max length of beacon: 752/752
[21:13:45]I (1777) wifi: Init max length of beacon: 752/752
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[21:13:45][V][wifi:1de : sta (30:ae:a4:42:ef:bc)
[21:13:45]I (1791) wifi: mode : sta (30:ae:a4:42:ef:bc)
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:13:45][V][wifi:1311]: Event: WiFi STA start
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:13:45][D][wifi:271]: Starting scan...
[21:13:45][V][wifi:1311]: Event: WiFi STA start
[21:13:45][VV][esp32_ble_tracker:537]: Parse Result:
`

pashar1, 2019-03-20:
```
21:11:38][V][esp32_ble_tracker:425]: Already discovered device F0:18:98:A6:52:1A
[21:11:38][E][wifi:102]: Can't connect to WiFi, rebooting...
[21:11:38][I][helpers:256]: Forcing a reboot... Reason: 'wifi'
[21:11:38][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39]\033 (300990) wifi: stop sw txqWiFi STA stop
[21:11:39]I (300990) wifi: stop sw txq
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
[21:11:39]ets Jun  8 2016 00:22:57
[21:11:39]
[21:11:39]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:11:39]configsip: 0, SPIWP:0xee
[21:11:39]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:11:39]mode:DOUT, clock div:2
[21:11:39]load:0x3fff0018,len:4
[21:11:39]load:0x3fff001c,len:928
[21:11:39]ho 0 tail 12 room 4
[21:11:39]load:0x40078000,len:9280
[21:11:39]load:0x40080400,len:5860
[21:11:39]entry 0x40080698
[21:11:39][I][logger:119]: Log initialized
[21:11:39][VV][preferences:036]: LOAD 0: valid=YES, 0=0x0000000A 1=0x1AFA847E (Type=233825507, CRC=0x1AFA847E)
[21:11:39][C][ota:479]: There have been 10 suspected unsuccessful boot attempts.
[21:11:39][VV][preferences:049]: SAVE 0: 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[21:11:39][E][ota:484]: Boot loop detected. Proceeding to safe mode.
[21:11:39][C][wifi:027]: Setting up WiFi...
[21:11:39][V][wifi:1049]: Enabling STA.
[21:11:39]I (74) wifi: wifi driver task: 3ffd21c0, prio:23, stack:3584, core=0
[21:11:39]I (309) wifi: wifi firmware version: 6b44342
[21:11:39]I (310) wifi: config NVS flash: enabled
[21:11:39]I (312) wifi: config nano formating: disabled
[21:11:39]I (333) wifi: Init dynamic tx buffer num: 32
[21:11:39]I (334) wifi: Init data frame dynamic rx buffer num: 10
[21:11:39]I (334) wifi: Init management frame dynamic rx buffer num: 10
[21:11:39]I (337) wifi: Init static rx buffer size: 1600
[21:11:39]I (341) wifi: Init static rx buffer num: 4
[21:11:39]I (345) wifi: Init dynamic rx buffer num: 10
[21:11:39]I (417) wifi: mode : softAP (30:ae:a4:42:ef:bd)
[21:11:39]I (418) wifi: Init max length of beacon: 752/752
[21:11:39]I (419) wifi: Init max length of beacon: 752/752
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[21:11:39][V][wifi:1353]: Event: WiFi AP start
[21:11:39][V][wifi:1302]: Event: WiFi ready
[21:11:39]I (440) wifi: mode : sta (30:ae:a4:42:ef:bc)
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39][V][wifi:1311]: Event: WiFi STA start
[21:11:39][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:11:39]I (458) wifi: Set ps type: 0
[21:11:39]
[21:11:39][D][wifi:271]: Starting scan...
[21:11:42][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[21:11:42][V][wifi:1307]: Event: WiFi Scan Done status=0 number=22 scan_id=128
[21:11:42][D][wifi:286]: Found networks:
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 1
[21:11:42][D][wifi:322]:     RSSI: -54 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 1
[21:11:42][D][wifi:322]:     RSSI: -65 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 6
[21:11:42][D][wifi:322]:     RSSI: -66 dB
[21:11:42][I][wifi:320]: - 'MAIN' [redacted]▂▄▆█
[21:11:42][D][wifi:321]:     Channel: 11
[21:11:42][D][wifi:322]:     RSSI: -79 dB
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[21:11:42][D][wifi:324]: - [redacted] [redacted]▂▄▆█
[... 82 lines cut ...]
[21:13:45][V][wifi:1049]: Enabling STA.
[21:13:45]I (780) wifi: wifi driver task: 3ffea0fc, prio:23, stack:3584, core=0
[21:13:45]I (1721) wifi: wifi firmware version: 6b44342
[21:13:45]I (1722) wifi: config NVS flash: enabled
[21:13:45]I (1724) wifi: config nano formating: disabled
[21:13:45]I (1748) wifi: Init dynamic tx buffer num: 32
[21:13:45]I (1748) wifi: Init data frame dynamic rx buffer num: 10
[21:13:45]I (1748) wifi: Init management frame dynamic rx buffer num: 10
[21:13:45]I (1752) wifi: Init static rx buffer size: 1600
[21:13:45]I (1756) wifi: Init static rx buffer num: 4
[21:13:45]I (1760) wifi: Init dynamic rx buffer num: 10
[21:13:45]I (1775) wifi: mode : softAP (30:ae:a4:42:ef:bd)
[21:13:45]I (1776) wifi: Init max length of beacon: 752/752
[21:13:45]I (1777) wifi: Init max length of beacon: 752/752
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[21:13:45][V][wifi:1de : sta (30:ae:a4:42:ef:bc)
[21:13:45]I (1791) wifi: mode : sta (30:ae:a4:42:ef:bc)
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:13:45][V][wifi:1311]: Event: WiFi STA start
[21:13:45][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[21:13:45][D][wifi:271]: Starting scan...
[21:13:45][V][wifi:1311]: Event: WiFi STA start
[21:13:45][VV][esp32_ble_tracker:537]: Parse Result:
```

OttoWinter, 2019-03-20:
Is `[21:13:45][VV][esp32_ble_tracker:537]: Parse Result:` the very last message you're getting? After that nothing?

Also, have you tried without very verbose mode before? Because of some complicated reasons, very verbose mode can hurt stability when BLE is enabled - I'm aware of the issue but it would be a pain to fix.

The only restart reason I'm seeing is `[21:13:44][E][ota:503]: No OTA attempt made, restarting.` - which would show up when the ESP repeatedly resets. But the logs for such resets are missing. Could you capture some more log data? (Specifically looking for lines after `[21:13:44]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)` and similar)

pashar1, 2019-03-20:
disabling very verbose helped...

OttoWinter, 2019-03-20:
Ok I think in that case I will close this. Maybe a small note can be added in the docs about this but very verbose mode is not meant for production use anyway.

closed 2019-03-20 by OttoWinter

pashar1, 2019-03-20:
I think you closed it too fast... not it worked for about 3-4 hours and stoped responding