esphome/issues #147: Ultrasonic Distance Sensor issues in 1.12

mack0352, 2019-03-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Addon v1.12

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMcu LUA CP2102

**Affected component:**

https://esphome.io/components/sensor/ultrasonic.html

**Description of problem:**
Returns distance measurement of 0.1 m 
Does not change values return. Tested on two different boards and two different HC-SR04. 
I noticed there were a few comments in the compontent section as well saying that 1.12 broke their distance measurement, but I didn't see any bugs logged against it. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: node2
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: myssid
  password: mypw
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.86.46
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.86.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: ultrasonic
    trigger_pin: D1
    echo_pin: D2
    name: "node2 ultrasonic"

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

jfournierphoto, 2019-03-20:
I have the exact same issue, stuck at 0,1m 

I have it on a wemos D1 : 

esphome:
  name: garage_controle
  platform: ESP8266
  board: d1_mini

wifi:
  networks:
  - ssid: wifi-ssid
    password: wifi-password
    manual_ip:
      static_ip: 192.168.x.x
      gateway: 192.168.x.x
      subnet: 255.255.255.0
# Enable logging

logger:
  level: VERBOSE
# Enable Home Assistant API
api:                       

ota:
dallas:
  - pin: D2

binary_sensor:
  - platform: status
    name: "garage-controle-statut"
  
sensor:
  - platform: wifi_signal
    name: "garage_controle signal"
    update_interval: 60s

  - platform: ultrasonic
    trigger_pin: D7
    echo_pin: D6            
    name: "Garage Distance"
    update_interval: 60s

  - platform: dallas
    address: 0x690118669F8CFF28
    name: "Temperature Garage"

    
switch:
  - platform: gpio
    pin: D1
    id: relay
  - platform: template
    name: "Porte de garage"
    icon: "mdi:garage"
    turn_on_action:
    - switch.turn_on: relay
    - delay: 500ms
    - switch.turn_off: relay

OttoWinter, 2019-03-20:
@jfournierphoto  Please put code blocks in triple backticks.

Can you try this:
```yaml
esphome:
  # ...
  esphome_core_version:
    branch: fix-ultrasonic
```
and see if it works?

mack0352, 2019-03-20:
Hi @OttoWinter 

I added that branch to my yaml, and I still get the issue. However, now it says: 

[10:30:49][D][sensor.ultrasonic:065]: Got distance: 0.08 m
[10:30:54][D][sensor.ultrasonic:065]: Got distance: 0.08 m

This was the value that was previously showing up in my Hass.io instance.

jfournierphoto, 2019-03-20:
Added the branch : 

'''
esphome:
  name: garage_controle
  platform: ESP8266
  board: d1_mini
  esphome_core_version:
    branch: fix-ultrasonic
'''

Got same results, stuck at 0.08 (0.1 in HA)

jfournierphoto, 2019-03-20:
@OttoWinter here is what I get:

[16:42:41][C][sensor.ultrasonic:075]: Ultrasonic Sensor 'Garage Distance'
[16:42:41][C][sensor.ultrasonic:075]:   Unit of Measurement: 'm'
[16:42:41][C][sensor.ultrasonic:075]:   Accuracy Decimals: 2
[16:42:41][C][sensor.ultrasonic:075]:   Icon: 'mdi:arrow-expand-vertical'
[16:42:41][C][sensor.ultrasonic:076]:   Echo Pin: GPIO12 (Mode: INPUT)
[16:42:41][C][sensor.ultrasonic:077]:   Trigger Pin: GPIO13 (Mode: OUTPUT)
[16:42:41][C][sensor.ultrasonic:078]:   Pulse time: 10 µs
[16:42:41][C][sensor.ultrasonic:079]:   Timeout: 17492 µs
[16:42:41][C][sensor.ultrasonic:080]:   Update Interval: 60000 ms

[16:44:10][D][sensor.ultrasonic:065]: Got distance: 0.08 m
[16:44:10][V][sensor.sensor:020]: 'Garage Distance': Received new state 0.078204

I tried to put logger to very verbose but the wemos just stopped when I did, had to reflash thru USB to make it respond again...

Brianckramer, 2019-03-21:
Same problem.

Config:
```
sensor:
  - platform: ultrasonic
    trigger_pin: D1
    echo_pin: D5
    name: "Ultrasonic Sensor"
    timeout: 4.0m
    update_interval: 1s
    filters:
      - sliding_window_moving_average:
          window_size: 5
          send_every: 5
  - platform: ultrasonic
    trigger_pin: D2
    echo_pin: D6
    name: "Ultrasonic Sensor2"
    timeout: 4.0m
    update_interval: 1s
    filters:
      - sliding_window_moving_average:
          window_size: 5
          send_every: 5
logger:
api:
ota:
  safe_mode: true
```

Log:
```

[19:53:32][C][wifi:371]: WiFi:
[19:53:32][C][wifi:253]:   SSID: [redacted]
[19:53:32][C][wifi:254]:   IP Address: 192.168.0.118
[19:53:32][C][wifi:256]:   BSSID: [redacted]
[19:53:32][C][wifi:257]:   Hostname: 'garagecarsensor'
[19:53:32][C][wifi:261]:   Signal strength: -64 dB ▂▄▆█
[19:53:32][C][wifi:262]:   Channel: 6
[19:53:32][C][wifi:263]:   Subnet: 255.255.255.0
[19:53:32][C][wifi:264]:   Gateway: 192.168.0.1
[19:53:32][C][wifi:265]:   DNS1: 192.168.0.1
[19:53:32][C][wifi:266]:   DNS2: 0.0.0.0
[19:53:32][C][logger:142]: Logger:
[19:53:32][C][logger:143]:   Level: DEBUG
[19:53:32][C][logger:144]:   Log Baud Rate: 115200
[19:53:32][C][logger:145]:   Hardware UART: UART0
[19:53:32][C][sensor.ultrasonic:061]: Ultrasonic Sensor 'Ultrasonic Sensor'
[19:53:32][C][sensor.ultrasonic:061]:   Unit of Measurement: 'm'
[19:53:32][C][sensor.ultrasonic:061]:   Accuracy Decimals: 2
[19:53:32][C][sensor.ultrasonic:061]:   Icon: 'mdi:arrow-expand-vertical'
[19:53:32][C][sensor.ultrasonic:062]:   Echo Pin: GPIO14 (Mode: INPUT)
[19:53:32][C][sensor.ultrasonic:063]:   Trigger Pin: GPIO5 (Mode: OUTPUT)
[19:53:32][C][sensor.ultrasonic:064]:   Pulse time: 10 µs
[19:53:32][C][sensor.ultrasonic:065]:   Timeout: 23323 µs
[19:53:32][C][sensor.ultrasonic:066]:   Update Interval: 1000 ms
[19:53:32][C][sensor.ultrasonic:061]: Ultrasonic Sensor 'Ultrasonic Sensor2'
[19:53:32][C][sensor.ultrasonic:061]:   Unit of Measurement: 'm'
[19:53:32][C][sensor.ultrasonic:061]:   Accuracy Decimals: 2
[19:53:32][C][sensor.ultrasonic:061]:   Icon: 'mdi:arrow-expand-vertical'
[19:53:32][C][sensor.ultrasonic:062]:   Echo Pin: GPIO12 (Mode: INPUT)
[19:53:32][C][sensor.ultrasonic:063]:   Trigger Pin: GPIO4 (Mode: OUTPUT)
[19:53:32][C][sensor.ultrasonic:064]:   Pulse time: 10 µs
[19:53:32][C][sensor.ultrasonic:065]:   Timeout: 23323 µs
[19:53:32][C][sensor.ultrasonic:066]:   Update Interval: 1000 ms
[19:53:32][C][api:101]: API Server:
[19:53:32][C][api:102]:   Address: garagecarsensor.local:6053
[19:53:32][C][ota:127]: Over-The-Air Updates:
[19:53:32][C][ota:128]:   Address: garagecarsensor.local:8266
[19:53:33][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:33][D][sensor.sensor:100]: 'Ultrasonic Sensor': Sending state 0.09165 m with 2 decimals of accuracy
[19:53:33][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:33][D][sensor.sensor:100]: 'Ultrasonic Sensor2': Sending state 0.09158 m with 2 decimals of accuracy
[19:53:34][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:34][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:35][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:35][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:36][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:36][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:37][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:37][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:38][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:38][D][sensor.sensor:100]: 'Ultrasonic Sensor': Sending state 0.09151 m with 2 decimals of accuracy
[19:53:38][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:38][D][sensor.sensor:100]: 'Ultrasonic Sensor2': Sending state 0.09155 m with 2 decimals of accuracy
[19:53:39][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:39][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:40][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:40][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[19:53:41][D][sensor.ultrasonic:049]: Got distance: 0.1 m
```

talondnb, 2019-03-21:
Same issue here.

talondnb, 2019-03-21:
[16:11:44][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[16:11:44][D][sensor.sensor:100]: 'Ultrasonic Car Sensor': Sending state 0.08266 m with 2 decimals of accuracy
[16:12:08][D][sensor.dht:066]: Got Temperature=1.0°C Humidity=1.0%
[16:12:14][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[16:12:14][D][sensor.sensor:100]: 'Ultrasonic Car Sensor': Sending state 0.08301 m with 2 decimals of accuracy
[16:12:44][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[16:12:44][D][sensor.sensor:100]: 'Ultrasonic Car Sensor': Sending state 0.08335 m with 2 decimals of accuracy

RomanWojnar, 2019-03-24:
Thank you Otto Winter, the suggested correction helped me. The ultrasonic sensor measures again.

mack0352, 2019-03-24:
Thank you @OttoWinter! The branch: fix-ultrasonic is now working for me as well. Appreciate the support as always!

jfournierphoto, 2019-03-24:
Works for me too, big thank you @OttoWinter, I appreciate all the hard work you put in this project so we can easily make our homes smarter!!!

closed 2019-03-24 by OttoWinter