esphome/issues #148: status_use_ping does not seem to work.

lvefferen, 2019-03-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** 
Hass.io

fully routed multi subnet network. ESP devices reside in 192.168.2.x, Hassio resides in subnet 192.168.1.x. (no firewalls enabled, ICMP traffic can traverse) 

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 devices (sonoff basic/Pow/S20)

**Affected component:**

the function status_use_ping to determine online/offline status in the dashboard. devices are fully functional. but status appears as "Unknown"

**Description of problem:**

the function status_use_ping to determine online/offline status in the dashboard. devices are fully functional. but status appears as "Unknown"

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

Ping from Hassio (rpi) host to ESP8266 devices are succesfull

OttoWinter, 2019-03-20:
You were the one that asked about this on Discord, right?

The issue was (probably) that `ping` isn't installed in the docker container. Can you try removing and re-adding the addon repo and trying with the dev version?

If that works for you, it will go live with the next version

lvefferen, 2019-03-20:
No i did not contact you on discord, but it seems very logical, will try dev build. And let you know.

lvefferen, 2019-03-21:
Tried DEV build (1.13.0-dev), but esphome devices still appear with status "Unknown".

OttoWinter, 2019-03-22:
Did you remove and re-add the repo? I just tried it myself and it worked for me.

lvefferen, 2019-03-22:
> Did you remove and re-add the repo? I just tried it myself and it worked for me.

Yep, just tried again, but no cigar. I have portainer installed (underlying server is ubuntu server 18.10) but all is running from docker. With portainer I can open Esphome (DEV) console. ping is not recognized as a command.

OttoWinter, 2019-03-22:
That is a different thing, you wrote Hass.io in the original issue report - the issue *is* fixed for the Hass.io dev image, just not for the esphome *docker* image (those are two different things). Maybe I can also release the same fix for docker today

lvefferen, 2019-03-23:
> That is a different thing, you wrote Hass.io in the original issue report - the issue _is_ fixed for the Hass.io dev image, just not for the esphome _docker_ image (those are two different things). Maybe I can also release the same fix for docker today

I'm using Hassio on but not on rPi I installed it on top of ubuntu server. so technically i'm using Hassio like i mentioned. found the solution in the other issue you are referring.  Also had this issue on rPi3 installed portainer and installing ping resolves the issue. perhaps I was pulling the dev container from cache?

apt-get update
apt-get install iputils-ping

kenmaples, 2019-04-01:
Dev version is failing to install, but I have not looked into why. However, apt-get update && apt-get install iputils-ping resolved this on Hass.io running ESPHome 1.12.2.

OttoWinter, 2019-04-25:
Fixed in https://github.com/esphome/esphome/pull/519

closed 2019-04-25 by OttoWinter