esphome/issues #149: Wifi still on when using deep sleep

john-arvid, 2019-03-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Esp-01
**Affected component:**

https://esphome.io/components/deep_sleep.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp01_window1

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

        
wifi:
  ssid: !secret ssid
  password: !secret wpa
  
  #Recommend a static ip for faster wifi connection
  manual_ip:
    static_ip: 192.168.1.41
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Disable logging for faster execution?
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

deep_sleep:
  id: deep_sleep_1
  #If it takes very long time for the API to connect we have some time
  run_duration: 30s
  #Sleep for a year
  sleep_duration: 8760h

binary_sensor:
  #Specify to get input from HA
  - platform: homeassistant
    #Name not really needed when using "internal:", found out another place in the documentation, that if you define an "id:" but not a "name:", "internal:" is implicit true
    #name: "Input Boolean From Home Assistant"
    #What entity to get input from in HA, this correspond to what you see in HA
    entity_id: input_boolean.ota_update
    #Define that this sensor is not being shown in HA (Is a bit stupid to get input from HA and then send the same back)
    #internal:
    id: ota_update

#Get information about the battery status, not yet implemented good enough
sensor:
  - platform: adc
    pin: VCC
    name: $devicename VCC Voltage
    update_interval: 15s

#Using scripts since I can't do service calls in lambda
script:
  #Script to handle the OTA update that I want to run
  - id: script1
    then:
      #Preventing the 300s deep sleep to occur
      - deep_sleep.prevent: deep_sleep_1
      #Turn off the input_binary in HA
      - homeassistant.service:
          service: input_boolean.turn_off
          data:
            entity_id: input_boolean.ota_update
  #Script to toggle the input binary in HA
  - id: script2
    then:
      - homeassistant.service:
          service: input_boolean.toggle
          data:
            entity_id: input_boolean.keep_ventilation_on
  #Script to enter deep sleep immediately
  - id: script3
    then:
      - deep_sleep.enter: deep_sleep_1

#Using globals to retain variable status. Not sure if necesarry
globals:
  - id: ota_ready
    type: bool
    initial_value: 'false'
  - id: has_state
    type: bool
    initial_value: 'false'

#Since I don't know when the HA api will connect or how I can force it I just have to check now and then. Not to often since that will avoid the connection to be made
interval:
  - interval: 7s
    then:
      - lambda: |-
          //Check if the binary from HA has a state, that means that the API connection is ok and we can proceed with certainty
          if (id(ota_update).has_state()){
            id(has_state) = true;
            // Check if the device should be updated
            if (id(ota_update).state == true){
              id(script1).execute();
              // Avoid toggling the input_boolean
              id(ota_ready) = true;
            }
          }
          else id(has_state) = false;
          
          // When HA api is connected and a update is not waiting
          if (id(has_state) && !id(ota_ready)){
            id(script2).execute();
            id(script3).execute();
          }
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

Tried with a esp-01 were gpio16 is connected to RST, and one that is not.
Both shows logs stating that Beginning Deep Sleep.
But I can still ping the devices.
API is not working, and it does not give anything else out on serial.

john-arvid, 2019-03-20:
The yaml I have pasted is work in progress. Many things that are not working as it should.

brandond, 2019-03-20:
Have you tried changing: https://esphome.io/components/wifi.html#wifi-power-save-mode

john-arvid, 2019-03-20:
Tested it now with HIGH. Did not help, just caused it to spend 20 seconds more to connect to the HA api.

OttoWinter, 2019-03-20:
Interesting, I thought deep sleep would automatically turn off the radio too. But it could probably be different too. However, what I find weird is that it would still respond to ICMP pings, as those are surely handled in the normal CPU code which is definitely shut down during deep sleep.

What happens when you put a `WiFi.mode(WIFI_OFF);` in a on_shutdown action lambda?

john-arvid, 2019-03-20:
Yes it turns radio off when I test with the deep sleep in a clean arduino sketch.

I added it and now it stops the radio. But I would need an oscilloscope to see if it is really in deep sleep.

john-arvid, 2019-03-21:
I have let the device be connected to a battery for some hours now, and it is quite clear that it is not in deep sleep.
The battery is 2700mah+ and during the last 14 hours the battery has gone down about 24,14%, so around 652mah in 14 hours.
And I have removed the LED so there is no power draw there. There is neither any LDO between the battery and the ESP-01. Just the battery, a push button and the ESP-01.

closed 2019-03-31 by OttoWinter