esphome/issues #151: Esphome Docker - devices show as offline when they are infact online

seaverd, 2019-03-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker running in Unraid
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Sonoff
**Affected component:**

No issue with component

**Description of problem:**

Esphome 1.12.1 running as a docker on Unraid 6.6.7

When docker is configured in bridge mode the device shows as offline in the Esphome webgui even though the device is online and the show logs shows that it is active. Device behaves as it should. Only issue is that the status is shown incorrectly in the webgui.

In docker I am forwarding two ports from the host to the container...they are ports 6052 for the webgui and port 6123.

Does this docker need any other ports for the status to show up correctly?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

If I run the docker in host mode the devices show online as they should, but I am trying to avoid running in host mode.

TheHackmeister, 2019-03-22:
I was having the same issue, and spent some time digging into it today. I needed to fix 2 things to get it working.

1. Add `ESPHOME_DASHBOARD_USE_PING="true" ` as an env var. 
2. Ping wasn't installed, so I had to add that to my Dockerfile. I saw that @OttoWinter  addressed this in https://github.com/esphome/esphome-docker-base/commit/e70deec81639fece1f226edb57af38cb42ad5c7b. That's only 2 days old, so you might need to update your image. 

Side note, https://hub.docker.com/r/esphome/esphome-base-amd64/dockerfile should probably point to or reference an up to date Dockerfile.

seaverd, 2019-03-22:
Thanks Hackmeister!

I tried updating the image and the resulting docker container still did not have ping installed.  My docker template is pulling from the following:

[(https://hub.docker.com/r/esphome/esphome)]

For those using Unraid I went into the console of the docker and executed the following to get ping installed:
`apt-get update
apt-get install iputils-ping`

Also in the docker template I added the env variable by adding the following to the extra parameters in the advanced view of the docker template:
`-e ESPHOME_DASHBOARD_USE_PING="true"`

digiblur, 2019-03-22:
Nice!  The green dots are back!  Now if we can just get the iputils-ping included in each build or maybe I'll look into an automated build with it included.

OttoWinter, 2019-03-22:
Duplicate of https://github.com/esphome/issues/issues/148#issuecomment-475124128

And yes, the package will be included in the next build.

seaverd, 2019-04-03:
@OttoWinter 

I see the commit you make adding ping to docker.  I just did a clean docker pull this morning and ping is still missing from the container.  Any suggestions?  I also checked with another user and they confirmed the same.

TheHackmeister, 2019-04-03:
@seaverd 
Is it possible you're using the ottowinter/esphomeyaml image? I'm pretty sure after the name change to ESPHome, you'd need to use esphome/esphome image. 

It's possibly something else, using the old image is just my best guess.

digiblur, 2019-04-03:
Using esphome/esphome here and I had to do the apt-get thing again on the recent update of the container.

seaverd, 2019-04-03:
Yes, I am using the esphome/esphome repository.  I actually didn't start using until after the project name change.

OttoWinter, 2019-04-04:
Huh, you're right. It's not in `esphome/esphome:latest` - I must have messed up tagging it with the automated builds

OttoWinter, 2019-04-25:
Fixed in https://github.com/esphome/esphome/pull/519

closed 2019-04-25 by OttoWinter