esphome/issues #152: Got data coming in HA (native API) but cannot connect to get logs

akasma74, 2019-03-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io 0.89.0

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, Wemos D1 mini

**Affected component:**
logger

**Description of problem:**
I noticed that sometimes my HA receives data from esphome node (I can open temperature graph and it says "updated 15 seconds ago"), but command `esphome bathroom_extractor_fan_node.yaml logs` cannot connect for quite a long time:

> INFO Reading configuration...
> INFO Starting log output from 192.168.0.53 using esphome API
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: timed out). Re-Trying in 1 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: [Errno 64] Host is down). Re-Trying in 2 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: [Errno 64] Host is down). Re-Trying in 4 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: [Errno 64] Host is down). Re-Trying in 8 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: timed out). Re-Trying in 16 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.53: timed out). Re-Trying in 32 seconds
> INFO Connecting to 192.168.0.53:6053 (192.168.0.53)
> INFO Successfully connected to 192.168.0.53
> [08:01:21][I][application:097]: esphome-core version 1.12.1 compiled on Mar 20 2019, 13:38:52

and then in finally connects. The node remains Connected according to the status sensor.

Also, soon before this time I got the following in HA log:

> 2019-03-21 07:32:54 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
> 2019-03-21 07:37:49 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!

I'm not saying that it happens constantly - completely the opposite, it usually connects and provides logs very quickly if the node is online (not rebooting).

But it's alarming that we have different status on different channels (data/logs).
And I have no idea what those warnings in HA log mean.

According to the status sensor , the node was unavailable at 07:32 and 07:36-37, but uptime is high. Couldn't see any significant dips on WiFi signal level graph.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

wifi:
  fast_connect: true
  ssid: !secret wifi_ssid
  password: !secret wifi_passwd
  manual_ip:
    static_ip: 192.168.0.53
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_passwd

ota:
  password: !secret ota_passwd

  - platform: status
    name: ${devicename}_status
```

**Traceback (if applicable):**
```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-22:
This will require more info to debug. From the logs there's not enough data to see what is happening. Please attach USB logs during this happening.

akasma74, 2019-03-22:
I cannot see how it's possible as my node is located inside an extractor fan.. 8()
How should I get USB logs anyway?

OttoWinter, 2019-03-23:
As I said, there is no way for me to help here without more verbose logs - so enable verbose mode and capture logs via USB. You could also try enabling the web server component and AP wifi mode and view the logs with your phone, but I'm not sure that will work (JS is hosted on esphome.io), so USB logs are necessary.

akasma74, 2019-03-23:
Is there any difference between USB and network logs?

OttoWinter, 2019-03-23:
USB logs contain more info because they contain data from the setup phase, before WiFi etc is even setup. Also, when the ESP is having connectivity issues you obviously won't get any logs because well it is not connected.

kcsoft33, 2019-03-28:
I'm hitting something similar but I think I root caused to network isolation.  I put my devices on a secondary VLAN via a different Access Point and when on there I can't do any OTA updates.  I assume something is being blocked since I can actually get logs from the devices but when I goto do the OTA update I get this annoying error that is is waiting for the version or something (I lost the error text).

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter