esphome/issues #155: ESP32 wifi info_sensor not working?

erazor666, 2019-03-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 Wemos32

**Affected component:**
[wifi_info](https://esphome.io/components/text_sensor/wifi_info.html)

**Description of problem:**
Wemos32 and another generic esp32 board I have will both throw boot loops with the new sensor.
These are the the boards tested:
[Wemos](https://www.aliexpress.com/item/For-Wemos-MINI-D1-ESP32-ESP-32S-Wireless-WiFi-Bluetooth-BLE-For-Wemos-D1-Mini-Esp8266/32838275003.html)
[Generic](https://www.aliexpress.com/item/ESP-32-ESP-32S-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP32-Board/32796032726.html)

Terminal output:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
▒[I][logger:119]: Log initialized
[C][ota:479]: There have been 0 suspected unsuccessful boot attempts.
[I][application:055]: Running through setup()...
[C][wifi:027]: Setting up WiFi...
[D][wifi:271]: Starting scan...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandle                                                                                                                                                                                                                                             d.
Core 1 register dump:
PC      : 0x4000c26c  PS      : 0x00060a30  A0      : 0x800d6159  A1      : 0x3f                                                                                                                                                                                                                                             fb1d00
A2      : 0x00000000  A3      : 0x3ffb8ced  A4      : 0x00000006  A5      : 0x00                                                                                                                                                                                                                                             000000
A6      : 0x3ffc1cf4  A7      : 0x3ffb8b30  A8      : 0x00000001  A9      : 0x00                                                                                                                                                                                                                                             000000
A10     : 0x0000300f  A11     : 0x400840c0  A12     : 0x3ffc3d7c  A13     : 0x3f                                                                                                                                                                                                                                             fbef40
A14     : 0x3ffb8058  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00                                                                                                                                                                                                                                             00001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xff                                                                                                                                                                                                                                             ffffff

Backtrace: 0x4000c26c:0x3ffb1d00 0x400d6156:0x3ffb1d20 0x400d3d42:0x3ffb1d90 0x4                                                                                                                                                                                                                                             00d358f:0x3ffb1db0 0x400d2392:0x3ffb1e00 0x400dc32b:0x3ffb1fb0 0x40091245:0x3ffb                                                                                                                                                                                                                                             1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
[I][logger:119]: Log initialized
[C][ota:479]: There have been 1 suspected unsuccessful boot attempts.
[I][application:055]: Running through setup()...
[C][wifi:027]: Setting up WiFi...
[D][wifi:271]: Starting scan...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandle                                                                                                                                                                                                                                             d.
Core 1 register dump:
PC      : 0x4000c26c  PS      : 0x00060a30  A0      : 0x800d6159  A1      : 0x3f                                                                                                                                                                                                                                             fb1d00
A2      : 0x00000000  A3      : 0x3ffb8ced  A4      : 0x00000006  A5      : 0x00                                                                                                                                                                                                                                             000000
A6      : 0x3ffc1cf4  A7      : 0x3ffb8b30  A8      : 0x00000001  A9      : 0x00                                                                                                                                                                                                                                             000000
A10     : 0x0000300f  A11     : 0x400840c0  A12     : 0x3ffc3d7c  A13     : 0x3f                                                                                                                                                                                                                                             fbef40
A14     : 0x3ffb8058  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00                                                                                                                                                                                                                                             00001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xff                                                                                                                                                                                                                                             ffffff
[... 526 lines cut ...]
    gateway: 10.0.0.1
    subnet: 255.255.255.0
logger:
api:
ota:
text_sensor:
  - platform: wifi_info
    ip_address:
      icon: "mdi:ip"
      name: Garage IP
    ssid:
      name: Garage SSID
      icon: "mdi:access-point-network"
    bssid:
      name: Garage BSSID
      icon: "mdi:access-point-network"
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-22:
What esphome version is this? There has been a fix in 1.12.1 for wifi info sensor.

erazor666, 2019-03-22:
It's in 1.12.1, the update fixed the sensor on 8266, cause i have them working. Its just on esp32 the sensor seems broken now. 
(For the record i never tested the sensor on esp32 on 1.12)

closed 2019-03-24 by OttoWinter