esphome/issues #160: MCP23017 problems

ddean6232, 2019-03-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO
**ESP (ESP32/ESP8266, Board/Sonoff):**

WEMOS DI MINI
**Affected component:**

https://esphome.io/components/mcp23017.html?highlight=mcp23017

**Description of problem:**
Cannot get MCP23017 to operate as per instructions.  I have tested the module with the manufacture instructions on an Arduino Uno and it works as documented.  I have attached the manf. instructions to this issue for you ref.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: tempnode
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'xxxx'
  password: 'xxxx'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# I2C INTERFACE
i2c:
  sda: D2
  scl: D1
  scan: False
  frequency: 400kHz

mcp23017:
  - id: 'mcp23017_hub'
    address: 0x27

# Individual outputs
switch:
  - platform: gpio
    name: "MCP23017 Pin #0"
    pin:
      mcp23017: mcp23017_hub
      # Use pin number 0
      number: 0
      # One of INPUT, INPUT_PULLUP or OUTPUT
      mode: OUTPUT
      inverted: False

**Traceback (if applicable):**
<!--
Please copy the traceback here if compilation is failing. If possible, also connect to the ESP over USB and copy its logs into the backticks.
-->
```

```

**Additional information and things you've tried:**

Arduino Uno and manf. instructions and code
[MCP2307_IO_Expansion_Board_User_Manual_EN.pdf](https://github.com/esphome/issues/files/2999014/MCP2307_IO_Expansion_Board_User_Manual_EN.pdf)
[MCP23017-Datasheet.pdf](https://github.com/esphome/issues/files/2999015/MCP23017-Datasheet.pdf)

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

Melkor82, 2019-03-23:
Having the same issue. INPUT mode is working though

ddean6232, 2019-03-24:
Have not tested inputs. Need output.

puerari, 2019-03-25:
I'm having the same issue with OUTPUT. I have not tested inputs because my project is on a PCB.

redstar2k1, 2019-03-26:
I'm also having problems with OUTPUT mode.

puerari, 2019-03-27:
I did a pull request with the solution to the problem.
Waiting for analyse.

redstar2k1, 2019-03-27:
@puerari I've tested your diff and it works for me. Great job.

ddean6232, 2019-03-28:
No sure how I can test. Knowledge limited.

Ausonius, 2019-03-28:
Hi I also have the same problem OUTPUT does not want to work with esp8266 d1_mini while for INPUTs it works cm of can solve

Questo è il codice:

esphome:
  name: esp_realy_8ch
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "my SSID"
  password: "my password"
  manual_ip:
    static_ip: my ip
    subnet: 255.255.255.0
    gateway: my gateway

# Enable logging
logger:
  
# Enable Home Assistant API
api:
  password: 'my_password'

ota:
  password: 'my_password'

i2c:
  sda: 4
  scl: 5
  scan: True

# Example configuration entry
mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20

binary_sensor:
  - platform: gpio
    name: "Pulsante 1"
    pin:
      mcp23017: mcp23017_hub
      number: 0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
    - switch.toggle: relay_1

switch:
  - platform: gpio
    name: "Relay 1"
    id: relay_1
    pin:
      mcp23017: mcp23017_hub
      number: 15
      mode: OUTPUT
      inverted: True

nguyentien1271990, 2019-03-31:
In my Nodemcu, INPUT error when Validate

```
INFO Reading configuration...
Failed config

switch.gpio: [source /config/esphome/tang3.yaml:90]
  platform: gpio
  name: PCF8574 Pin #0
  pin:  [source /config/esphome/tang3.yaml:93]
    pcf8574: pcf8574_hub
    number: 0
    
    Unknown value 'INPUT', must be one of 'OUTPUT'. Got 'INPUT'
    mode: INPUT
    inverted: False
```

### Example configuration entry

i2c:
  sda: D2
  scl: D1
  scan: True
 

pcf8574:
  - id: 'pcf8574_hub'
    address: 0x21
    pcf8575: False

 - platform: gpio
    name: "PCF8574 Pin #0"
    pin:
      pcf8574: pcf8574_hub
      # Use pin number 0
      number: 0
      # One of INPUT, INPUT_PULLUP or OUTPUT
      mode: INPUT
      inverted: False

puerari, 2019-04-01:
Dear @nguyentien1271990 

This issue is about MCP23017.
Your configuration is referring to PCF8574.