esphome/issues #161: Dallas component crashes on GPIO0

maciekczwa, 2019-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip, python2.7

**ESP (ESP32/ESP8266, Board/Sonoff):**

H801 - ESP8266

**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
Dallas component doesn't work on GPIO0. There is crash during boot up.

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: salon_led
  platform: ESP8266
  board: esp01_1m

dallas:
  - pin: GPIO0

wifi:
  ssid: '!secret!'
  password: '!secret!'
  manual_ip:
    static_ip: 192.168.100.51
    gateway: 192.168.100.1
    subnet: 255.255.255.0

# Enable logging
logger:
  hardware_uart: UART1
  level: VERY_VERBOSE

api:
  password: '!secret!'

ota:
  password: '!secret!'

```

**Traceback (if applicable):**

```
[11:20:18]
[11:20:18]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[11:20:18][I][logger:122]: Log initialized
[11:20:18][VV][preferences:034]: LOAD 32: valid=YES, 0=9 1=1242973987 (Type=233825507, CRC=1242973987)
[11:20:18][C][ota:480]: There have been 9 suspected unsuccessful boot attempts.
[11:20:18][VV][preferences:048]: SAVE 32: 0=10 1=452625534 (Type=233825507, CRC=452625534)
[11:20:18][I][application:055]: Running through setup()...
[11:20:18][V][application:056]: Sorting components by setup priority...
[11:20:18][C][sensor.dallas:041]: Setting up DallasComponent...
[11:20:18]Fatal exception 29(StoreProhibitedCause):
[11:20:18]epc1=0x40206d7c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
WARNING Decoded 0x40206d7c: esphome::ESPOneWire::reset_search()
[11:20:18]
[11:20:18]Exception (29):
WARNING Exception type: unknown
[11:20:18]epc1=0x40206d7c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000
WARNING Decoded 0x40206d7c: esphome::ESPOneWire::reset_search()
[11:20:18]
[11:20:18]ctx: cont
[11:20:18]sp: 3ffffc50 end: 3fffffd0 offset: 01a0
[11:20:18]
[11:20:18]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[11:20:18]3ffffdf0:  3ffe8d0f 3fff1344 3fff1344 402078e5
WARNING Decoded 0x402078e5: esp_log_vprintf_(int, char const*, __FlashStringHelper const*, __va_list_tag)
[11:20:18]3ffffe00:  3ffffe50 3ffffe40 0000000c 40204890
WARNING Decoded 0x40204890: _ZZN7esphome11Application5setupEvENKUlPKNS_9ComponentES3_E_clES3_S3_$isra$1 at application.cpp
[11:20:18]3ffffe10:  3ffffe50 3ffffe40 0000000c 40211995
WARNING Decoded 0x40211995: esphome::Component::get_actual_setup_priority() const
[11:20:18]3ffffe20:  3fff13a8 3fff1344 3fff0d34 40207908
WARNING Decoded 0x40207908: esp_log_printf_(int, char const*, __FlashStringHelper const*, ...)
[11:20:18]3ffffe30:  3ffffe50 3ffffe40 3ffe852c 3fff1344
[11:20:18]3ffffe40:  3ffe8d0f 00000004 4020dc64 3fffefb0
WARNING Decoded 0x4020dc64: esp_yield
[11:20:18]3ffffe50:  00000029 008e0000 3fff1344 4020dd1e
WARNING Decoded 0x4020dd1e: yield
[11:20:18]3ffffe60:  3ffffeb0 3fff1344 3fff1344 402092d7
WARNING Decoded 0x402092d7: esphome::sensor::DallasComponent::setup()
[11:20:18]3ffffe70:  00000000 00000000 00000000 00000005
[11:20:18]3ffffe80:  3fff13a8 3fff13c4 3fff13a0 00000004
[11:20:18]3ffffe90:  00000001 3fff1344 3fff1344 402065c0
WARNING Decoded 0x402065c0: esphome::PollingComponent::setup_()
[11:20:18]3ffffea0:  3fff13c4 3ffe981c 3fff13bc 401009e4
WARNING Decoded 0x401009e4: free
[11:20:18]3ffffeb0:  3fff13c4 00000005 3ffef93c 4021879c
WARNING Decoded 0x4021879c: operator delete(void*, std::nothrow_t const&) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/src/gcc-4.8.5/libstdc++-v3/libsupc++/del_opnt.cc:35
[11:20:18]3ffffec0:  3fff13c4 3fff1344 3ffef93c 40205ca6
WARNING Decoded 0x40205ca6: esphome::Application::setup()
[11:20:18]3ffffed0:  3fff13c4 00000005 3fff1344 40208b79
WARNING Decoded 0x40208b79: esphome::sensor::DallasComponent::DallasComponent(esphome::ESPOneWire*, unsigned int)
[11:20:18]3ffffee0:  3fff1274 3ffef93c 3fff1344 3fff1394
[11:20:18]3ffffef0:  00000000 0000ea60 3fff11b4 3ffefaac
[11:20:18]3fffff00:  3fff1274 3ffe88ef 3ffef93c 40202850
WARNING Decoded 0x40202850: setup
[11:20:18]3fffff10:  3fff1108 00000000 00000000 3fff1138
[11:20:18]3fffff20:  00000000 00000001 40212468 3364a8c0
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:18]3fffff30:  40212468 0164a8c0 40212468 00ffffff
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffff40:  40212468 01010101 40212468 01000001
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffff50:  00000000 3fff1328 40212468 3364a8c0
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffff60:  40212468 0164a8c0 40212468 00ffffff
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffff70:  40212468 01010101 40212468 01000001
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffff80:  40212468 3364a8c0 40212468 0164a8c0
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffff90:  40212468 00ffffff 40212468 01010101
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffffa0:  40212468 01000001 feefeffe feefeffe
WARNING Decoded 0x40212468: AsyncServer::setNoDelay(bool)
[11:20:19]3fffffb0:  3fffdad0 00000000 3ffefaa4 4020dce8
WARNING Decoded 0x4020dce8: loop_wrapper() at core_esp8266_main.cpp
[11:20:19]3fffffc0:  feefeffe feefeffe 3ffe852c 40100a2d
WARNING Decoded 0x40100a2d: cont_wrapper
```

**Additional information and things you've tried:**
Changing dallas pin to GPIO1 makes board bootup, however I didn't tried connecting dallas sensor to GPIO1. Dallas normally works on GPIO0 on Espurna firmware

maciekczwa, 2019-03-24:
I found out what is the problem. When you put GPIO0 into config it is translated into code
```sensor::DallasComponent *sensor_dallascomponent = App.make_dallas_component(0);```

then it runs following method
```
DallasComponent *Application::make_dallas_component(ESPOneWire *one_wire, uint32_t update_interval) {
  return this->register_component(new DallasComponent(one_wire, update_interval));
}
```
instead of
```
DallasComponent *Application::make_dallas_component(const GPIOOutputPin &pin, uint32_t update_interval) {
  return this->make_dallas_component(new ESPOneWire(pin.copy()), update_interval);
}
```

So ESPOneWire is not created it is NULL that's why it crashes.

After making following changes it works:
```
#ifdef USE_DALLAS_SENSOR
//DallasComponent *Application::make_dallas_component(ESPOneWire *one_wire, uint32_t update_interval) {
//  return this->register_component(new DallasComponent(one_wire, update_interval));
//}
DallasComponent *Application::make_dallas_component(const GPIOOutputPin &pin, uint32_t update_interval) {
  return this->register_component(new DallasComponent(new ESPOneWire(pin.copy()), update_interval));
}
#endif
```

OttoWinter, 2019-03-24:
@maciekczwa lol, it just had to be something stupid like that :D Darn C implicit casts

That method was never used anyway so I deleted it, fix will follow thanks to your idea!

closed 2019-03-24 by OttoWinter