esphome/issues #162: Help understanding what went wrong updating to latest version

tatunts, 2019-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Sonoff Basic

**Affected component:**

Sensor template

**Description of problem:**

Hello everyone. Sorry to bother again but I've been wondering what I could have done to broke a few sonoffs basics that are not able to connect back to the wifi network after updating to version 1.12 using the exact same yaml files to compile and flash OTA. Those devices were working perfeclty, no issues at all, after an OTA flash the led keeps flashing, I can see they are trying to connect to my router (Asus RT-AC68U) with incorrect values of transfer rate, lease time and so on, after a few seconds they simply disconnect.

After dozens os tries, even erasing the flash memory, I've noticed that if I take this template sensor out and flashing using cables, those devices return immediately. I don't get it how they were working just fine before the update and also why all my other 20+ sonoffs that was updated using the exact same method came back working flawlessly.

I've flashed 40 devices (including D1 mini, Sonoff Dual, 4CH but mostly Basics) using that template sensor and only 5 of them (all basics) broke.

This template sensor is used to calculate the RSSI quality in percentage using a non-linear formula.

Thanks!

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ${project_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: ${wifi_ssid}
    password: ${wifi_psw}
  - ssid: Backup
    password: ${wifi_psw}
  ap:
    ssid: ${friendly_name}
    password: ${ap_psw}
    channel: 6

logger:

web_server:

api:
  password: ${api_psw}

ota:

binary_sensor:

## Buton 00

  - platform: gpio
    id: button_01
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    filters:
      - delayed_on: 600ms
      - delayed_off: 600ms
    on_state:
      then:
        - switch.toggle: main_switch

output:
  - platform: gpio
    pin: GPIO12
    id: main_relay

switch:
  - platform: output
    name: ${switch_entity_id}
    id: main_switch
    output: main_relay
    icon: ${switch_icon}

sensor:
  - platform: wifi_signal
    id: wifi_rssi
    update_interval: 60s

  - platform: template
    name: ${friendly_name} - Sinal Wifi
    unit_of_measurement: '%'
    accuracy_decimals: 0
    update_interval: 60s
    icon: "mdi:wifi"
    lambda: |-
        if (id(wifi_rssi).state < -92.0)
          return 100.0;
        if (id(wifi_rssi).state > -21.0)
          return 1.0;
        else
          return round(( -0.0154 * id(wifi_rssi).state * id(wifi_rssi).state ) - ( 0.3794 * id(wifi_rssi).state ) + 98.182 );

status_led:
  pin:
    number: GPIO13
    inverted: true
```

**Traceback (if applicable):**

```
[32mINFO Reading configuration...[0m
[32mINFO Generating C++ source...[0m
[32mINFO Compiling app...[0m
[32mINFO Running:  platformio run -d casa_sonoffbasic_ventilador_casa[0m
Processing casa_sonoffbasic_ventilador_casa (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.1
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size .pioenvs\casa_sonoffbasic_ventilador_casa\firmware.elf
Checking size .pioenvs\casa_sonoffbasic_ventilador_casa\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  42.3% (used 34656 bytes from 81920 bytes)
PROGRAM: [====      ]  36.9% (used 377976 bytes from 1023984 bytes)
============================================= [SUCCESS] Took 4.04 seconds =============================================
```

**Additional information and things you've tried:**

Flashed many times OTA/cables and erased flash memory using ESPTOOL.

tatunts, 2019-03-24:
I am not sure if this is related but I am now struggling to compile some firmwares using similar templates and method:

```yaml
esphome:
  name: ${project_name}
  platform: ESP8266
  board: esp01_1m
  esphome_core_version: 1.11.2

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_psw}
  fast_connect: true
  ap:
    ssid: ${friendly_name}
    password: ${ap_psw}

logger:

web_server:

api:
  password: ${api_psw}

ota:

binary_sensor:

## Buton 00

  - platform: gpio
    id: button_00
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at least 400ms
        then:
          - switch.toggle: main_switch

output:
  - platform: gpio
    pin: GPIO12
    id: main_relay

switch:
  - platform: output
    name: ${switch_entity_id}
    id: main_switch
    output: main_relay
    icon: ${switch_icon}

sensor:
  - platform: wifi_signal
    id: wifi
    update_interval: 60s

  - platform: template
    name: ${friendly_name} - Sinal Wifi
    unit_of_measurement: '%'
    accuracy_decimals: 0
    icon: "mdi:wifi"
    lambda: |-
        if (id(wifi).state < -92.0)
[... 134 lines cut ...]
^
src\main.cpp:60:71: note: candidate is:
In file included from .piolibdeps\esphome-core\src/esphome/controller.h:8:0,
from .piolibdeps\esphome-core\src/esphome/api/api_server.h:9,
from .piolibdeps\esphome-core\src/esphome/application.h:6,
from .piolibdeps\esphome-core\src/esphome.h:1,
from src\main.cpp:3:
.piolibdeps\esphome-core\src/esphome/switch_/switch.h:106:20: note: template<class T> esphome::switch_::ToggleAction<T>* esphome::switch_::Switch::make_toggle_action()
ToggleAction<T> *make_toggle_action();
^
.piolibdeps\esphome-core\src/esphome/switch_/switch.h:106:20: note:   template argument deduction/substitution failed:
src\main.cpp:60:71: note:   couldn't deduce template parameter 'T'
switch_::ToggleAction<> *action = main_switch->make_toggle_action<>();
^
src\main.cpp:61:15: error: request for member 'add_actions' in '* automation', which is of non-class type 'int'
automation->add_actions({action});
Compiling .pioenvs\casa_sonoffbasic_chuveiro_social\libac1\ESP8266WiFi\WiFiClient.cpp.o^

Compiling .pioenvs\casa_sonoffbasic_chuveiro_social\libac1\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs\casa_sonoffbasic_chuveiro_social\libac1\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
*** [.pioenvs\casa_sonoffbasic_chuveiro_social\src\main.cpp.o] Error 1
================================ [ERROR] Took 17.93 seconds ================================
```

Thanks.

RomanWojnar, 2019-03-26:
I have a very similar problem.
Sonoff Basic after adding code:

`text_sensor:
   - platform: wifi_info
     ip_address:
       name: Wemos ESP IP Address
     ssid:
       name: Wemos ESP Connected SSID`

Sonoff Basic does not connect to WiFi.
Sonoff S26 is not the problem.

OttoWinter, 2019-03-27:
>   esphome_core_version: 1.11.2

Your core and esphome versions need to be in sync. You also need to install v1.11.2 of the python tool to get it to compile.

> using this config with a switch component alongside that code of template sensor .

very unlikely, I believe it is just random - esphome is a highly modularized system, components can *crash* the system (resulting in a stacktrace), but not affect things like wifi.

For me to help with this issue the compile logs will not help - they're basically useless to me. I will need logs over USB and preferable verbose log setting.

@RomanWojnar 
WiFi Info has been fixed in 1.12.1 - that's a separate issue.

RomanWojnar, 2019-03-28:
@OttoWinter 

tatunts napsal:

> **Operating environment/Installation (Hass.io/Docker/pip/etc.):**
>
> **ESP (ESP32/ESP8266, Board/Sonoff):**
>
> ESP8266, Sonoff Basic
>
> **Affected component:**
>
> Sensor template
>
> **Description of [...]

and that's exactly what happens to me.
Only Sonoff Basic, Sonoff S26 is OK.
I just added this code:
```text_sensor:
  - platform: wifi_info
    ip_address:
      name: Sonoff ESP IP Address
    ssid:
      name: Sonoff ESP Connected SSID
```
ESPHome  version 1.12.1

tatunts, 2019-03-30:
Hello again.

I was trying to get a valid log data from the USB connection but with the Sonoff Basic it is impossible since when the pins are disconnected or if I reset the device by disconnecting the USB after flashing, the connection gets cut.

Since I need this device up and running, I've removed it from where it was, connected over USB, erased the flash using ESPTool and re-flashed with the `1.12.1` version with no luck. Same behavior as before. Then I've once again erased the flash memory and flashed with a backup that I had with the previous working version (`1.11.2`) and it worked flawlessly, exact same yaml file, exact flashing method but with previous core version.

I really think there is something that have changed with the latest release, unfortunately.

For now I will try to restore the other devices with the previous version and see how it goes.

I wonder if there is a plan of releasing a new version soon? I would like to try again and see if it happens again.

Thanks!

tatunts, 2019-03-30:
Another problematic code:

```yaml
esphome:
  name: ${project_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_psw}
  fast_connect: true
  ap:
    ssid: ${friendly_name}
    password: ${ap_psw}

logger:

web_server:

api:
  password: ${api_psw}

ota:

binary_sensor:

## Buton 00

  - platform: gpio
    id: button_00
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at least 400ms
        then:
          - light.toggle: main_light
      - timing:
          - ON for at most 1s
          - OFF for at most 399ms
          - ON for at most 1s
          - OFF for at least 400ms
        then:
          - text_sensor.template.publish:
              id: buttons
              state: "button_pressed"
          - delay: 50ms
          - text_sensor.template.publish:
              id: buttons
              state: "button_00_double"
      - timing:
          - ON for at most 1s
          - OFF for at most 399ms
          - ON for at most 1s
          - OFF for at most 399ms
          - ON for at most 1s
          - OFF for at least 400ms
        then:
          - text_sensor.template.publish:
              id: buttons
              state: "button_pressed"
          - delay: 50ms
          - text_sensor.template.publish:
[... 162 lines cut ...]
          - ON for at most 1s
          - OFF for at least 400ms
        then:
          - text_sensor.template.publish:
              id: buttons
              state: "button_pressed"
          - delay: 50ms
          - text_sensor.template.publish:
              id: buttons
              state: "button_00_many"
      - timing:
          - ON for at least 2s
        then:
          - text_sensor.template.publish:
              id: buttons
              state: "button_pressed"
          - delay: 50ms
          - text_sensor.template.publish:
              id: buttons
              state: "button_00_long"
```

Very strange, indeed.

Thanks.

tatunts, 2019-04-11:
Hello.

Unfortunately, the problem was not solved with the latest update.

It seems that the problem revolves around the device type, as being most complains of people (like me) running ESPHome on Sonoff Basics, like this: https://github.com/esphome/issues/issues/163#issuecomment-482100334

I've flashed yesterday another Sonoff basic that was not able to connect to wifi after a update and the problem persisted after re-flashing with the latest version. The solution was removing the `binary_sensor` component. Here is the code used:

```yaml
esphome:
  name: ${project_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_psw}
  fast_connect: false
  ap:
    ssid: ${friendly_name}
    password: ${ap_psw}

logger:

web_server:

api:
  password: ${api_psw}

ota:

binary_sensor:

## Buton 00

  - platform: gpio
    id: button_00
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at least 400ms
        then:
          - switch.toggle: main_switch

output:
  - platform: gpio
    pin: GPIO12
    id: main_relay

switch:
  - platform: output
    name: ${switch_entity_id}
    id: main_switch
    output: main_relay
    icon: ${switch_icon}

sensor:
  - platform: wifi_signal
    id: wifi
    update_interval: 60s

  - platform: template
    name: ${friendly_name} - Sinal Wifi
    unit_of_measurement: '%'
    accuracy_decimals: 0
    icon: "mdi:wifi"
    lambda: |-
        if (id(wifi).state < -92.0)
          return 100.0;
        if (id(wifi).state > -21.0)
          return 1.0;
        else
          return round(( -0.0154 * id(wifi).state * id(wifi).state ) - ( 0.3794 * id(wifi).state ) + 98.182 );

status_led:
  pin:
    number: GPIO13
    inverted: true
```

Everything worked perfectly after the first flash then broke down after the update...

Thanks!

mcpgza, 2019-04-27:
I hope it will help, it was dumped to my usb from this borad: https://www.banggood.com/Wemos-Mega-WiFi-R3-ATmega2560ESP8266-32Mb-Memory-USB-TTL-CH340G-Compatible-For-Arduino-Mega-p-1205437.html 

[0;36m[D][custom:033]:  Time: 2019-04-24 22:51:57[0m
[0;36m[D][api:246]: Error from client 'ESPHome v1.12.2 (192.168.0.111)': -14[0m

Exception (9):
epc1=0x40104b0c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys 
sp: 3fffeac0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffec60:  4023df5e 3ffee1a0 4023df0f 3fff145b  
3fffec70:  00000000 3ffe8308 3fffed20 402209d4  
3fffec80:  00000000 ffffffff ffffffff 00000000  
3fffec90:  00000002 00000051 3f302064 00000001  
3fffeca0:  3fff0674 00000329 00000329 4010020c  
3fffecb0:  00000080 3fff2b0c 3fff31d4 401006dc  
3fffecc0:  40104d38 06078228 3fff0620 00000000  
3fffecd0:  3ffefd20 3fff0620 00000080 40204aaa  
3fffece0:  00000000 06077c0e 4020184c 00000005  
3fffecf0:  3fff0620 3fff1461 00000004 00000004  
3fffed00:  3fff14b2 00000058 3fff31c4 4020323b  
3fffed10:  001d5500 3fffed44 3fff1461 40204cf2  
3fffed20:  00000051 3fffed03 3fff1461 40204db5  
3fffed30:  3fff1461 00000000 3fff31c4 402038b7  
3fffed40:  3fff1461 3fff31d4 40106760 0004f09a  
3fffed50:  00000000 3fff13d4 3fff050c 3ffe8ce8  
3fffed60:  00000004 3fff25e8 3fff25e4 4020391a  
3fffed70:  3fff1461 401037d6 3fff050c 3ffe8ce8  
3fffed80:  3fff1461 3fff1a0c 3fff1461 3ffe8ce8  
3fffed90:  3fff1461 3fff2734 3fff2724 40208a72  
3fffeda0:  00000004 3fffee40 0000000c 00000000  
3fffedb0:  00000022 4010325e 00040000 3fff1461  
3fffedc0:  3ffe8ce8 00000004 3fff13d4 40208ae5  
3fffedd0:  00000051 00000000 50572200 4000050c  
3fffede0:  00000014 3fff3040 3fff3254 3fff0908  
3fffedf0:  3fff303c fffffff2 3fff31c4 402088c9  
3fffee00:  3fffee50 3fffee40 0000000c 00000006  
3fffee10:  3fffee50 3fffee40 0000000c 00000022  
3fffee20:  3fffee30 00000200 fffff1ff 402088ec  
3fffee30:  3fffee50 3fffee40 0000000c 4022c068  
3fffee40:  3fff0b50 00000000 00000000 3ffe8ce8  
3fffee50:  000000f6 3fff2000 40106760 4022c08e  
3fffee60:  3fff0b50 00000000 00000000 402030fe  
3fffee70:  fffffff2 000003d7 000003d7 4010020c  
3fffee80:  00000006 00000001 3fff30e4 4020311c  
3fffee90:  00000430 3fff3040 00000003 40211cad  
3fffeea0:  3fff0b50 3fff08ec 3fff08ed 40211cd4  
3fffeeb0:  00000036 00000000 00000014 40227e49  
3fffeec0:  8d00a8c0 000014e9 00000000 00000000  
3fffeed0:  00000000 00000036 00000048 3fff0fb2  
3fffeee0:  3fff0b50 3fff0f94 3fff0c38 4022be21  
3fffeef0:  00000014 0000027e 0000027e 3fff0b50  
3fffef00:  3fffdc80 3fff1214 3fff2044 3fff12b4  
3fffef10:  00000008 3fff0b50 3fff0f94 402252b0  
3fffef20:  3fffdc80 3fff1214 3fff2044 402250eb  
3fffef30:  402481d2 3fff1214 3fff2044 402481e3  
3fffef40:  3fff0fa4 3fff0f94 00000000 3fffdcb0  
3fffef50:  4023cc7e 00000000 3fff2044 402439d7  
3fffef60:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffef70:  40000e19 40001878 00000002 12e499f0  
3fffef80:  3fffff10 aa55aa55 000000ed 4010441c  
3fffef90:  40104422 00000002 12e499f0 0ec57a9d  
3fffefa0:  4010000d a265a2cd a3069888 9c070dcc  
3fffefb0:  40100764 3fffef4c 40100705 3fffff48  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  feefeffe feefeffe feefeffe feefeffe  
3ffff150:  feefeffe feefeffe feefeffe feefeffe  
3ffff160:  feefeffe feefeffe feefeffe feefeffe  
3ffff170:  feefeffe feefeffe feefeffe feefeffe  
3ffff180:  feefeffe feefeffe feefeffe feefeffe  
3ffff190:  feefeffe feefeffe feefeffe feefeffe  
3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff200:  feefeffe feefeffe feefeffe feefeffe  
3ffff210:  feefeffe feefeffe feefeffe feefeffe  
3ffff220:  feefeffe feefeffe feefeffe feefeffe  
3ffff230:  feefeffe feefeffe feefeffe feefeffe  
3ffff240:  feefeffe feefeffe feefeffe feefeffe  
3ffff250:  feefeffe feefeffe feefeffe feefeffe  
3ffff260:  feefeffe feefeffe feefeffe feefeffe  
3ffff270:  feefeffe feefeffe feefeffe feefeffe  
3ffff280:  fee
ets Jan  8 2013,rst cause:2,
[0;33m[W][wifi:399]: Error whi
[0;32m[I][wifi:167]: WiFi Conn

mcpgza, 2019-04-27:
Ok, here is a stacktrace for the dump:

pi@raspberrypi:~ $ java -jar EspStackTraceDecoder.jar ./.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-addr2line nixienode/.pioenvs/nixienode/firmware.elf ./dump.txt
Exception Cause: 9  [LoadStoreAlignmentCause: Load or store to an unaligned address]

0x40104b0c: ets_timer_setfn at ??:?
0x4022c111: ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:685
0x402255a0: ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:182
0x40244078: ets_timer_handler_isr at ??:?
0x402484d3: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:371
0x402440ca: ets_timer_handler_isr at ??:?
0x40243cc7: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000e19: ?? ??:0
0x40001878: ?? ??:0
0x4010441c: call_user_start_local at ??:?
0x40104422: call_user_start_local at ??:?
0x4010000d: call_user_start at ??:?
0x40100764: cont_ret at cont.S.o:?
0x40100705: cont_continue at cont.S.o:?

mcpgza, 2019-04-28:
Another stack trace:

```
Exception Cause: 9  [LoadStoreAlignmentCause: Load or store to an unaligned address]

0x40104b0c: ets_timer_setfn at ??:?
0x4023e35a: pp_attach at ??:?
0x40220cc4: _svfprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c:531
0x4010020c: _umm_free at umm_malloc.c:?
0x401006dc: free at ??:?
0x40104d38: ets_timer_arm_new at ??:?
0x40204ac6: _ZNSt12_Vector_baseIhSaIhEE13_M_deallocateEPhj$isra$9 at util.cpp:?
0x4020184c: delay at ??:?
0x40203257: esphome::api::APIConnection::send_buffer(esphome::api::APIMessageType) at ??:?
0x40204d0e: esphome::api::APIBuffer::encode_varint_raw(unsigned int) at ??:?
0x40204dd1: esphome::api::APIBuffer::encode_string(unsigned int, char const*, unsigned int) at ??:?
0x402038d3: esphome::api::APIConnection::send_log_message(int, char const*, char const*) at ??:?
0x40106760: millis at ??:?
0x40203936: std::_Function_handler<void (int, char const*, char const*), esphome::api::APIServer::setup()::{lambda(int, char const*, char const*)#2}>::_M_invoke(std::_Any_data const&, int, char const*, char const*) at api_server.cpp:?
0x40208ad2: esphome::LogComponent::log_message_(int, char const*, char*, int) at ??:?
0x40208b45: esphome::LogComponent::log_vprintf_(int, char const*, __FlashStringHelper const*, __va_list_tag) at ??:?
0x40104317: lmacRxDone at ??:?
0x40101c2d: trc_NeedRTS at ??:?
0x40208929: esp_log_vprintf_(int, char const*, __FlashStringHelper const*, __va_list_tag) at ??:?
0x4020894c: esp_log_printf_(int, char const*, __FlashStringHelper const*, ...) at ??:?
0x402350a2: sta_input at ??:?
0x4020311a: esphome::api::APIConnection::on_error_(signed char) at ??:?
0x40203138: std::_Function_handler<void (void*, AsyncClient*, signed char), esphome::api::APIConnection::APIConnection(AsyncClient*, esphome::api::APIServer*)::{lambda(void*, AsyncClient*, signed char)#1}>::_M_invoke(std::_Any_data const&, void*, AsyncClient*, signed char) at api_server.cpp:?
0x40211f31: AsyncClient::_error(long) at ??:?
0x40211f58: AsyncClient::_s_error(void*, long) at ??:?
0x40228139: tcp_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:382
0x4022c111: ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:685
0x402255a0: ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:182
0x402253db: esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:439
0x402484c2: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:363
0x402484d3: ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:371
0x4023d09b: ppPeocessRxPktHdr at ??:?
0x40243cc7: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000e19: ?? ??:0
0x40001878: ?? ??:0
0x4010441c: call_user_start_local at ??:?
0x40104422: call_user_start_local at ??:?
0x4010000d: call_user_start at ??:?
0x40100764: cont_ret at cont.S.o:?
0x40100705: cont_continue at cont.S.o:?

```

It seems a known bug by the way

mcpgza, 2019-04-28:
One step further:

@OttoWinter 

Here is a part of code from the above stacktrace:

```
bool APIConnection::send_buffer(APIMessageType type) {
  uint8_t header[20];
  header[0] = 0x00;
  uint8_t header_len = 1;
  encode_varint(header + header_len, &header_len, this->send_buffer_.size());
  encode_varint(header + header_len, &header_len, static_cast<uint32_t>(type));

  size_t needed_space = this->send_buffer_.size() + header_len;

  if (needed_space > this->client_->space()) {
    delay(5);
    if (needed_space > this->client_->space()) {
      if (type != APIMessageType::SUBSCRIBE_LOGS_RESPONSE) {
        ESP_LOGV(TAG, "Cannot send message because of TCP buffer space");
      }
      delay(5);
      return false;
    }
  }
```

According to exception dump, we are under sys context, and it seems delays are not allowed in sys context. What do you think?

OttoWinter, 2019-05-01:
@mcpgza Yeah that is true - log functions shouldn't be called from sys context. I will remove that. However, that still shouldn't get a node to be unresponsive. It will only mean that if the API disconnects with an error (not a clean disconnect) the ESP will reboot - but that can't get the ESP to be totally unresponsive.

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter