esphome/issues #163: After upgrade to 1.21.1 Sonoff Touch cannot connect to Wi-Fi and becomes unstable

davidvidmar, 2019-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS.IO

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Touch

**Description of problem:**

After uploading a new version of firmware with version 1.12.1 the switch will not connect to Wifi. Status LED keeps blinking. After a while switch might or might now freeze - touch button stops responding, LED is not blinking and switch is not connected to wifi.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass  
  power_save_mode: light
  fast_connect: true  
  manual_ip:
    static_ip: 192.168.xx.yy
    gateway: 192.168.xx.1
    subnet: 255.255.255.0   

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_pass

ota:
  password: !secret ota_pass

# Status sensors

sensor:    
  - platform: wifi_signal
    name: "ESPHome $upper_devicename WiFi Signal"
    update_interval: 10s    
  - platform: uptime
    name: "ESPHome $upper_devicename Uptime"

text_sensor:    
  - platform: version
    name: "ESPHome $upper_devicename Version"

switch:
  - platform: restart
    name: "ESPHome $upper_devicename Restart"

binary_sensor:
  - platform: status
    name: "ESPHome $upper_devicename Status"

# https://esphome.io/devices/sonoff.html#sonoff-touch

# Pin     Function
# ------------------------------
# GPIO0   Button (inverted)
# GPIO12  Relay and Red LED
# GPIO13  Blue LED (inverted)

  - platform: gpio    
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true    
    on_press:
      - fan.toggle: fan

output:
  - platform: gpio
    id: relay    
    pin: GPIO12

status_led:
  pin:
    number: GPIO13
    inverted: false

fan:
  # https://esphome.io/components/fan/binary.html
  - platform: binary
    id: fan
    name: $upper_devicename
    output: relay

```

**Additional information and things you've tried:**

I took board out of the switch and pluged it in programmator near wifi router, I could get the switch to connect to wifi. This was I could reflash, but nothing changed. When I put board back into switch I have no wifi connection. Checking logs it keeps connecting then reboots. 

I use static IP. I tried fast connect and light power safe with no notable changes to behavior.

Right not I'm back to Tasmota and switch seems stable.

tatunts, 2019-03-25:
Hello. 

I am also facing this problem. I’ve opened and issue here: https://github.com/esphome/issues/issues/162

Have you tried to take the switch component out? Somehow I think the problem is somehow related to the switch component.

Thanks.

davidvidmar, 2019-03-25:
Hi, @tatunts. I saw you issue but I believe it is different enough to warrant opening a new one. In my case I don't even have a switch (appart from from the reset) and I'm working with fan. I guess I could simplify my config a bit and test it out but it worked rock solid for couple of weeks on 1.11.2 before so I'm not very enthusiastic about it. 

I'm using ESPhome with HASS and don't see a way to downgrade it. I will probably try that route and confirm that this indeed is an issue with version of some packaged library, probably of Core library and possibly next version will be better.

tatunts, 2019-03-25:
Hello @davidvidmar. Have you tried disabling the logger component? Please take a look at here: https://github.com/esphome/issues/issues/82

Thanks.

davidvidmar, 2019-03-25:
I'll have a look. Is your recommendation to turn off / remove logger or setting verbose mode (which would be good idea in my case anyway, to get more info what the disconnect happens)?

tatunts, 2019-03-25:
Yes, I have tried with one of my Sonoffs that is connecting to the wifi and worked for a couple of times, then it became unstable again :( I've flashed 40 devices here in my home and office with ESPHome because this is an amazing work that Otto have done here but the last few days have been a lot of problems here with the new updated version. I am sure there is a way of fixing those issues and soon enough we will have our devices working just fine once more. I am very positive :) Thanks!

davidvidmar, 2019-03-26:
I "downgraded" back to Tasmota for now. I'll keep my fingers crossed that Otto and a team find a culprit and a fix soon.

OttoWinter, 2019-03-27:
>  power_save_mode: light

Here's likely your problem. light power save mode is not the default _because_ it works worse in many cases. (Though it can help in others, but then it would work right away)

davidvidmar, 2019-03-27:
Hi, Otto. Thanks for replying. The power_save_mode was one of the things I tried after the problem started. I didn't help, but it also didn't make things worse. Same thing with fast_connect.

KevinTurnbull, 2019-04-11:
I'm a new user of Sonoff Basic devices who is having this problem porting working code over from other ESP8266 devices. Of note; I have theSonoff Basic V1.1 which has ESP8266EX chips on it if that makes a difference.

As reported by others; a trivial .yaml config for the ESPhome's results in a working install. But if a "switch" is used then it fails on sonoff basic devices but not either ESP32 or Wemos D1 Mini Pro (I've tested my main firmware on those. I've tried running without Logger and that doesn't cause it to be able to connect. However; my main firmware is somewhat complicated making use of a custom C++ header to make and instantiate a radio group class built on top of a switchgroup. Migrating away from using switches to lights as another user had suggested isn't likely the right solution in my case.

mcpgza, 2019-04-26:
Just an additional info. I have wemos  ESP8266, board: esp01_1m, and my board was unstable too. I changed my Wifi AP channel (in my case 1->8), and my problems went away. It seems in my case it was antenna issue.

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter