esphome/issues #164: PWM output on LC01 only binary, no dimming

daenny, 2019-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

Arilux LC01
**Affected component:**

esp8266_pwm

**Description of problem:**
I am coming from Tasmota, but the neat integration of esphome into hass triggered me to switch my devices from Tasmota to Esphome. However, I am running into a problem. I have a LC-01 dimmer, which has only one channel connected (GPIO12). I am running the yaml below. When I toggle the light at 100% all works as expected. But dimming does not work at all. When being off and dimming to some level, it stays off, when at 100% and dimming to less the light stays at 100%.

In Tasmota dimming is not problem.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: stairs
  upper_devicename: Stairs
  ip_end: "215"

esphome:
  name: ${devicename}_light
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'xxxx'
  password: 'xxxx'
  manual_ip:
    static_ip: "192.168.2.${ip_end}"
    gateway: 192.168.2.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'xxxxx'

ota:
  password: 'xxxxx'

output:
  - platform: esp8266_pwm
    pin: GPIO12
    id: pwm_output1
    
# Example configuration entry
light:
  - platform: monochromatic
    name: "${upper_devicename} Light"
    output: pwm_output1
    default_transition_length: 0.1s

switch:
  - platform: restart
    name: "${upper_devicename} Light Restart"

# Example configuration entry
web_server:
  port: 80

```

**Traceback (if applicable):**

```
[00:41:31][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:31][D][light.state:425]:   Brightness: 15%
[00:41:31][D][light.state:459]:   Transition Length: 100 ms
[00:41:33][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:33][D][light.state:425]:   Brightness: 100%
[00:41:33][D][light.state:459]:   Transition Length: 100 ms
[00:41:35][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:35][D][light.state:425]:   Brightness: 28%
[00:41:35][D][light.state:459]:   Transition Length: 100 ms
[00:41:40][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:40][D][light.state:425]:   Brightness: 47%
[00:41:40][D][light.state:459]:   Transition Length: 100 ms
[00:41:41][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:41][D][light.state:425]:   Brightness: 100%
[00:41:41][D][light.state:459]:   Transition Length: 100 ms
[00:41:41][D][light.state:418]: 'Stairs Light' Turning OFF
[00:41:41][D][light.state:459]:   Transition Length: 100 ms
[00:41:43][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:43][D][light.state:459]:   Transition Length: 100 ms
[00:41:44][D][light.state:418]: 'Stairs Light' Turning OFF
[00:41:44][D][light.state:459]:   Transition Length: 100 ms
[00:41:46][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:46][D][light.state:425]:   Brightness: 47%
[00:41:46][D][light.state:459]:   Transition Length: 100 ms
[00:41:48][D][light.state:418]: 'Stairs Light' Turning ON
[00:41:48][D][light.state:425]:   Brightness: 0%
```

**Additional information and things you've tried:**

I have tried different PWM frequencies (540, 1111, 880) These were the references that i could find in the Tasmota source code for the dimming.

voodoobg, 2019-03-25:
I can concur. I have Arilux LC01 RGB and the main colors work, but no dimming and no colors different from main ones.

OttoWinter, 2019-03-27:
Have you tried unplugging and replugging it to power?

daenny, 2019-03-27:
I rebooted a couple of times, but I did not unplug it from power. I will try that later.

daenny, 2019-03-27:
Ok thanks, unplugging and replugging solved it... :man_facepalming: 
Is that mentioned somewhere?

closed 2019-03-27 by daenny

voodoobg, 2019-03-28:
It fixed it for me too. I agree that It should be documented somewhere.

OttoWinter, 2019-03-29:
Documentation updates are always welcome, just add a note in the generic sonoff page somewhere.

I don't own such a device, so it was just a generic suggestion - just had a hunch that could be it.