esphome/issues #165: Home Assistant addon toggles switches on start

chriscrowe, 2019-03-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- HassIO running on Ubuntu VM in Synology DS918
- ESPHome addon running in HassIO

**ESP (ESP32/ESP8266, Board/Sonoff):**

- Teckin SP20 switch, ESP8266, running ESPHome v 1.11.2

**Affected component:**

https://esphome.io/guides/getting_started_hassio.html

**Description of problem:**

When the Home Assistant ESPHome addon (container) starts, it toggles my ESPHome Teckin SP20 switches off and back on again. This obviously presents an issue if the switches are hooked inline on something like a NAS, which could result in data corruption (luckily a well-placed UPS helps in this case).

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  plug_name: teckin1
  # Higher value gives lower watt readout
  current_res: "0.002142347"
  # Lower value gives lower voltage readout
  voltage_div: "946"

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "**********"
  password: "**********"

# Enable logging
logger:

# Enable Web server
web_server:
  port: 80

# Enable Home Assistant API
api:
  password: "**********"

ota:
  password: "**********"

time:
  - platform: homeassistant
    id: homeassistant_time
    
binary_sensor:
  - platform: gpio
    pin:
      number: 13
      inverted: True
    name: "${plug_name}_button"
    on_press:
      - then:
        - switch.toggle: relay
        - if:
            condition:
              switch.is_on: relay
            then:
              switch.turn_on: blue_led
            else:
              switch.turn_off: blue_led
    
switch:
  - platform: gpio
    name: "${plug_name}_Relay"
    pin: 04
    restore_mode: ALWAYS_ON
    id: relay
  - platform: gpio
    name: "${plug_name}_LED_Blue"
    pin:
      number: 2
      inverted: True
    restore_mode: RESTORE_DEFAULT_OFF
    id: blue_led
  - platform: gpio
    name: "${plug_name}_LED_Red"
    pin:
      number: 0
      inverted: True
    restore_mode: RESTORE_DEFAULT_OFF
    id: red_led

sensor:
  - platform: hlw8012
    sel_pin:
      number: 12
      inverted: True
    cf_pin: 5
    cf1_pin: 14
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    current:
      name: "${plug_name}_Amperage"
      unit_of_measurement: A
    voltage:
      name: "${plug_name}_Voltage"
      unit_of_measurement: V
    power:
      name: "${plug_name}_Wattage"
      unit_of_measurement: W
      id: "${plug_name}_Wattage"
    change_mode_every: 8
    update_interval: 10s
  - platform: total_daily_energy
    name: "${plug_name}_Total Daily Energy"
    power_id: "${plug_name}_Wattage"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
      
# Extra sensor to keep track of plug uptime
  - platform: uptime
    name: ${plug_name}_Uptime Sensor
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-03-25:
Can you provide logs from the device when this occurs? I suspect that's what's actually happening is that the devices are restarting because there's no API connection while your addon is down. You could work around this by increasing or disabling the reboot timeout.

chriscrowe, 2019-03-26:
So using a [restart switch](https://esphome.io/components/switch/restart.html) I was able to debug a big more. Found out that on-reset there's a VERY brief blink on the switch-- ie, if the switch is closed, and my config is set for restore_mode: ALWAYS_ON, when the switch resets it'll open and shut very fast (maybe less than 50ms). Unfortunately I think this is enough to cause issues in some situations (eg, I think this triggers my UPS into battery mode which has a chain reaction to other devices subscribing to this power state).

I'm still not fully aware of how to capture useful logs after the fact... The remaining question is still why is the switch having issues that require resets so regularly...

chriscrowe, 2019-03-26:
Updating more of my Teckin switches with this `restart switch` and continuing to see this behavior being troublesome. Anytime the ESPHome software restarts there's a brief toggle On->Off->On on the relay that is long enough to shut down whatever equipment is attached to the switch (example: my home entertainment center, and the cable box takes >5min to reboot). This is pretty annoying honestly, and sometimes happens randomly, presumably because the switch is wedged, perhaps due to the ["low level IP stack issues"](https://esphome.io/components/wifi.html) mentioned on the WiFi component page.

Does ESPHome have the ability to keep the relay closed during reboot? For reference this is the switch I'm using:

https://www.amazon.com/Outlet-Compatible-Monitoring-Function-Required/dp/B079Q5W22B/

CMDR-Sloma, 2019-03-27:
I have UK version of this switch and mine is tripping RCD switch sometimes during reboot loops (especially when it cannot connect to Hassio if Hassio is down for some reason. This is really terrible.

brandond, 2019-03-27:
Unfortunately I this is a limitation of the ESP chipset. They set the pins to default on reset, while some of them output debug logs or a clock signal until the bootloader has started. I'm not aware of any way to fix this, as it's a hardware limitation and not something that can be fixed by code. You'd likely have the same issue with tasmota, etc.

chriscrowe, 2019-03-27:
Had another blink on the plug which powers my NAS overnight... I've got the timeout set for `30min` right now... I could accept that the switch has a hardware limitation that requires a blink on reset but why does it need to reset so often? This seems like a bug in the ESPHome firmware, no?

chriscrowe, 2019-03-27:
Wondering if the ESPHome API could be at fault here for the untimely resets... Do you suppose MQTT is more stable? Looking at my Home Assistant logbook I see some of these switches going in and out of "unavailable" state every 10-30 minutes like clockwork, and they have extremely good wifi signal (I have mesh in my house)

CMDR-Sloma, 2019-03-27:
Mine is super stable as long it is connected to Home Assistant, if the connection drops or HA's restart takes longer than 4 minutes it executes reboot loop.

chriscrowe, 2019-03-27:
Interestingly the switch that did a reset overnight (on my NAS) doesn't show any disconnection/unavailable events in my Home Assistant log, so frankly I'm not at all sure why it did a reset.

brandond, 2019-03-27:
Have y'all tried increasing or disabling the API reboot timeout as I mentioned earlier? Have you pulled logs from any of these devices to see if they show reboots occurring?

chriscrowe, 2019-03-27:
All of my reboot timeouts are set to `30min`

How do I pull useful logs after the fact? I can see the live logs through the ESPHome UI...

OttoWinter, 2019-03-27:
> How do I pull useful logs after the fact? I can see the live logs through the ESPHome UI...

You don't. ESPs don't have the storage for that. You need to capture the logs live - best through USB.

And yes, as @brandond said the relays will trigger for a short amount of time while it boots up, way before any esphome code even runs. So this is expected and correct.

Reboot timeout can also be configured if you don't want it to toggle by itself.

chriscrowe, 2019-03-29:
Copy that-- I'll close this issue then.

closed 2019-03-29 by chriscrowe

chriscrowe, 2019-03-29:
@OttoWinter can you point me to the issues that are tracking the unexplained reboots?

OttoWinter, 2019-03-29:
See the logs, it's in there - and read the automation page

chriscrowe, 2019-03-30:
Sorry I was referring to this statement:

> Why it's rebooting in the first place is a different question, but there are already enough issues open about those - the discussion should be moved there.

Where can I find these discussions?

OttoWinter, 2019-03-30:
> Where can I find these discussions?

See the issue tracker (that you're in right now). Mostly the ones tagged as connection issue.