esphome/issues #167: ESPHome Dashboard Terminated Process

lalank, 2019-03-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassbian on a Pi 3B+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Board/Sonoff

**Affected component:**

Dashboard

**Description of problem:**

Dashboard Edit command works, Show Logs or Validate return a black bar and "Terminated Process".
The command line works, just not the browser interface.  Tried on Firefox 66.01 and Chrome.  Did not try other commands.
The address used to interface with Dashboard:  hassbian.local:6052

**Problem-relevant YAML-configuration entries:**
```yaml
Not relevant - command line works.  This appears to be the browser interface

```

**Traceback (if applicable):**

```
INFO 304 GET /edit?configuration=lohas_color.yaml (192.168.1.125) 1.99ms
INFO 101 GET /logs (192.168.1.125) 4.28ms
INFO Running command 'esphome --dashboard lohas_color//lohas_color.yaml logs --serial-port OTA'
ERROR Uncaught exception GET /logs (192.168.1.125)
HTTPServerRequest(protocol='http', host='hassbian.local:6052', method='GET', uri='/logs', version='HTTP/1.1', remote_ip='192.168.1.125')
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/tornado/websocket.py", line 546, in _run_callback
    result = callback(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/esphome/dashboard/dashboard.py", line 116, in on_message
    stderr=subprocess.STDOUT)
  File "/home/pi/.local/lib/python2.7/site-packages/tornado/process.py", line 240, in __init__
    self.proc = subprocess.Popen(*args, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
INFO 304 GET /edit?configuration=lohas_color.yaml (192.168.1.125) 2.46ms

```

**Additional information and things you've tried:**
Researched web for hours -- no information found to fix this, although there was another reference to the same problem with no responses and another related to a proxy issue.  I am not using a proxy, so that response did not help.
Used wsdump to check the RPi and local win 7 machine - got "Handshake status 200 OK"
I can use the command line, but would really like to use the browser interface.

OttoWinter, 2019-03-27:
> Used wsdump to check the RPi and local win 7 machine - got "Handshake status 200 OK"

That has got nothing to do with the error here.

The issue is with the way you installed esphome I think. I think the `esphome` command is not correctly installed in your system. What happens when you try to run `esphome` from the context the dashboard is executed?

lalank, 2019-03-28:
ESPHome functions perfectly from the command line, a;l the way through compile and OTA to the device.  Verified functional on two different devices.  The Homeassistant API also works - the card associated with the device works perfectly.

I followed the basic command line install instructions for Hassbian but did not install ESPHome in a virtual environment.  ESPHome was installed as user "pi" using "pip install esphome".  This is run under Python 2.7.  It did not sound like the virtual environment was required.

ESPHome was called on the command line by "./esphome test_config/ dashboard" and the FireFox browser was used to access the dashboard at "hassbian.local:6052".

The Dashboard Edit command works; Log and Validate do not.

OttoWinter, 2019-03-29:
> "./esphome test_config/ dashboard"

There's your problem. The dashboard will run `esphome test_config/lohas_color.yaml run` which works for standard installs, but you apparently got it installed as an executable file somehow.

I don't know what happened to your py environment or why you're running it directly as an executable file, you will need to install it in such a way that you have esphome in your PATH.

brandond, 2019-03-29:
Why are you running it as `./esphome`? Did you not put the location of the esphome executable in your $PATH, and instead are just running it out of your working directory? No wonder the dashboard can't run it; it expects it to be in your path and will not run it with an absolute path even if you start the dashboard that way.

lalank, 2019-03-30:
OttoWinter and brandond were correct.  I fixed the path and all is now functional.  I must have missed something in the installation for ESPHome.  Thank you all for taking the time to comment and help me fix my issues!

closed 2019-03-30 by lalank

hoatienii, 2019-04-10:
i have problem

> OttoWinter and brandond were correct. I fixed the path and all is now functional. I must have missed something in the installation for ESPHome. Thank you all for taking the time to comment and help me fix my issues!

how to fix path, thanks, i have same problem

brandond, 2019-04-10:
@hoatienii You can't run the python module directly like that (python -m esphome). You need to have the actual esphome script in your path.