esphome/issues #171: Errors compiling directly from platformio

gnkarn, 2019-03-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP8266, Board/wemos D1):**

**Affected component:**

**Description of problem:**
not able to compile from platformio , just added  2 lines of code , to the autogenerated source code , included the core library, in order to setup the development environment ( not from hassio, as i need to include custom code , so will need to compile directly from plaformio .

**Problem-relevant YAML-configuration entries:**
```yaml
// ========== AUTO GENERATED INCLUDE BLOCK BEGIN ===========
#include "esphome.h"
using namespace esphome;
// ========== AUTO GENERATED INCLUDE BLOCK END ===========

void setup() {
  // ===== DO NOT EDIT ANYTHING BELOW THIS LINE =====
  // ========== AUTO GENERATED CODE BEGIN ===========
  App.set_name("testing");
  App.set_compilation_datetime(__DATE__ ", " __TIME__);
  LogComponent *logcomponent = App.init_log(115200, 512, UART_SELECTION_UART0);
  logcomponent->set_global_log_level(ESPHOME_LOG_LEVEL_DEBUG);
  WiFiComponent *wificomponent = App.init_wifi();
  wificomponent->set_use_address("192.168.0.41");
  WiFiAP wifiap = WiFiAP();
  wifiap.set_ssid("WIFIxxxx");
  wifiap.set_password("xxxxxx");
  wifiap.set_manual_ip(ManualIP{
      .static_ip = IPAddress(192, 168, 0, 41),
      .gateway = IPAddress(192, 168, 0, 1),
      .subnet = IPAddress(255, 255, 255, 0),
      .dns1 = IPAddress(1, 1, 1, 1),
      .dns2 = IPAddress(1, 0, 0, 1),
  });
  wificomponent->add_sta(wifiap);
  OTAComponent *otacomponent = App.init_ota();
  otacomponent->set_auth_password("xxxxx");
  otacomponent->start_safe_mode();
  WebServer *webserver = App.init_web_server(80);
  api::APIServer *api_apiserver = App.init_api_server();
  api_apiserver->set_password("xxxxx");
  // =========== AUTO GENERATED CODE END ============
  // ========= YOU CAN EDIT AFTER THIS LINE =========
  App.setup();
    Serial.begin(115200);
}

void loop() {
  App.loop();
  Serial.print(".>");
}

```

**Traceback (if applicable):**

```
src/main.cpp: In function 'void setup()':
src/main.cpp:29:17: error: 'class esphome::OTAComponent' has no member named 'set_auth_password'
otacomponent->set_auth_password("2394");
^
src/main.cpp:31:3: error: 'WebServer' was not declared in this scope

WebServer *webserver = App.init_web_server(80);
^
src/main.cpp:31:14: error: 'webserver' was not declared in this scope
WebServer *webserver = App.init_web_server(80);
^
src/main.cpp:31:30: error: 'class esphome::Application' has no member named 'init_web_server'
WebServer *webserver = App.init_web_server(80);
^
*** [.pioenvs/d1_mini/src/main.cpp.o] Error 1
```

**Additional information and things you've tried:**

OttoWinter, 2019-03-27:
You also need to copy the platformio.ini files with the build flags defined there. Also, esphome doesn't officially support using the system directly through platformio anymore.

gnkarn, 2019-03-27:
perfect , thankyou  !!

gnkarn, 2019-03-28:
this is a different matter , and of course for sure something on my code .

i was able to merge my code , and compile ok ., it even works , but is suffering resets
it runs, completes setup, completes  Loop() , and after a while it resets ...
```

Exception (29):
epc1=0x4020c372 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00d2114c depc=0x00000000

ctx: sys
sp: 3fffeb80 end: 3fffffb0 offset: 01a0

>>>stack>>>

```
and at the end 
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
␛[0;32m[I][logger:119]: Log initialized␛[0m
␛[0;35m[C][ota:479]: There have been 6 suspected unsuccessful boot attempts.␛[0m
```
the code i merged , was also based on asynctcp, asyncmqtt , i have replaced my wifi begin for the esphome wifi component,  mqtt im using my own .

 will continue investigating, but just in case you eye ball something that may be critical and i may be missing . tks

OttoWinter, 2019-07-11:
Closing because this way of compiling is not supported

closed 2019-07-11 by OttoWinter