esphome/issues #172: ESP32: ble_rssi does not create globally unique id's

syphernl, 2019-03-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- Pip (Python 2.7.15 / 3.6.7)
- Esphome 1.12.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-WROOM-32 (configured as board `lolin_d32`)

**Affected component:**

https://esphome.io/components/sensor/ble_rssi.html

**Description of problem:**
In my current setup I have two ESP32's (`livingroom` & `office`) with the same config (only the `name` differs in the sensor setting.

However, this confuses HomeAssistant since on the integration page: 
- RSSI sensor seems to be missing on livingroom
- Office contains the livingroom RSSI sensor

The problem lies in the fact that the `unique_id` in `.homeassistant/.storage/core.entity_registry` doesn't contain an unique ID: `ble-0000E1DD2FXXXXXX-rssi`

This unique ID appears to have been formed based on the MAC address, which is not globally unique across the devices.

The ID can probably better be changed to a normalized version of the sensor name (in my case that would be `miband_office_signal`) or appended with a value that ties this particular value to the device (e.g. esphome name or mac address) e.g. `ble-0000E1DD2FXXXXXX-rssi-miband-office-signal`.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: ble_rssi
    mac_address: E1:DD:2F:XX:XX:XX
    name: "Miband Office Signal"
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
Setting `id` explicitly didn't seem to help

NoedelVreter, 2019-04-17:
Fixed for me in latest dev branch. Thanks Otto!

OkhammahkO, 2019-04-29:
I was experiencing the same issue, and happen to be using it for Mi band tracking on 2 ESP32's too (very happy to be able to do that!).  I also happen to try setting id explicitly:).   I can also confirm that this is fixed in the dev branch.

 @syphernl  , I am pretty new to BLE presence detection and if you could give me any steer/share your config/tell me about reliability/performance/detection intervals you are achieving, that would be appreciated.  Is there a Home Assistant thread for this yet? Cheers.

OttoWinter, 2019-05-01:
Ok, closing this issue as the problem is solved.

@OkhammahkO I don't know of such a thread, but users are welcome to share their experiences here.

closed 2019-05-01 by OttoWinter