esphome/issues #174: Deep Sleep / Connection Time to API

S3NS375, 2019-03-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io v2.11
HA v0.90.2
ESPHome v1.12.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP12F

Sensors: BME280 / TSL2561 / ADC

**Affected component:**

Home Assistant API / Wifi

**Description of problem:**

Hello!

I try to make an battery driven device for capturing temp, brightness etc.
I have use deep sleep for testing but i have connection problems to the API. Or better say it last way too long.
It sometimes don´t connect even on 30s on-time. Most of the time it connects to the api within 20s.
So i tested it with 12 seconds and 10 minutes deep sleep over night. Only around 6 times it was able to deliver the data.

Can i do something to connect faster and more relieable to the API? If i let it run for a minute or more the battery is empty in some days.

I have added the log where it took 8 seconds to connect to the API after connecting to my WiFi network.

The second question is about the procedure. 

It is possible to let him completly connect to WiFi and API and then read and transmit the sensor data once and go to sleep? So that the DeepSleep Timer is only a safety option? If he in this time don´t manage to do anything correct it goes sleep anyway? 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: node_balkon
  platform: ESP8266
  board: esp12e

wifi:
  ssid: 'SSID'
  password: 'PASSWORD'
  fast_connect: true
  manual_ip:
    static_ip: 192.168.0.22
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1
    dns2: 192.168.0.1

# Enable logging
logger:
#  level: VERBOSE

# Enable Home Assistant API
api:
  password: 'PASSWORD'

ota:
  password: 'PASSWORD'
  
i2c:
  sda: 4
  scl: 5
  scan: false
  
  
  
[... 176 lines cut ...]
19:02:28.639 -> [0;36m[D][sensor.tsl2561:138]: Got illuminance=822.9lx[0m
19:02:30.092 -> [0;36m[D][sensor.bme280:217]: Got temperature=22.1°C pressure=1007.1hPa humidity=39.9%[0m
19:02:30.139 -> [0;36m[D][sensor.adc:087]: 'Node_Balkon Batterie': Got voltage=3.28V[0m
19:02:30.139 -> [0;36m[D][sensor.sensor:100]: 'Node_Balkon Batterie': Sending state 3.21332 V with 2 decimals of accuracy[0m
19:02:30.655 -> [0;36m[D][sensor.tsl2561:138]: Got illuminance=822.9lx[0m
19:02:32.061 -> [0;36m[D][sensor.adc:087]: 'Node_Balkon Batterie': Got voltage=3.28V[0m
19:02:32.108 -> [0;36m[D][sensor.sensor:100]: 'Node_Balkon Batterie': Sending state 3.21430 V with 2 decimals of accuracy[0m
19:02:32.108 -> [0;36m[D][sensor.bme280:217]: Got temperature=22.1°C pressure=1007.1hPa humidity=39.9%[0m
19:02:32.624 -> [0;36m[D][sensor.tsl2561:138]: Got illuminance=822.4lx[0m
19:02:34.123 -> [0;36m[D][sensor.bme280:217]: Got temperature=22.1°C pressure=1007.1hPa humidity=40.0%[0m
19:02:34.123 -> [0;36m[D][sensor.adc:087]: 'Node_Balkon Batterie': Got voltage=3.28V[0m
19:02:34.123 -> [0;36m[D][sensor.sensor:100]: 'Node_Balkon Batterie': Sending state 3.21430 V with 2 decimals of accuracy[0m
19:02:34.639 -> [0;36m[D][sensor.tsl2561:138]: Got illuminance=822.4lx[0m
19:02:36.092 -> [0;36m[D][sensor.adc:087]: 'Node_Balkon Batterie': Got voltage=3.28V[0m
19:02:36.092 -> [0;36m[D][sensor.sensor:100]: 'Node_Balkon Batterie': Sending state 3.21332 V with 2 decimals of accuracy[0m
19:02:36.092 -> [0;36m[D][sensor.bme280:217]: Got temperature=22.1°C pressure=1007.1hPa humidity=40.0%[0m
19:02:36.655 -> [0;36m[D][sensor.tsl2561:138]: Got illuminance=822.9lx[0m
19:02:38.108 -> [0;36m[D][sensor.bme280:217]: Got temperature=22.1°C pressure=1007.1hPa humidity=40.0%[0m
19:02:38.108 -> [0;36m[D][sensor.adc:087]: 'Node_Balkon Batterie': Got voltage=3.28V[0m
19:02:38.108 -> [0;36m[D][sensor.sensor:100]: 'Node_Balkon Batterie': Sending state 3.21430 V with 2 decimals of accuracy[0m
19:02:38.436 -> [0;32m[I][deep_sleep:072]: Beginning Deep Sleep[0m

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-29:
The native API is not made for deep sleep (yet). The ESP is the server in this case and so HA needs to check every few seconds whether the ESP is online without killing the system's resources. This interval happens to be 60s. For deep sleep MQTT should be used.

See https://github.com/esphome/feature-requests/issues/46 for more information.

I will close this issue as this is expected and the correct behavior right now.

closed 2019-03-29 by OttoWinter

S3NS375, 2019-03-29:
Ah, ok, thanks!

I will temporarily check out the MQTT function.