esphome/issues #176: Device unavailable in Home Assistant

bphermansson, 2019-03-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant in virtualenv, Esphome from Docker.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos
**Affected component:**

**Description of problem:**
I set up a Esphome device. I see it in the dashboard, it is online and I can read the logs. I can control it via Mqtt. I also can add it as an integration in Home Assistant. But there the device shows as unavailable, even the day after I set it up. What can be wrong?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-30:
That depends on your HA and MQTT setup. unavailable means the topic `<node_name>/status` can not be read by HA or written by the ESP. Please check your MQTT config, HA logs and with an MQTT message viewer.

bphermansson, 2019-03-30:
Ok, thanks. Somehow the IP in HA got wrong. I deleted the device and added a new with the correct IP, now it's working. I've found this by comparing the HA log with the Esphome log.

closed 2019-03-30 by bphermansson