esphome/issues #179: My device is dead. I need to understand before I start with the second device.

onlize, 2019-03-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Virtual Environment
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
Hello,

I used the Teckin SP20 switch. I know that it has an ESP8266 chip. I updated the firmware on this device to Sonoff-Tasmota.

I do not have Hass.io. I followed this article to create a firmware.bin file: https://esphome.io/guides/migrate_sonoff_tasmota.html
First, I created the switch.yaml file:
```
sphome:
  name: teckin_sp20_1
  platform: ESP8266
  board: esp01_1m
wifi:
  ssid: myssid
  password: mypassword
api:
logger:
ota:
```
After that, I ran:
```
esphome switch.yaml compile
```

Following the article https://esphome.io/guides/migrate_sonoff_tasmota.html, I uploaded the firmware.bin file using the Tasmota firmware option, and my device is dead. Is there any way to bring it back to life?

In any case, I have the second device, and I want to understand what I did wrong before I start uploading firmware on the second device.

Thank you.

onlize, 2019-03-30:
Actually, I was not able to run "esphome switch.yaml compile" because of the error:
```
pi@raspberrypi:~ $ esphome switch.yaml compile
-bash: esphome: command not found
pi@raspberrypi:~ $
```

I was able to do it when I run:
```
python2 -m esphome wsitch.yaml compile
```
I do not think it makes a difference, but wanted to mention it.

redstar2k1, 2019-04-01:
Are you sure it's dead, or it just doesn't react to the physical button press? Did you actually configured the relay/button/led/sensor or you just uploaded that generic yaml?
Check your router if you see the device connecting. Make sure you upload a proper complete yaml as described on [https://esphome.io/devices/sonoff.html#teckin](https://esphome.io/devices/sonoff.html#teckin)

onlize, 2019-04-01:
Thank you for your reply. I am sure it is dead because it is not getting IP address. I checked on the router.

The same happened with my second device. This issue can be closed, I am not going to purchase more switches like this. I tried to open them, but I cannot get to pins without physically breaking the plastic (or whatever it is made of) cover. I am not going to have broken switches in my house.

I will find and use a different device.

Thank you.

closed 2019-04-01 by onlize

brandond, 2019-04-01:
@onlize they're not dead just because they don't get an IP address. Have you actually tried to collect logs or reflash the device via UART bootloader? Unless you've actually fried the thing with the incorrect voltage or something, the odds are very low that the devices can't be recovered.